realsenseai/librealsense #11747: IntelrealSense D455 cannot detect the device in mac environment with python programming

HiWiQiao, 2023-04-28:
profile = pipe.start()
RuntimeError: No device connected

MartyG-RealSense, 2023-04-28:
Hi @HiWiQiao  A RealSense user at https://github.com/IntelRealSense/librealsense/issues/9687 created a pre-built version of pyrealsense2 for Mac, though it has only been tested with MacOS Catalina and Big Sur.

MartyG-RealSense, 2023-05-06:
Hi @HiWiQiao  Was the pre-built version of pyrealsense2 for Mac helpful to you, please?  Thanks!

MartyG-RealSense, 2023-05-12:
Hi @HiWiQiao  Do you require further assistance with this case, please?  Thanks!

HiWiQiao, 2023-05-14:
> Hi @HiWiQiao Was the pre-built version of pyrealsense2 for Mac helpful to you, please? Thanks!

I can successfully import pyrealsense2, but when I connect to D455, it shows that the device is not detected.

HiWiQiao, 2023-05-14:
> Hi @HiWiQiao Do you require further assistance with this case, please? Thanks!

Still not working, my macOS system is Ventura 13.3.1 and when I use the Intel version of MacBook Big sur is no problem.

MartyG-RealSense, 2023-05-15:
At the time of writing this, the most recent MacOS version that has pyrealsense2 setup advice available for it is Ventura's predecessor Monterey - see https://github.com/IntelRealSense/librealsense/issues/5275#issuecomment-1059876482

closed 2023-05-27 by MartyG-RealSense