realsenseai/librealsense #11748: D415 HDR on Color Sensor

electro-logic, 2023-04-28:
Hello,
The OV2740 sensor used by D415 according to the Omnivision product brief has some kind of HDR support
https://www.ovt.com/wp-content/uploads/2023/03/OV2740-PB-v1.3b-WEB.pdf
Any chance of having HDR on the color channel too?

MartyG-RealSense, 2023-04-29:
Hi @electro-logic  I will ask my Intel RealSense colleagues about your request.  Thanks!

electro-logic, 2023-04-29:
Thanks a lot

MartyG-RealSense, 2023-05-01:
Hi @electro-logic  My colleagues have responded to your question.  The HDR that you mentioned will need to use a companion ISP (Image Signal Processing) chip which is not in RealSense products.  It is therefore not supported.

electro-logic, 2023-05-02:
Hello Marty,
Thank you for your response. I am wondering if maybe we can only get the raw "row interleaved HDR", without merging the two half-resolution frames. Please see https://forums.developer.nvidia.com/t/jetson-xavier-nx-interleave-hdr/185225 for an example of row interleaved HDR. Thanks.

MartyG-RealSense, 2023-05-02:
I will consult again with my colleagues about your suggestion.  Thank you.

MartyG-RealSense, 2023-05-03:
My colleagues responded that if “row interleaved HDR” is done at the host side then you are welcome to try it out in manual exposure mode.

electro-logic, 2023-05-03:
Hello, thank you for asking.
Is there a way to access the color OV2740 cmos sensor register?
By writing the bit 0 to 0x3820 is possible to enable the HDR-lite mode that i'm interested in.
Thank you again

MartyG-RealSense, 2023-05-03:
I have passed your question to my colleagues.

MartyG-RealSense, 2023-05-04:
The answer was no, you cannot change the OV2740 register, unfortunately.

electro-logic, 2023-05-04:
Oks, thank you for asking anyway.

closed 2023-05-04 by electro-logic

Camilochiang, 2023-09-05:
Hei @MartyG-RealSense but actually tthere is a ISP chip in the D4 board....