realsenseai/librealsense #11749: D415 RGB 720p

electro-logic, 2023-04-28:
Hello,
How is the 1280x720 resolution obtained? I need to find the pixel size (in micrometers) of 720p images, I was thinking there was a simple cropping, but the FOV is the same of 1080p, so looks like there is an interpolation from 1080p.
Can anyone confirm me that RGB 1280x720 is obtained by scaling from 1920x1080?
Thank you

MartyG-RealSense, 2023-04-29:
Hi @electro-logic  In 2018 a RealSense team member advised: "During the rectification process, the image gets downsized.  In order to get the 1280x720 resolution after rectification, the extra pixels are extrapolated".

electro-logic, 2023-04-29:
Hello Marty,
Fx and Fy are the focal lenght

MartyG-RealSense, 2023-04-29:
Yes, sorry, you are correct.  My apologies.  My research found that the D415 pixel size is 1.4 µm × 1.4 µm

[image]

electro-logic, 2023-04-29:
Hello,
1.4um is the pixel size of the native resolution, the "equivalent" pixel size for a downsampled 1280x720 image I found out is

1920x1080x1.4x1.4=4064256 (pixel area)
4064256/(1280*720)=4.41
sqrt(4.41)=2.1um

Thanks anyway

MartyG-RealSense, 2023-04-29:
You are very welcome!

closed 2023-05-04 by electro-logic