realsenseai/librealsense #11755: Aligning pixel to 3D point on D435i (deprojection): undistortion, alignment, frame synchronization, best practices

Stack-it-up, 2023-05-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.14 | 
| Operating System & Version |  Ubuntu 18.04 (L4T)  | 
| Kernel Version (Linux Only)    |                                         | 
| Platform                                 | Jetson TX2  |
| SDK Version                            | 2                         | 
| Language                            |  C++                          | 
| Segment			| Robot                        | 

### Issue Description
Hello, I am trying to retrieve 3D point coordinates from pixels using a D435i. 

I have some questions about the preferred way of doing this, and I would like to be referred to relevant documentation if possible, since the questions I'm asking have been answered multiple times but never referring to "authoritative" resources such as SDK docs or device datasheets.

Relevant context: I am aligning depth to RGB frames since I will be using RGB for processing with a neural net. I am not sure this is the recommended approach, I just figured it would be easier for me to query the correct data points, since the NN output is provided in the RGB reference frame. I receive two streams (depth and color) at the same resolution (848x480) and framerate (30fps).

1. Are the color and depth streams already undistorted by the SDK? Would it be of any use to calibrate intrinsics and distortion parameters before performing depth-to-color alignment?
2. When using `rs2::pipeline::wait_for_frames()`, are the depth and color frames always synchronized or should I take care of that? Ideally, I would love the hardware to take care of frame synchronization, but I don't know how to achieve that (in case it is even possible).
3. All postprocessing should be applied to the depth frame after alignment to color, is this correct?

MartyG-RealSense, 2023-05-01:
Hi @Stack-it-up  Thanks very much for your questions.

1.  Most of the streams have a distortion model applied to them (they are **rectified**).  An exception to this is the **Y16** infrared format, which is unrectified as it is used for camera calibration.  Performing calibration with the **Dynamic Calibration** tool can be a useful way to improve alignment if you suspect that there is mis-alignment, as the tool calibrates both the depth and RGB sensors.

https://www.intel.com/content/www/us/en/download/645988/intel-realsense-d400-series-dynamic-calibration-tool.html

Alternatively, you could try resetting the camera to its factory-new default calibration using the instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487

2.  When wait_for_frames() is used, it will try to find the best timestamp match between depth and RGB frames.  Also, RGB-depth sync on the same camera  should kick in automatically once both RGB and depth have the same actual FPS.

3.  It is recommended that post-processing filters are applied **before** alignment in order to avoid distortion effects such as **aliasing** (jagged lines).

Stack-it-up, 2023-05-01:
Hello @MartyG-RealSense and thanks for the quick reply.

1.  You gave this verbatim answer on the Realsense help forum: I think it would be really useful if there was some reference table regarding which streams are rectified and which aren't. How do I know if frames in my stream need to be aligned using OpenCV?
2. I found some mentions of hardware sync being disabled by default on some cameras (which ones?). Do you mean to tell me that hardware sync is always enabled on D435 whenever the two streams have the same framerate?
3. I see, thanks for the heads up!

MartyG-RealSense, 2023-05-01:
1.  To the best of my knowledge, there is not a documentation table showing which streams are rectified or unrectified because most are rectified.  
 
The SDK's official **Projection** documentation at the link below states that "Left and right infrared images are rectified by default (Y16 format is not)", "The two infrared streams have identical intrinsics" and "The two infrared streams have no distortion".

https://dev.intelrealsense.com/docs/projection-in-intel-realsense-sdk-20#appendix-model-specific-details

Streams are not aligned by default in the RealSense SDK, so if you have not specified in your program that the streams should be aligned then they will not be aligned.

2.  https://github.com/IntelRealSense/librealsense/issues/10926#issuecomment-1298818196 has further information about sync between the depth and RGB sensors.  It states that on the D435 / D435i models, sync between the sensors is achieved by making RGB the 'master' and depth the 'slave' that follows RGB.

Stack-it-up, 2023-05-02:
Thank you! The links above have been useful to me