realsenseai/librealsense #11757: Getting error "Invalid Value in rs2_get_options(..... Auto Gain Limit Toggle)

Chris45215, 2023-05-02:
I get the error mentioned above (and in the attached screenshot) when starting the Realsense Viewer with the sensor attached. I have other D455's which do not have the error. If I click "OK" enough times (probably 10 times), then I can start the sensor - and both the RGB and depth seem to work fine - though the error message will periodically return. I am using Windows 10, fully updated.

A prior issue, https://github.com/IntelRealSense/librealsense/issues/11230, suggested reverting to firmware version 5.13.0.50 - but I cannot seem to figure out the way to do that. I found SDK v2.51.1, and downloaded the Intel.RealSense.FW.Update.exe file, which should be for firmware v5.13.0.50. But when I run it, it just shows the message "Nothing to do. Run again with -h for help", says that my sensor is on the most recent firmware v5.14, and then the console window automatically closes itself. 
There are instructions to update firmware with a rs-fw-update tool and command "rs-fw-update -l", but I think those instructions are missing a step - I probably need to direct the command-prompt window to the directory with that tool, but I cannot find that directory. And that page directs me to the .exe installer if I use Windows 10, which I do.

Also, the sensor works fine when I use the rs-pointcloud and rs-multicam tools. I use this sensor regularly, and in fact it's the sensor that I take with me when I work remotely. 

[image]

MartyG-RealSense, 2023-05-02:
Hi @Chris45215  To resolve this error, if you have firmware 5.14.0.0 installed in your camera then you must use RealSense SDK version **2.53.1**.  Firmware 5.14.0.0 is designed for 2.53.1.

If you need to use SDK 2.51.1 then I confirm that the firmware version must be downgraded to **5.13.0.50** which is the recommended firmware for 2.51.1.

If you are using Windows then you can obtain the 2.53.1 Viewer by visiting the link below and downloading and running the file **Intel.RealSense.Viewer.exe** in the 'Assets' file list at the bottom of the page.  This Viewer should be run from the location on your computer that the .exe file is downloaded to and not with a shortcut launch icon on the Windows desktop if you have one.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.53.1

You can also install the full 2.53.1 RealSense SDK for Windows by downloading and running the **Intel.RealSense.SDK-WIN10** installer file from the same Assets list.  Doing so will also install the 2.53.1 Viewer and place a launch shortcut for it on the Windows desktop.  Do use the shortcut for the SDK-installed Viewer.

Chris45215, 2023-05-03:
Thanks, updating the viewer to v2.53.1 seems to work - I don't know how I missed that.

The viewer v2.53.1 starts and runs with the sensor, but it does give me the red notice box in the top-right corner that reports "Right MIPI error". 
I closed the viewer and started it again, and it ran without producing the red box, but the debug console window did report the error "ds5-options.cpp:108 - Asic Temperature value is not valid!". When I closed the stream, it reported: "ds5-thermal-monitor.cpp:94 - Error during thermal compensation handling: query is available during streaming only". I am using it in the office, which is 70 degrees Fahrenheit and a normal room temperature.

I restarted the Viewer one more time, and it started up perfectly without any error at all. But when I disable the stereo data, the debug console reports "synthetic-stream.cpp:48 - Exception was thrown during callback: Out of frame resources!" (it's fine that it has run out of frames, I expect it to run out of frames because I turned off that stream, but I don't know if it should report this as an error).

So, it's working, but throwing occasional errors. As a programmer myself, I completely understand if those errors are 'supposed' to happen and are just a result of the startup or shutdown process of an external device; but I wanted to mention them just in case.

MartyG-RealSense, 2023-05-04:
The errors that you mention sound like momentary glitches rather than serious problems.  For example, the **Asic Temperature value is not valid** message can occur if the camera is not ready to respond to a request to check the camera temperature at the instant in time that the request was made but then report the temperature correctly afterwards.

Chris45215, 2023-05-04:
Thanks Marty! I'll close the issue, as all seems well.

closed 2023-05-04 by Chris45215