realsenseai/librealsense #11763: 'Place the chessboard on the plane where the object needs to be detected..'

pgswag, 2023-05-03:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 }                                                   | 
| Firmware Version                   | 05.13.00.50| 
| Operating System & Version |   {Win 10   | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python }  | 
| Segment			|  {others }                          | 

### Issue Description
<When i run the"box_dimensioner_multicam_demo" on pycharm,it always show like below.But,I have placed the checkerboard in the area being checked>
[image]

MartyG-RealSense, 2023-05-03:
Hi @pgswag  The box_dimensioner_multicam.py example program is sensitive to the distance of the camera from the chessboard. There will be a 'sweet spot' that is not too close to the board and not too far.  I recommend trying a distance that is not less than 0.5 m and not more than 1 m if possible.

pgswag, 2023-05-03:
> I adjusted the distance again, so that the distance is 0.7 meters, and the above information is still displayed
[image]

MartyG-RealSense, 2023-05-03:
The box_dimensioner_multicam program provides an example image of how to orient the camera and chessboard, putting the camera off to the side of the chessboard instead of directly above it.

[image]

pgswag, 2023-05-04:
I have readjust the camera position and angle, but it still doesn't solve the problem, whether the following may lead to the failure to detect the checkerboard:
1.Whether to use the specified checkerboard, for example, the number of black and white checkerboard is specified.
2.Whether I need to make sure that there are not too many irrelevant things in the camera's field of view.

pgswag, 2023-05-04:
> The box_dimensioner_multicam program provides an example image of how to orient the camera and chessboard, positioning the camera off to the side of the chessboard instead of directly above it.
> 
> [image]

I have readjust the camera position and angle, but it still doesn't solve the problem, whether the following may lead to the failure to detect the checkerboard:
1.Whether to use the specified checkerboard, for example, the number of black and white checkerboard is specified.
2.Whether I need to make sure that there are not too many irrelevant things in the camera's field of view.

MartyG-RealSense, 2023-05-04:
The program is configured to use a 6x9 chessboard by default, which you are doing.

Are you using a print-out of the PNG chessboard image provided by the box_dimensioner_multicam project at the link below, please?

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/box_dimensioner_multicam/pattern_chessboard.png

That png has some faint light grey writing on the bottom of it, but I do not see that text on your printout when zooming into it.

[image]

[image]

It is posible to modify box_dimensioner_multicam to use it with a single camera without the chessboard if the camera's intrinsics and extrinsics are provided, as described at https://github.com/IntelRealSense/librealsense/issues/9951#issuecomment-966039933

pgswag, 2023-05-04:
Thank you very much for your detailed answer！When I replaced the checkerboard you gave me, the problem was solved. Thank you again！！！

MartyG-RealSense, 2023-05-04:
It's great to hear that changing the chessboard to the official image resolved your problem.  Thanks very much for the update!

closed 2023-05-05 by pgswag