realsenseai/librealsense #11764: GenLock feature support

nv-kn, 2023-05-03:
Hi, I have a question about GenLock feature. It seems to be the case that the documentation was removed from Intel RealSense website, is there a reason for that? Does that mean the feature is not supported or is going to be removed?

I am asking because it is an important feature for my project, even though it has its own issues. Is there maybe a plan to implement a new feature that is more stable and robust? Knowing that would be valuable so I can plan the future lifecycle for my project.

It would be extremely helpful if anyone more knowledgeable could provide me with more information about this!

MartyG-RealSense, 2023-05-03:
Hi @nv-kn  The documentation for the genlock feature was removed because it was an experimental proof of concept rather than an official Intel RealSense feature and it was immature and unvalidated.  The officially supported, mature and validated method of hardware sync is Mode 1 and 2.

However, genlock mode is still supported within the SDK at the time of writing this if you need to use it.  You can access an archived PDF document version of the removed documentation at the link below.

[External Synchronization.pdf](https://github.com/IntelRealSense/librealsense/files/11382074/External.Synchronization.pdf)

MartyG-RealSense, 2023-05-03:
I will highlight your question to my Intel RealSense colleagues.  Thanks very much for your patience!

MartyG-RealSense, 2023-05-09:
Not yet.  I have sent a follow-up message to my colleagues about it.  Thanks very much for your patience!

MartyG-RealSense, 2023-05-15:
Hi @nv-kn  Bearing in mind the advice in the comment above, do you require further assistance with this case please?  Thanks!