realsenseai/librealsense #11767: How to access RS depth output without rendering to screen?

nataliesettles, 2023-05-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.14.00.00 | 
| Operating System & Version |  Win 10 |  
| Platform                                 | PC |
| SDK Version                            |  2.53.1.4623                        | 
| Language                            |  unity                          | 

### Issue Description
Hi - How can I output the texture/buffer of depth points from the rs camera without rendering an image to the screen? I'd like to use the points as a data source to inform an interactive response from the visualization I've built. I have successfully managed to use the stream of data from the RawImages in the TexturesDepthAndColor Scene in the Unity examples, but in the end, I don't want the depth images to render to the screen as a UI - I'd rather simply have a buffer or RenderTexture or similar that I can simply use as a data source. Is there a way you'd recommend doing this? - perhaps using the Depth or ColorizedDepth children of the RSDevice? Thanks!

MartyG-RealSense, 2023-05-05:
Hi @nataliesettles  One way to clear the pointcloud from the screen would be to set the **Culling Mask** on the Main Camera to **Nothing** so that the pointcloud data is still being processed but it is not visually rendered.

[image]

********

This approach could cause problems though if you need other visual elements to be rendered on the screen, as the Nothing cull removes everything.

A more precise approach would be to create a new layer name using **Add Layer**.

[image]

You could then go to the **Pointcloud** object that is a child of RsDevice in the PointCloudDepthAndColor scene and use the Layer drop-down in the top corner of the Unity window to change the Pointcloud object's layer to the new one that you just created.

[image]

Then go to the Main Camera's culling layers list and untick the name of the new layer.  This will cause the pointcloud to disappear whilst all other layer types can still be rendered.

nataliesettles, 2023-05-05:
Yes - thanks. That wasn't quite where I was stuck, however, I just solved it moments ago.

I was having trouble accessing the texture data stream without using the RawImages to pass the texture stream between scripts. It turns out I simply needed to initialize a connection to the texture stream by binding a public setter method to a texture field in my own script to the textureBinding event in the RsStreamTextureRenderer script, and then referencing that newly bound field in the update method of my script in order to continue to get the updates to the stream.

MartyG-RealSense, 2023-05-05:
It's great to hear that you achieved a solution.  Thanks very much for sharing the details for the benefit of other RealSense Unity users who may be in a similar situation in future!

nataliesettles, 2023-05-05:
Thanks for this alternate approach with culling - I hadn't thought of that.
I guess I'd assumed that if the content was culled it might not continue to
render in the background - but I hadn't tested to see. Cheers!

*Natalie Settles*

*A Pilgrim in Art and Science:*

Part memoir, part travelogue, part philosophical reflection

nataliesettles.com/pilgrim

closed 2023-05-05 by nataliesettles