realsenseai/librealsense #11768: sporadic no such interface supported error

rajrajeswaran, 2023-05-04:
I am currently investigating an issue where our application stops receiving any frames from D415 camera using C# SDK.  Our production clients are reporting a sporadic issues where they loose frames from Intel RealSense D415 embedded (bare metal D415) camera.  I have turned on LRS_LOG_LEVEL to DEBUG to capture more details.  Review LRS_LOG file more information.  I have also captured the logs from our c# application.  Review APPLICATION_LOG.  In addition, we also get "Set Address Failed" error in device manager.  Once the "Set Address Failed" error start, we will have to shutdown the PC, unplug power, wait 30 seconds and then start the PC to resolve it.  I need help understanding what causes "No such interface supported" error sporadically.   We request following steams via SDK

Depth: 15 FPS 720p
Color: 15 FPS 1080p BGRA8
IR: 15 FPS 720p

Sample code also attached.

[LRS_LOG.txt](https://github.com/IntelRealSense/librealsense/files/11400332/LRS_LOG.txt)
[APPLICATION_LOG.txt](https://github.com/IntelRealSense/librealsense/files/11400361/APPLICATION_LOG.txt)
[image]
[Sample Code.cs.txt](https://github.com/IntelRealSense/librealsense/files/11400494/Sample.Code.cs.txt)

Camera Model: D415 - Bare metal (embedded)
Firmware Version: 05.12.07.100
Operating System & Version: Windows 10 iOT LTSC
SDK Version: 2.38.1
Language: C# (.Net 4.8)

MartyG-RealSense, 2023-05-05:
Hi @rajrajeswaran  From looking at the Device Manager image, it seems that the camera is disconnected and disappears from the Device Manager (as there is no 'Cameras' category listed).  It may attempt to re-connect but the camera cannot be successfully initialized and so appears as 'Unknown USB Device (Set Address Failed) with a yellow warning sign beside it.

Looking at the C# code, I do not see an obvious problem with it.

Is the D415 module using a USB cable?  If so, is it a cable that you chose yourself?  The **USB CAM overflow** message on the other PC could in itself be a sign of an issue with the USB cable.  It is important that USB cables used with RealSense are high quality ones suited to data transfer and not inexpensive ones designed only for device sync or charging.

MartyG-RealSense, 2023-05-05:
You are very welcome!  As you suggest, I will close the case. Thanks!

closed 2023-05-05 by MartyG-RealSense