realsenseai/librealsense #11769: Is there any function look like: rs2_project_color_pixel_to_depth_pixel in Csharp?

tri3891, 2023-05-05:
In C++, there is a function: rs2_project_color_pixel_to_depth_pixel. How can we use it in C#?

MartyG-RealSense, 2023-05-05:
Hi @tri3891  There is not an example script available for using rs2_project_color_pixel_to_depth_pixel in C#, though a RealSense C# tutorial at the link below has scripting under the **Mapping 2D to 3D** heading for transforming 2D screen coordinates to 3D world coordinates (like how rs2_project_color_pixel_to_depth_pixel converts a 2D color pixel to a 3D depth pixel).

https://lightbuzz.com/intel-realsense-coordinate-mapping/

MartyG-RealSense, 2023-05-05:
Depth-color alignment is typically more difficult to achieve in C# than it is in C++ or Python languages.

The C# wrapper has an example project called **Processing** that demonstrates alignment, though it does not output the XYZ coordinates.  See the **Window.xaml.cs** file for the program code.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/csharp/tutorial/processing

closed 2023-05-16 by MartyG-RealSense