realsenseai/librealsense #11771: Is there any RGB image quality report for D455 series camera

seliber, 2023-05-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS } | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 |  NVIDIA Jetson / etc..  |
| SDK Version                            |  { 2.50.1 }                          | 
| Language                            |  {C/C++ }                          | 
| Segment			|  {others }                          | 

### Issue Description

Is there any image quality test report for RGB stream? I can find Depth image quality in the whitepaper, but cannot find any pages related with RGB image quality.

MartyG-RealSense, 2023-05-05:
Hi @seliber  There is not a white paper about RGB quality testing that is an equivalent of the depth quality testing paper.

seliber, 2023-05-06:
> 

Thus do we have any related unformal report or test steps about the RGB image quality?

MartyG-RealSense, 2023-05-06:
The only RGB controls that most people are likely to adjust are **Exposure** and **White Balance**.  They might also maximize the value of the **Sharpness** option to help to remove blurring.

You can also use the **Dynamic Calibration** tool to calibrate RGB in addition to depth calibration.

https://www.intel.com/content/www/us/en/download/645988/intel-realsense-d400-series-dynamic-calibration-tool.html

Page 52 of the PDF user guide for the tool has information about RGB calibration and its settings.

https://www.intel.co.uk/content/www/uk/en/support/articles/000026723/emerging-technologies/intel-realsense-technology.html

seliber, 2023-05-06:
Thanks a lot for the information!

closed 2023-05-06 by seliber