realsenseai/librealsense #11772: D435i camera unavailable in realsense viewer

hassan-tht, 2023-05-05:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.14.00.00 | 
| Operating System & Version |   Ubuntu 20  | 
| Kernel Version (Linux Only)    |  5.15.0-71-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.53.1                           | 

### Issue Description
Hi, I have the D435i camera and it was working pretty well with both `realsense-viewer` and with the `ROS` 
wrapper. I stopped using it for some time and when I tried using it again, I can't get it to work on the same machine I used it with. I tried with another machine, and it works. My initial guess was the versions and so I updated to `2.53` following the instructions [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#upgrading-the-packages), but still no luck. I thought this might be due to having the `ROS` wrapper installed which might create some confusion. So, I uninstalled the `ROS` wrapper and tried again and still it won't work. Of course all of this included the many times of restarting the machine and un/pluging the camera.

When I launch the `realsense-viewer`, I can't get any stream into it and it gets unresponsive that I gave to kill it using the terminal. Below you can find the `realsense-viewer` and the output in the terminal.

[image]

```
hassan> realsense-viewer 
 05/05 12:34:11,618 INFO [140709388408640] (backend-v4l2.cpp:656) Enumerating UVC video0 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.4/2-8.4:1.0/video4linux/video0
 05/05 12:34:11,854 INFO [140709388408640] (backend-v4l2.cpp:656) Enumerating UVC video1 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.4/2-8.4:1.0/video4linux/video1
 05/05 12:34:11,854 INFO [140709388408640] (backend-v4l2.cpp:656) Enumerating UVC video2 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.4/2-8.4:1.0/video4linux/video2
 05/05 12:34:11,855 INFO [140709388408640] (backend-v4l2.cpp:656) Enumerating UVC video3 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.4/2-8.4:1.0/video4linux/video3
 05/05 12:34:11,855 INFO [140709388408640] (backend-v4l2.cpp:656) Enumerating UVC video4 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.4/2-8.4:1.3/video4linux/video4
 05/05 12:34:11,855 INFO [140709388408640] (backend-v4l2.cpp:656) Enumerating UVC video5 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.4/2-8.4:1.3/video4linux/video5
 05/05 12:34:11,952 INFO [140709388408640] (synthetic-stream-gl.cpp:80) Initializing rendering, GLSL=1
 05/05 12:34:11,952 INFO [140709388408640] (synthetic-stream-gl.cpp:89)  0 GPU objects initialized
 05/05 12:34:11,952 INFO [140709388408640] (synthetic-stream-gl.cpp:132) Initializing processing, GLSL=1
 05/05 12:34:11,977 INFO [140709388408640] (synthetic-stream-gl.cpp:146)  0 GPU objects initialized
 05/05 12:34:12,019 INFO [140709388408640] (backend-v4l2.cpp:656) Enumerating UVC video0 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.4/2-8.4:1.0/video4linux/video0
 05/05 12:34:12,019 INFO [140709388408640] (backend-v4l2.cpp:656) Enumerating UVC video1 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.4/2-8.4:1.0/video4linux/video1
 05/05 12:34:12,019 INFO [140709388408640] (backend-v4l2.cpp:656) Enumerating UVC video2 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.4/2-8.4:1.0/video4linux/video2
 05/05 12:34:12,019 INFO [140709388408640] (backend-v4l2.cpp:656) Enumerating UVC video3 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.4/2-8.4:1.0/video4linux/video3
 05/05 12:34:12,019 INFO [140709388408640] (backend-v4l2.cpp:656) Enumerating UVC video4 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.4/2-8.4:1.3/video4linux/video4
 05/05 12:34:12,020 INFO [140709388408640] (backend-v4l2.cpp:656) Enumerating UVC video5 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.4/2-8.4:1.3/video4linux/video5
 05/05 12:34:12,020 INFO [140709388408640] (context.cpp:382) Found 1 RealSense devices (mask 0xff)
 05/05 12:34:12,092 INFO [140709287593728] (backend-v4l2.cpp:656) Enumerating UVC video0 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.4/2-8.4:1.0/video4linux/video0
 05/05 12:34:12,092 INFO [140709287593728] (backend-v4l2.cpp:656) Enumerating UVC video1 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.4/2-8.4:1.0/video4linux/video1
 05/05 12:34:12,092 INFO [140709287593728] (backend-v4l2.cpp:656) Enumerating UVC video2 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.4/2-8.4:1.0/video4linux/video2
 05/05 12:34:12,092 INFO [140709287593728] (backend-v4l2.cpp:656) Enumerating UVC video3 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.4/2-8.4:1.0/video4linux/video3
 05/05 12:34:12,092 INFO [140709287593728] (backend-v4l2.cpp:656) Enumerating UVC video4 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.4/2-8.4:1.3/video4linux/video4
 05/05 12:34:12,092 INFO [140709287593728] (backend-v4l2.cpp:656) Enumerating UVC video5 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.4/2-8.4:1.3/video4linux/video5
 05/05 12:34:12,092 INFO [140709287593728] (backend-v4l2.cpp:656) Enumerating UVC video0 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.4/2-8.4:1.0/video4linux/video0
 05/05 12:34:12,093 INFO [140709287593728] (backend-v4l2.cpp:656) Enumerating UVC video1 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.4/2-8.4:1.0/video4linux/video1
 05/05 12:34:12,093 INFO [140709287593728] (backend-v4l2.cpp:656) Enumerating UVC video2 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.4/2-8.4:1.0/video4linux/video2
 05/05 12:34:12,093 INFO [140709287593728] (backend-v4l2.cpp:656) Enumerating UVC video3 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.4/2-8.4:1.0/video4linux/video3
 05/05 12:34:12,093 INFO [140709287593728] (backend-v4l2.cpp:656) Enumerating UVC video4 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.4/2-8.4:1.3/video4linux/video4
 05/05 12:34:12,093 INFO [140709287593728] (backend-v4l2.cpp:656) Enumerating UVC video5 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.4/2-8.4:1.3/video4linux/video5
 05/05 12:34:12,094 INFO [140709388408640] (rs.cpp:2701) Framebuffer size changed to 1879 x 1376
 05/05 12:34:12,094 INFO [140709388408640] (rs.cpp:2701) Window size changed to 1879 x 1376
 05/05 12:34:12,094 INFO [140709388408640] (rs.cpp:2701) Scale Factor is now 1
 05/05 12:34:12,094 INFO [140709287593728] (backend-v4l2.cpp:656) Enumerating UVC video0 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.4/2-8.4:1.0/video4linux/video0
 05/05 12:34:12,094 INFO [140709287593728] (backend-v4l2.cpp:656) Enumerating UVC video1 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.4/2-8.4:1.0/video4linux/video1
 05/05 12:34:12,094 INFO [140709287593728] (backend-v4l2.cpp:656) Enumerating UVC video2 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.4/2-8.4:1.0/video4linux/video2
 05/05 12:34:12,094 INFO [140709287593728] (backend-v4l2.cpp:656) Enumerating UVC video3 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.4/2-8.4:1.0/video4linux/video3
 05/05 12:34:12,095 INFO [140709287593728] (backend-v4l2.cpp:656) Enumerating UVC video4 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.4/2-8.4:1.3/video4linux/video4
 05/05 12:34:12,095 INFO [140709287593728] (backend-v4l2.cpp:656) Enumerating UVC video5 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.4/2-8.4:1.3/video4linux/video5
 05/05 12:34:12,362 INFO [140709388408640] (backend-v4l2.cpp:656) Enumerating UVC video0 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.4/2-8.4:1.0/video4linux/video0
 05/05 12:34:12,362 INFO [140709388408640] (backend-v4l2.cpp:656) Enumerating UVC video1 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.4/2-8.4:1.0/video4linux/video1
 05/05 12:34:12,362 INFO [140709388408640] (backend-v4l2.cpp:656) Enumerating UVC video2 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.4/2-8.4:1.0/video4linux/video2
[... cut ...] /2-8/2-8.4/2-8.4:1.0/video4linux/video3
 05/05 12:34:12,362 INFO [140709388408640] (backend-v4l2.cpp:656) Enumerating UVC video4 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.4/2-8.4:1.3/video4linux/video4
 05/05 12:34:12,362 INFO [140709388408640] (backend-v4l2.cpp:656) Enumerating UVC video5 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8.4/2-8.4:1.3/video4linux/video5
 05/05 12:34:12,362 INFO [140709388408640] (context.cpp:382) Found 1 RealSense devices (mask 0xfe)
```

MartyG-RealSense, 2023-05-05:
Hi @hassan-tht  It sounds as though it is a software issue (the RealSense SDK) rather than a hardware issue (the camera or USB cable / USB port) as the camera works correctly on another computer.

As you used the distribution_linux instructions to install the SDK from Debian packages, did you then install the ROS1 wrapper (Kinetic, Melodic, Noetic) from packages too with apt-get using **Method 1**?  

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy#method-1-the-ros-distribution

If you did use Method 1 then this installs both the SDK and wrapper at the same time.  So if you installed the SDK with the distribution_linux instructions and then used ROS wrapper method 1 then you could have two SDK installations on the same computer that may conflict with each other.

MartyG-RealSense, 2023-05-05:
When you start the Viewer, are there any other RealSense programs such as the ROS wrapper that are already active and streaming?  Does the Viewer work if you launch it as the first thing you do after start-up of the computer has completed and do not open any other programs first?

MartyG-RealSense, 2023-05-05:
Are you able to launch any other RealSense tools and examples such as rs-capture or rs-hello-realsense.  These should be able to be found in the **usr/local/bin** folder of Ubuntu.  

If rs-hello-realsense (which does not have graphics) worked but rs-capture (which does use graphics) freezes when launched, it might indicate that there is a problem with the graphics drivers on that particular PC and the drivers might need updating.

hassan-tht, 2023-05-05:
Thanks for all the help, I figured out the problem but don't know the reason. Actually, both `rs-hello-realsense` and `rs-capture` had the same problem. So, the (wireless) mouse was connected to the `USB` port right beside the one where the camera is connected (both same `PCI`). When I disconnected the mouse, everything worked out (including the `ROS` wrapper).
NB: I mentioned the wireless mouse because the wired one doesn't create that problem.

closed 2023-05-05 by hassan-tht

MartyG-RealSense, 2023-05-05:
Thanks very much for the update.  It's great to hear that you were successful.

I have only seen that wireless mouse issue once before.  It is because RealSense 400 Series cameras can interfere with non-RealSense devices nearby that use infrared signals.