realsenseai/librealsense #11773: D435i works when on USB2, but D415 doesn't

mtandy1, 2023-05-05:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415, D435i | 
| Firmware Version                   | 05.14.00.00 | 
| Operating System & Version |   Linux (Ubuntu 22.04)  | 
| Kernel Version (Linux Only)    |  5.19.0-41-generic               | 
| Platform                                 | PC  |
| SDK Version                            |  2.53.1                 | 
| Language                            |  n/a      | 
| Segment			|  n/a                       | 

### Issue Description

#### USB2 video works on the D435i

When the D435i is connected via USB 2, it advertises a lower framerate - but it can still be used.

I tested with a D435i (S/N 013222072862, firmware 05.14.00.00) and when connected via USB2 the RGB sensor was still happy to provide me with 1280x720 video at 6fps. The RGB feed could be viewed in realsense-viewer, could be captured via v4l2 with the command `ffmpeg -f video4linux2 -framerate 6 -video_size 1280x720 -i /dev/video8 -t 00:10:00 temp.mp4` and the resolution and framerate were reported as available when checked with `v4l2-ctl --list-formats-ext --device=/dev/video8`

[image]
[image]

The D435i also, of course, works fine when on USB3.

#### USB2 video does not work on the D415

I connected a D415 (S/N 102122061880, firmware 05.14.00.00) via USB3 and confirmed it was in full working order.

However, when switched to USB2 it _appeared_, it _reported video formats as available_ and it _allowed me to choose the video format in realsense-viewer_ - but it didn't display any frames.

[image]

```
 05/05 16:40:17,126 WARNING [139719757252160] (backend-v4l2.cpp:1435) Frames didn't arrived within 5 seconds
 05/05 16:40:17,126 INFO [139719757252160] (types.h:912) Frames didn't arrived within 5 seconds
 05/05 16:40:22,130 WARNING [139719757252160] (backend-v4l2.cpp:1435) Frames didn't arrived within 5 seconds
 05/05 16:40:22,130 INFO [139719757252160] (types.h:912) Frames didn't arrived within 5 seconds
```
I was also unable to capture any frames using v4l2 directly (such as `ffmpeg -f video4linux2 -framerate 6 -video_size 1280x720 -i /dev/video8 -t 00:10:00 temp.mp4`).

However, the D415 on USB2 does *report* that video is available:
```
$ v4l2-ctl --list-formats-ext --device=/dev/video8
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'YUYV' (YUYV 4:2:2)
		Size: Discrete 424x240
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.167s (6.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.167s (6.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.167s (6.000 fps)
		Size: Discrete 1920x1080
			Interval: Discrete 0.125s (8.000 fps)
```

#### Things seem better for D415s on older firmware?

I also had a D415 with firmware version `05.11.01.100` (S/N 105422061206) which seems to provide frames over USB2 (albeit with some warnings, and without the 1920x1080 resolution offered)

[image]

#### Things I tried, to isolate the root cause

I tried _rebooting_ and _detaching all other USB devices_. I wasn't able to detach my laptop's built in camera and trackpad, naturally. In all the tests the _camera was plugged directly into my laptop_, never via a USB hub. I tested on _every USB port on my laptop_ and tried classic options like flipping over the USB-C connector.

I tested _the depth feeds_ and they exhibited the same behaviour. I tested _using the camera from python_ and saw the same behaviour as realsense-viewer showed.

I tested with _several different D415s_ on the latest firmware - S/N 102122061880, S/N 105422060904, S/N 148122061154 - all of which refused to provide frames over USB2.

I tested with _a different D435i_ (S/N 148122061154) and confirmed that it too works OK over USB2.

**OS Version might make a difference,** as I tested on an Ubuntu 16.04 machine with kernel 4.15.0-142-generic and things seemed to work somewhat better. But as that host didn't have a screen, I wasn't able to confirm things fully with realsense-viewer etc. And as you'll appreciate, downgrading to an end-of-life operating system isn't on the cards :grin: 

#### Comparing the USB traffic in wireshark
As we have a camera that isn't working and one that is, we can compare the USB traffic:

[image]

As you can see in row 207, the working camera starts sending large `URB_BULK in` messages within a second, and keeps on doing so. Whereas the non-working camera sends nothing for 15 seconds, sending only some short messages as the stream ends.

#### Thanks for reading this far!

I'd appreciate it if you could test the same feature on your end and tell me whether you see the same problem :smile:

MartyG-RealSense, 2023-05-06:
Hi @mtandy1 I tested the RGB on a D415 with firmware 5.14.0.0 on a USB 2 connection at 1280x720, YUYV format and 6 FPS but was not able to replicate the No Frames Received problem that you experienced.  I tested with the Viewer on two different PCs.

Have you tried using 1280x720 resolution at 15 FPS instead of 6 or a lower resolution such as 640x480, please?

mtandy1, 2023-05-09:
@MartyG-RealSense Yes, I've tried every resolution and framerate, none of them work. Here's an example at the 15fps setting you suggest:

[image]

MartyG-RealSense, 2023-05-09:
If you are capturing mp4 video through V4L2 then it sounds as though you built the librealsense SDK from source code in 'native' mode (RSUSB backend = **false**), and patched the kernel.  Is that correct, please?

I note that you had a D415 that worked fine on USB 3.  Is it a requirement of your project that USB 2 is used?

mtandy1, 2023-05-09:
No, we haven't built the librealsense SDK from source or anything like that, I'm just using the standard realsense-viewer and so on. I mention V4L2 merely to establish that I'm aware that the camera offers different framerates on USB2 and USB3, that I made sure I was using a framerate supported on USB2, and that as well as testing with Python code and realsense-viewer I've also tested using the D415 as a vanilla UVC device.

USB2 is indeed a requirement for the task I'm working on.

I've tested on a number of firmware versions, and can report the following:-

| Firmware  |  USB2 test results      |
| --------- | ------------------- |
| 05.12.00.00 | Partial success (12/24 frames) |
| 05.12.06.00 | Success (24/24 frames) |
| 05.12.07.100 | Partial success (12/24 frames) |
| 05.12.08.200 | Partial success (12/24 frames) |
| 05.12.09.00 | Partial success (12/24 frames) |
| 05.12.10.00 | Success (24/24 frames) |
| 05.12.11.00 | Partial success (12/24 frames) |
| 05.12.12.100 | Success (24/24 frames) |
| 05.12.13.50 | failed |
| 05.12.15.50 | failed |
| 05.13.00.50 | failed |
| 05.14.00.00 | failed |

So I think we're looking at a regression happened some time around firmware version 05.12.13.50.

mtandy1, 2023-05-09:
Sorry, perhaps I wasn't clear. When I say "we haven't built the librealsense SDK from source or anything like that" what I mean is: We [followed the instructions to install via apt-get,](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages) rather than checking out the librealsense source code and compiling it ourselves. realsense-viewer is provided by the `librealsense2-utils` package.

I'm afraid I don't know if we're using the "RSUSB backend" - we're using whatever comes as default when installing from apt-get.

```
$ apt list --installed | grep -i realsense
librealsense2-dkms/jammy,now 1.3.19-0ubuntu1 all [installed]
librealsense2-gl/jammy,now 2.53.1-0~realsense0.8251 amd64 [installed,automatic]
librealsense2-net/jammy,now 2.53.1-0~realsense0.8251 amd64 [installed,automatic]
librealsense2-udev-rules/jammy,now 2.53.1-0~realsense0.8251 amd64 [installed,automatic]
librealsense2-utils/jammy,now 2.53.1-0~realsense0.8251 amd64 [installed]
librealsense2/jammy,now 2.53.1-0~realsense0.8251 amd64 [installed,automatic]
```

MartyG-RealSense, 2023-05-09:
Thanks very much for the clarification.  An apt-get package installation for Ubuntu 22.04 Jammy is compatible with kernel 5.15, and kernel 5.19 is not officially supported at this time.  

[image]

closed 2023-05-21 by MartyG-RealSense