realsenseai/librealsense #11774: fallocate: fallocate failed: No space left on device libuvc_installation.sh build

darshit-desai, 2023-05-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
When I try to run the libuvc_installation.sh script it says that it has not space left at the end

```
rosnoetic:yoctohome$ ./libuvc_installation.sh
+ exec
+ exec
+ '[' 8 -ne 0 ']'
+ echo -e '\e[32m'

+ read -p 'Remove all RealSense cameras attached. Hit any key when ready'
Remove all RealSense cameras attached. Hit any key when ready
+ echo -e '\e[0m'

+ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.5 LTS
Release:	20.04
Codename:	focal
++ uname -r
+ echo 'Kernel version 3.18.71-perf'
Kernel version 3.18.71-perf
+ sudo apt-get update
Hit:1 http://packages.ros.org/ros/ubuntu focal InRelease
Hit:2 http://ports.ubuntu.com/ubuntu-ports focal InRelease                                                 
Hit:3 http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease                                         
Hit:4 http://ports.ubuntu.com/ubuntu-ports focal-backports InRelease
Hit:5 https://librealsense.intel.com/Debian/apt-repo focal InRelease
Hit:6 http://ports.ubuntu.com/ubuntu-ports focal-security InRelease
Reading package lists... Done
+ cd /root/
+ sudo rm -rf ./librealsense_build
+ mkdir librealsense_build
+ cd librealsense_build
++ sudo swapon --show
++ wc -l
+ '[' 0 -eq 0 ']'
+ echo 'No swapon - setting up 1Gb swap file'
No swapon - setting up 1Gb swap file
+ sudo fallocate -l 2G /swapfile
fallocate: fallocate failed: No space left on device

```

Anyone faced this issue, Note I am using a ubuntu 20.04 docker container to do this

MartyG-RealSense, 2023-05-06:
Hi @darshit-desai  The libuvc_installation.sh installation script usually does not work with Ubuntu 20.04 because the script was designed for earlier Ubuntu versions.  You can achieve a similar installation by building for 20.04 from source code with CMake using the **RSUSB backend** installation method described at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692

RSUSB backend is the successor of the older libuvc backend.  Both methods bypass the kernel, making the librealsense build non-dependent on Linux versions or kernel versions and not requiring the kernel to be patched.

darshit-desai, 2023-05-06:
Hi @MartyG-RealSense it was a memory issue since I was trying to install this on a system which had almost 10 GB of storage and the build itself is of 10 GB 

I understand that this method is got older distributions of Linux but at the end of it the build was completed successfully and upon install and building the Ros wrapper I was able to launch the node successfully. I am yet to test it with the camera connected with the computer.

The whole installation is occuring on a docker container.

The only way for me to test whether the build of the sdk worked was with launching of a launch file from Ros wrapper. Is there any other way to check it before I get into the hardware connection of the camera with the docker container?

I am attaching a photo of the Ros launch of the Realsense wrapper[image]

MartyG-RealSense, 2023-05-06:
You should be able to test the launch of the ROS1 wrapper without a camera attached if a **.bag** file of pre-recorded data is used instead of a live camera.  This can be done with the launch file **rs_from_file.launch** instead of rs_camera.launch, providing a path to the bag file.  For example:

`roslaunch realsense2_camera rs_from_file.launch rosbag_filename:="/home/user/project/test1.bag"`

darshit-desai, 2023-05-06:
Hi @MartyG-RealSense the Ros bag along with launch file did run. Since this was running in a UI less Docker nothing like this or gazebo opened up but I don't see any errors popping up. Does this validate SDK+wrapper installation?

`rosnoetic:catkin_ws$ roslaunch realsense2_camera rs_from_file.launch rosbag_filename:="/root/catkin_ws/outdoors.bag"

... logging to /root/.ros/log/b427a14a-ec2a-11ed-b761-0242ac110002/roslaunch-6ff06d51c199-776.log

Checking log directory for disk usage. This may take a while.

Press Ctrl-C to interrupt

Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://6ff06d51c199:35591/

SUMMARY

========

PARAMETERS

 * /camera/realsense2_camera/accel_fps: 0

 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame

 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...

 * /camera/realsense2_camera/align_depth: False

 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...

 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...

 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...

 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...

 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...

 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...

 * /camera/realsense2_camera/allow_no_texture_points: False

 * /camera/realsense2_camera/base_frame_id: camera_link

 * /camera/realsense2_camera/calib_odom_file: 

 * /camera/realsense2_camera/clip_distance: -1.0

 * /camera/realsense2_camera/color_fps: 0

 * /camera/realsense2_camera/color_frame_id: camera_color_frame

 * /camera/realsense2_camera/color_height: 0

 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...

 * /camera/realsense2_camera/color_width: 0

 * /camera/realsense2_camera/confidence_fps: 30

 * /camera/realsense2_camera/confidence_height: 480

 * /camera/realsense2_camera/confidence_width: 640

 * /camera/realsense2_camera/depth_fps: 0

 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame

 * /camera/realsense2_camera/depth_height: 0

[... 205 lines cut ...]

[ INFO] [1683390369.943303428]: Done Setting Dynamic reconfig parameters.

[ INFO] [1683390369.945492970]: depth stream is enabled - width: 1280, height: 720, fps: 30, Format: Z16

[ INFO] [1683390369.946168053]: infra1 stream is enabled - width: 1280, height: 720, fps: 30, Format: Y8

[ INFO] [1683390369.946724012]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8

[ INFO] [1683390369.946741303]: setupPublishers...

[ INFO] [1683390369.948470053]: Expected frequency for depth = 30.00000

[ INFO] [1683390369.976322053]: Expected frequency for infra1 = 30.00000

[ INFO] [1683390369.982786887]: Expected frequency for color = 30.00000

[ INFO] [1683390369.991605970]: setupStreams...

[ INFO] [1683390369.991755178]: SELECTED BASE:Depth, 0

[ INFO] [1683390369.992071012]: RealSense Node Is Up!

[ WARN] [1683390371.168919887]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
`

MartyG-RealSense, 2023-05-06:
The launch reached the **RealSense Node Is Up** launch completion message without errors in the log before or after and the topics are published, so it looks as though the SDK and wrapper are behaving correctly.  For further information you could perform a rostopic echo of the **/diagnostics** topic that the wrapper automatically publishes.

darshit-desai, 2023-05-10:
Hi @MartyG-RealSense I was able to load the image on the VOXL 1 Modal AI computer and when I tried to run it I ran into the following error, Do you have any ideas on what could be the issue?

```
rosnoetic:catkin_ws$ roslaunch realsense2_camera rs_from_file.launch rosbag_filename:="/root/stairs.bag"
... logging to /root/.ros/log/3ff378a0-ef6b-11ed-b0b8-18473db1bf59/roslaunch-apq8096-381.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.78.253:59110/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 0
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 0
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 0
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 0
 * /camera/realsense2_camera/confidence_fps: 30
 * /camera/realsense2_camera/confidence_height: 480
 * /camera/realsense2_camera/confidence_width: 640
 * /camera/realsense2_camera/depth_fps: 0
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 0
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 0
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 0
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 0
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 0
 * /camera/realsense2_camera/gyro_fps: 0
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
[... 38 lines cut ...]
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [389]
ROS_MASTER_URI=http://192.168.78.253:11311

setting /run_id to 3ff378a0-ef6b-11ed-b0b8-18473db1bf59
process[rosout-1]: started with pid [401]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [404]
process[camera/realsense2_camera-3]: started with pid [405]
[ INFO] [1683747945.899715516]: Initializing nodelet with 4 worker threads.
[ INFO] [1683747947.953942910]: RealSense ROS v2.3.2
[ INFO] [1683747947.954126245]: Built with LibRealSense v2.53.1
[ INFO] [1683747947.954221089]: Running with LibRealSense v2.53.1
[ INFO] [1683747948.022359264]: publish topics from rosbag file: /root/stairs.bag
[ INFO] [1683747950.186044521]: getParameters...
[ INFO] [1683747951.066723771]: setupDevice...
[ INFO] [1683747951.066938304]: JSON file is not provided
[ INFO] [1683747951.067061743]: ROS Node Namespace: camera
[ INFO] [1683747951.067182889]: Device Name: Intel RealSense 435
[ INFO] [1683747951.067289869]: Device Serial No: 722312070422
[ERROR] [1683747951.104213536]: An exception has been thrown: info Physical Port not supported by the device!
[ERROR] [1683747951.104872864]: Exception: info Physical Port not supported by the device!
```

darshit-desai, 2023-05-10:
Hi @MartyG-RealSense in the same hardware configuration when I run outdoors.bag it works and realsense nodes load successfully. We echoed the image_raw topic in the under water file it works. I will update after I connect the hardware

The diagnostics you requested is below in the screenshot:
[image]
```

rosnoetic:~$ roslaunch realsense2_camera rs_from_file.launch rosbag_filename:="/root/outdoors.bag"
... logging to /root/.ros/log/f21808fc-ef6b-11ed-8053-18473db1bf59/roslaunch-apq8096-728.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.78.253:55548/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 0
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 0
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 0
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 0
 * /camera/realsense2_camera/confidence_fps: 30
 * /camera/realsense2_camera/confidence_height: 480
 * /camera/realsense2_camera/confidence_width: 640
 * /camera/realsense2_camera/depth_fps: 0
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 0
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 0
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 0
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 0
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 0
[... 67 lines cut ...]
[ INFO] [1683748248.047953758]: Align Depth: Off
[ INFO] [1683748248.048055738]: Sync Mode: Off
[ INFO] [1683748248.048290740]: Device Sensors: 
[ INFO] [1683748248.048616888]: Stereo Module was found.
[ INFO] [1683748248.048874650]: RGB Camera was found.
[ INFO] [1683748248.049052933]: (Infrared, 2) sensor isn't supported by current device! -- Skipping...
[ INFO] [1683748248.049177622]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1683748248.049651011]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1683748248.056980285]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1683748248.057961490]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1683748248.061978865]: num_filters: 0
[ INFO] [1683748248.069136107]: Setting Dynamic reconfig parameters.
[ INFO] [1683748248.110650329]: Done Setting Dynamic reconfig parameters.
[ INFO] [1683748248.115787503]: depth stream is enabled - width: 1280, height: 720, fps: 30, Format: Z16
[ INFO] [1683748248.128738227]: infra1 stream is enabled - width: 1280, height: 720, fps: 30, Format: Y8
[ INFO] [1683748248.130728710]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1683748248.131462362]: setupPublishers...
[ INFO] [1683748248.164765792]: Expected frequency for depth = 30.00000
[ INFO] [1683748248.662894060]: Expected frequency for infra1 = 30.00000
[ INFO] [1683748249.038819836]: Expected frequency for color = 30.00000
[ INFO] [1683748249.401947806]: setupStreams...
[ INFO] [1683748249.402966615]: SELECTED BASE:Depth, 0
[ INFO] [1683748249.410886936]: RealSense Node Is Up!
[ WARN] [1683748249.412812524]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
```

darshit-desai, 2023-05-10:
Hi @MartyG-RealSense We were able to connect the stereo camera d435i with the docker image onboard the VOXL1 board. We also streamed the image topics back to rviz in our computer. 

Here's the screenshot of that:

[image]

The L515 camera is shown when I enumerate it, but there is some issue with it:

[image]

We are still trying to connect L515 Lidar camera but we are facing the below issue,

```
rosnoetic:catkin_ws$ roslaunch realsense2_camera rs_camera.launch 
... logging to /root/.ros/log/528e5e72-ef83-11ed-8f97-18473db1bf59/roslaunch-apq8096-851.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.78.253:53048/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
[... 162 lines cut ...]
 10/05 18:38:39,949 ERROR [546568122736] (sensor.cpp:664) acquire_power failed: failed to set power state
 10/05 18:38:39,954 WARNING [546568122736] (rs.cpp:310) null pointer passed for argument "device"
[ WARN] [1683758319.955711636]: Device 1/1 failed with exception: failed to set power state
[ERROR] [1683758319.956181431]: The requested device with  is NOT found. Will Try again.
^C[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```
After launching the file when I again do rs-enumerate-devices it doesn't show the device, even though it shows earlier:

```
rosnoetic:catkin_ws$ rs-enumerate-devices 
 10/05 18:44:17,791 ERROR [547687186864] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_NO_DEVICE
 10/05 18:44:17,791 ERROR [547738506304] (sensor.cpp:664) acquire_power failed: failed to set power state
Could not create device - failed to set power state . Check SDK logs for details
```

Thank you for your continued help 

Best,
Darshit

MartyG-RealSense, 2023-05-11:
Hi @darshit-desai  I note that the **Device USB type** in your launch log is **2.1**.  This indicates that the camera's USB connection is being detected as a slow USB 2.1 instead of the faster USB 3.

If an L515 is plugged into a USB 3 port but is detected as 2.1, this can be because the micro-sized end of the USB cable that is plugged into the base of the camera may need to be unplugged, turned around the other way and re-inserted.  

A characteristic of the L515 camera model is that it typically works correctly when the connector is plugged into one particular orientation of the two available (USB-C cables are two-way insertion on the micro-sized end) and may be mis-detected as USB 2.1 if the connector is the other way around.

darshit-desai, 2023-05-12:
Hi @MartyG-RealSense I did try to reverse the polarity of the cables but it didn't work. In other updates, we are not able to detect the D435i on board our drone's VOXL computer. A day ago everything worked when we tried the `lsusb` command for checking the USB devices but the same device is not detected anymore. We cross-checked it with the laptop by connecting the camera with the laptop and it detected it. We also connected L515 to cross-check whether there is something wrong with the drone's computer, L515 was detected but failed to launch realsense node as earlier. We are really not sure how to debug this issue

darshit-desai, 2023-05-12:
Also to add on that we had a spare D435i which we borrowed from one of our friends. The drone's computer did detect that camera too. So we are not sure is it the drone or the camera which is at fault here. Because the camera which doesn't connect to the drone does connect with the Laptop's USB ports

darshit-desai, 2023-05-13:
Hi @MartyG-RealSense. I wanted to provide more information on this:

When I type in dmesg in the voxl terminal I get this message:

```
[ 1592.790220] msm-dwc3 7600000.hsusb: DWC3 exited from low power mode
[ 1592.790305] type=1400 audit(1683949837.371:216): avc: denied { write } for pid=5394 comm="lsusb" name="001" dev="devtmpfs" ino=3279 scontext=root:sysadm_r:sysadm_t:s0 tcontext=system_u:object_r:usb_device_t:s0 tclass=chr_file permissive=1
[ 1593.941957] msm-dwc3 7600000.hsusb: DWC3 in low power mode
```

Waht does this mean? Ignore 1592.790305 message but the two other messages came as soon as I connected the USB cable

MartyG-RealSense, 2023-05-13:
The DWC3 messages suggest that the USB controller on the computer is going into low power mode, perhaps causing the camera to disconnect.  

Are you connecting the camera to a micro-sized **USB OTG** port on the computers instead of a full-size USB port?  OTG ports can cause insufficient power to be provided to a RealSene camera.  Also, if you are using your own choice of USB cable instead of the official 1 meter cable then quality issues in that cable may also cause connection problems.  For example, inexpensive cables designed only for charging or sync should not be used, and a high quality cable designed for data transfer used instead.

darshit-desai, 2023-05-13:
Hi @MartyG-RealSense Our setup consists of the following connections

Voxl 1 board-->USB 2.1 Expansion board-->Male USB A to Micro USB C (Official cable) --> Realsense Camera D435i

This worked a few times before, but now it doesn't detect the camera at all upon doing `lsusb`. Apart from this in the same setup above we tested another D435i it detected it. It also detects L515 lidar camera. 

We are not sure what went wrong and don't want to sacrifice any more D435i without getting into the root cause of this issue.

Regarding the power we are thinking of upgrading the expansion board power to 5v-2A. (USB 2.1 with higher throughput).

Other than this we have no clue why the camera stopped working. The same camera which isn't detected on the VOXL board gets detected on our Laptops when connected to USB3.0 ports.

Best,
Darshit

P.S. : Not important but still mentioning it
Specs of VOXL1 board:https://docs.modalai.com/voxl-flight-datasheet-connectors/
Specs of VOXL 1 USB 2.1 Expansion board: https://docs.modalai.com/usb-expander-and-debug-user-guide/

MartyG-RealSense, 2023-05-14:
Have you tried a cable swap, using the USB cable of the D435i that is detected on the D435i that is not detected to see whether the problem of non-detection on USB 2.1 on the VOX1 is due to a fault in a specific USB cable?

darshit-desai, 2023-05-14:
Hi @MartyG-RealSense we have been using the same cable for all 3 devices. 2 devices can connect.(D435i, L515 and another D435i) 1 used to get connected and now it's not showing up

MartyG-RealSense, 2023-05-14:
Please check the affected D435i on the laptop's USB connection to see which firmware driver version it currently has installed.  You can find this information in the RealSense Viewer tool by clicking on the 'Info' button at the top of the Viewer's options side-panel.

You could also try resetting the camera in the Viewer to its factory-new default calibration settings on the USB 3 connection using the instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487

darshit-desai, 2023-05-14:
Hi @MartyG-RealSense we already tried installing and reinstalling another version firmware, we were before using the latest version we uninstalled it since this issue happened with an older version. But since the older version also didn't work we switched back to the latest firmware v5.14.0.0. Here's a screenshot of the info screen

[image]

We also tried Hardware reset option earlier but it didn't work.

We also tried doing the factory reset now for calibration as you said but it seems like it's already at the same settings as the write table icon remains inactive
[image]

Same was tested on the drone computer but still it's not able to connect with it.

Best,
Darshit

MartyG-RealSense, 2023-05-14:
You have to enable the depth stream before opening the Camera Calibration window in order for the Write Table option to be selectable.

darshit-desai, 2023-05-14:
Hi @MartyG-RealSense We tried that right now. We were able to write the table but as I told earlier none of the values changed. We tried it with the drone and still we are unable to connect it

MartyG-RealSense, 2023-05-14:
The following advice usually only works on the L515 model but give it a try anyway please.  Unplug the micro sized USB connector from the side of the camera, turn the connector around the opposite direction and re-insert it.  Then unplug the full size connector from the VOX1 computer, wait a couple of seconds and re-insert it into the computer.  Is the camera then detected on USB 2.1?

MartyG-RealSense, 2023-05-20:
Hi @darshit-desai  Do you require further assistance with this case, please?  Thanks!

darshit-desai, 2023-05-20:
Hi @MartyG-RealSense I was able to connect to the drone by changing some of the hardware. I jacked up the power supplied from the USB 2.1 board to 2 Amperes and it was able to detect both the cameras. Thank you. I am willing to write a Wiki on how to connect the realsense cameras to the VOXL 1 computers if you like.

MartyG-RealSense, 2023-05-20:
You are very welcome.  

User-contributed guides are always a great resource that benefits others with similar problems, so please do share a guide if you create one!

closed 2023-06-08 by MartyG-RealSense