realsenseai/librealsense #11775: Mutli Object Volume Estimation Failed.

BVishal-Geek, 2023-05-06:
* When I tried working with realsense Box Dimensioner Multicam. It was working pretty good with single object volume estimation, but when two objects comes into the scene it is not working properly.

MartyG-RealSense, 2023-05-07:
Hi @BVishal-Geek  The box_dimensioner_multicam.py example program is designed for use with a single object on the chessboard.

BVishal-Geek, 2023-05-07:
Hi, I am actually working to estimate volume for 2 or more objects. Any leads with which you can help me achieve it.

MartyG-RealSense, 2023-05-07:
The link below has a RealSense pyrealsense2 program for measuring between points instead of measuring the distance from camera to point, though it would likely require the manual definition of the points rather than doing it automatically like the box dimensioner example does.

https://github.com/soarwing52/RealsensePython/blob/master/separate%20functions/measure_new.py

closed 2023-05-21 by MartyG-RealSense