realsenseai/librealsense #11778: I want to add depth streaming to the C# form. Is there an example?

gjdeorl, 2023-05-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D455} | 
| Firmware Version                   | {2.53.1} | 
| Operating System & Version |   {Win 10} | 
| Platform                                 | PC |
| SDK Version                            |  {2.5 }                          | 
| Language                            |  {C#}                          | 

I want to stream the depth using C#, but I'm asking because there doesn't seem to be an example using a form. If you know, please let me know. Thank you.

MartyG-RealSense, 2023-05-08:
Hi @gjdeorl  Does the C# script at https://github.com/IntelRealSense/librealsense/issues/3413#issuecomment-474451376 help to resolve your problem, please?  Thanks!

gjdeorl, 2023-05-09:
var frames = pipe.WaitForFrames()  <==== An error occurs here. Why does this error occur? The error content is as follows.

System.InvalidOperationException: 'wait_for_frames cannot be called before start()'
ExternalException: rs2_pipeline_wait_for_frames(pipe:000001ED5ECE9E10)

MartyG-RealSense, 2023-05-09:
I believe that https://github.com/IntelRealSense/librealsense/issues/3413#issuecomment-474451376 is not a complete script but is a section that is meant to be placed within a larger C# script.  Once the pipeline has been started (to resolve the problem that wait_for_frames cannot be called before start) then call the function name **RunInThread()** to jump to the block of code containing the depth display instructions.

MartyG-RealSense, 2023-05-15:
Hi @gjdeorl  Do you require further assistance with this case, please?  Thanks!

gjdeorl, 2023-05-16:
Thank you for your reply.
One problem arose.
using (var depth = frames.FirstOrDefault(x => x.Profile.Stream == Intel.RealSense.Stream.Depth) as DepthFrame) In this part, the depth value comes out as a null value and 'System.NullReferenceException'(Intel.RealSense .dll) throws an exception.
How should I do this part?
The color part also comes out with a null value in the same way.

MartyG-RealSense, 2023-05-16:
The RealSense SDK's **C# Cookbook** page of code snippets has a simpler way of defining FirstOr Default.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/Documentation/cookbook.md#wait-for-coherent-set-of-frames

gjdeorl, 2023-05-22:
Thank you for your reply. I will close this issue.

closed 2023-05-22 by gjdeorl