realsenseai/librealsense #11779: comparing the HIGH density param-set to the json online

avizipi, 2023-05-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5_12_8_200 | 
| SDK Version                            | 2.39.0 | 
| Language                            | C++  | 

### Issue Description

after changing the preset params to HIGH-DENSITY and reading the json file It seems different than the one in the [wiki page](https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets#short-range-preset).

I used this function to change the values - sensor.set_option(RS2_OPTION_VISUAL_PRESET, _val);
while _val was casted to "High Density" using this function rs2_rs400_visual_preset_to_string((rs2_rs400_visual_preset)_val)

After that I read the json params again using this code:
```
    rs2::device device =  m_profile.get_device();
    auto advanced = device.as<rs400::advanced_mode>();
    auto new_parmasAsJson = advanced.serialize_json();
    std::cout << new_parmasAsJson <<std::endl;
```

therefore these lines was printed:
```
{
"aux-param-autoexposure-setpoint": "1536",
"aux-param-colorcorrection1": "0.298828",
"aux-param-colorcorrection10": "-0",
"aux-param-colorcorrection11": "-0",
"aux-param-colorcorrection12": "-0",
"aux-param-colorcorrection2": "0.293945",
"aux-param-colorcorrection3": "0.293945",
"aux-param-colorcorrection4": "0.114258",
"aux-param-colorcorrection5": "-0",
"aux-param-colorcorrection6": "-0",
"aux-param-colorcorrection7": "-0",
"aux-param-colorcorrection8": "-0",
"aux-param-colorcorrection9": "-0",
"aux-param-depthclampmax": "65536",
"aux-param-depthclampmin": "0",
"aux-param-disparityshift": "0",
"controls-autoexposure-auto": "True",
"controls-autoexposure-manual": "8500",
"controls-color-autoexposure-auto": "True",
"controls-color-autoexposure-manual": "166",
"controls-color-backlight-compensation": "0",
"controls-color-brightness": "0",
"controls-color-contrast": "50",
"controls-color-gain": "64",
"controls-color-gamma": "300",
"controls-color-hue": "0",
"controls-color-power-line-frequency": "3",
"controls-color-saturation": "64",
"controls-color-sharpness": "50",
"controls-color-white-balance-auto": "True",
"controls-color-white-balance-manual": "4600",
"controls-depth-gain": "16",
"controls-laserpower": "150",
"controls-laserstate": "on",
"ignoreSAD": "0",
"param-amplitude-factor": "0",
"param-autoexposure-setpoint": "1536",
"param-censusenablereg-udiameter": "9",
"param-censusenablereg-vdiameter": "3",
"param-censususize": "9",
"param-censusvsize": "3",
"param-depthclampmax": "65536",
"param-depthclampmin": "0",
"param-depthunits": "1000",
"param-disableraucolor": "0",
"param-disablesadcolor": "0",
"param-disablesadnormalize": "0",
"param-disablesloleftcolor": "1",
"param-disableslorightcolor": "0",
"param-disparitymode": "0",
"param-disparityshift": "0",
"param-lambdaad": "850",
"param-lambdacensus": "19",
"param-leftrightthreshold": "32",
"param-maxscorethreshb": "1949",
"param-medianthreshold": "276",
"param-minscorethresha": "68",
"param-neighborthresh": "4",
"param-raumine": "7",
"param-rauminn": "1",
"param-rauminnssum": "6",
"param-raumins": "3",
"param-rauminw": "6",
"param-rauminwesum": "14",
"param-regioncolorthresholdb": "0.0107632",
"param-regioncolorthresholdg": "0.645793",
"param-regioncolorthresholdr": "0.0166341",
"param-regionshrinku": "4",
"param-regionshrinkv": "1",
"param-robbinsmonrodecrement": "1",
"param-robbinsmonroincrement": "4",
"param-rsmdiffthreshold": "5.0625",
"param-rsmrauslodiffthreshold": "1",
"param-rsmremovethreshold": "0.446429",
"param-scanlineedgetaub": "4",
"param-scanlineedgetaug": "542",
"param-scanlineedgetaur": "982",
"param-scanlinep1": "479",
"param-scanlinep1onediscon": "4",
"param-scanlinep1twodiscon": "25",
"param-scanlinep2": "342",
"param-scanlinep2onediscon": "69",
"param-scanlinep2twodiscon": "225",
"param-secondpeakdelta": "6",
"param-texturecountthresh": "1",
"param-texturedifferencethresh": "0",
"param-usersm": "1",
"param-zunits": "1000"
}
```

I do see that the values was changed because I also printed the json before the process.
comparing this json to the json in [the wiki](https://raw.githubusercontent.com/wiki/IntelRealSense/librealsense/d400_presets/HighDensityPreset.json) reveled that the values are different. I used text-compare.com to compare then and here is a small part of it.
[image]

you can see that the max/median/min score THR are different and that the lambda ad/census are also different.
Can you maybe check what should we use? the default hardcoded version or the wiki one?

MartyG-RealSense, 2023-05-08:
Hi @avizipi  If you are using one of the built-in presets then I recommend using the **RS2_OPTION_VISUAL_PRESET** instruction to apply it and not use the json file version of the preset on the Visual Presets web-page.  This should ensure that your script has the same configuration changes as when choosing that preset in the RealSense Viewer tool.

The scripting for loading a json from a file instead of applying a built-in preset is different and more complicated, so using a built-in preset in your script if possible is much easier.

avizipi, 2023-05-08:
Thanks for your replay.
Did you also notice the different between using the RS2_OPTION_VISUAL_PRESET preset values and the wiki values?

MartyG-RealSense, 2023-05-08:
I tested in the Viewer with both the built-in High Density preset and the json file version of it and there was no difference in the depth image for each.

avizipi, 2023-05-08:
Also in my indoor scene I didn't see any visible changes in the depth image. 
But what about the values them self, can you confirm that the wiki and hardcoded params are different? and if so can you elaborate on these differences?

MartyG-RealSense, 2023-05-08:
I checked the Advanced Mode values of the High Density built-in preset in my Viewer with a D435i and compared them to the json file values.  Some of the values for my built-in preset were different to yours.  I do not think that it is significant though.  Advanced Mode parameters are managed by a machine learning algorithm because of the complex ways in which they interact with each other, which is the reason why Advanced Mode options do not have documentation (though RealSense users are free to experiment with the options to see how changes affect the image).  

So if built-in and json files produce the same visual results, I would recommend not being concerned by it.

closed 2023-05-08 by avizipi