realsenseai/librealsense #11781: How can I export ply files

0DWADE0, 2023-05-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | (Open RealSense Viewer --> 2.53.1) | 
| Operating System & Version |   {Win (10)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.53.1.4623}                          | 
| Language                               |  {C++ }                          | 
| Segment			         |  {others }                          | 

### Issue Description
<I am developing L515, how to export ply format files, and what should I do if I cannot export them directly?>

MartyG-RealSense, 2023-05-09:
Hi @0DWADE0  There are not many references available about exporting a ply file with C++.  I recommend using the **save_to_ply** instruction, which has a range of options for configuring export.  A RealSense team member shares a C++ code snippet for save_to_ply at https://github.com/IntelRealSense/librealsense/issues/6677#issuecomment-648079238

A list of the supported configuration commands can be found at https://github.com/IntelRealSense/librealsense/pull/4906 and also at the link below.

https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1save__to__ply.html#a9971c1fbc2d761e7a871d55c5174770c

MartyG-RealSense, 2023-05-16:
Hi @0DWADE0  Do you require further assistance with this case, please?  Thanks!

0DWADE0, 2023-05-16:
@MartyG-RealSense Thanks for your advice, I got the example I wanted

MartyG-RealSense, 2023-05-16:
You are very welcome.  As you achieved a solution, I will close this case.  Thanks again!

closed 2023-05-16 by MartyG-RealSense