realsenseai/librealsense #11784: Does the D405 require alignment between RGB and depth streams?

xEnVrE, 2023-05-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D405 | 
| Firmware Version                   |05.12.14.100 | 
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  6.2.7 | 
| Platform                                 | PC |
| SDK Version                            | 2.53.1| 
| Language                            | C++                | 
| Segment			| Others  | 

### Issue Description

I made some experiments using the `rs-align` example in C++. I have modified it so that I can run it using the native resolution of my application, i.e. 640x480, and I can disable the alignment completely if needed.

It seems that using a D405 camera the software-based alignment is not working as expected.

I am attaching a video which shows the behavior.

https://user-images.githubusercontent.com/6014499/237053011-6b6dcb73-2b99-4ee2-91fd-41ed25ebb0d6.mp4

As you can see, when I enable the alignment the depth and RGB streams get not aligned and the depth is kinda zoomed.
Is such behavior of the `rs2::align` software block normal? Do I need to use alignment at all with the D405 camera?

Thank you

cc @MartyG-RealSense

MartyG-RealSense, 2023-05-09:
Hi @xEnVrE  The D405 camera model does not have a separate RGB sensor component and RGB is instead provided by the depth sensor.  This means that RGB is already in alignment with the left IR sensor that depth originates from.

It is fine to use D405 with depth-color alignment scripts such as **rs-align** for C++ and **align_depth2color.py** for Python to create an aligned image,

D405 uses a different **depth unit scale** scaling factor to other 400 Series models.  Most models have a default depth scale of 0.001, whilst the default scale of D405 is 0.01.

xEnVrE, 2023-05-09:
Hi @MartyG-RealSense thank you for your the prompt answer.

> This means that RGB is already in alignment with the left IR sensor that depth originates from

Thanks, then we can avoid using the alignment software-side.

> D405 uses a different depth unit scale scaling factor to other 400 Series models. Most models have a default depth scale of 0.001, whilst the default scale of D405 is 0.01.

We are aware of it however we did not change any scale within the `rs-align` executable.

> I note that your firmware version at the top of this case is listed as 5.12.14.100. The firmware version that must be used with SDK 2.53.1 is 5.14.0.0.

> Kernel version 6.2.7 is not supported by the RealSense SDK. The newest supported kernel at the time of writing this is 5.15. The SDK can be used with an unsupported kernel but there may be unpredictable consequnences in regards to stability. If you are unable to change your kernel version then you can build the SDK from source code with the RSUSB backend installation method so that the kernel is bypassed and the SDK does not need to be kernel-patched.

To avoid any confusion we moved to another setup with Kernel `5.11.0-46-generic` and we compiled using `RSUSB` in any case to avoid any issue. I wanted to also clarify that we are compiling with `CUDA` enabled.

> Does the alignment occur correctly without your resolution modification, please?

We repeated our tests.

The original `rs-align` executable is running at `1280x720` and seems fine in terms of calibration with the `D405`.

[image]

However, I think - please correct me if I am wrong - that the software-based alignment should work as well with a different resolution. Then we changed these lines

https://github.com/IntelRealSense/librealsense/blob/b874e42685aed1269bc57a2fe5bf14946deb6ede/examples/align/rs-align.cpp#L40-L52

to

```code

    const std::size_t width = 640;
    const std::size_t height = 480;

    // Create and initialize GUI related objects
    window app(width, height, "RealSense Align Example"); // Simple window handling
    ImGui_ImplGlfw_Init(app, false);      // ImGui library intializition
    rs2::colorizer c;                     // Helper to colorize depth images
    texture depth_image, color_image;     // Helpers for renderig images

    // Create a pipeline to easily configure and start the camera
    rs2::pipeline pipe;
    rs2::config cfg;
    if (!serial.empty())
        cfg.enable_device(serial);
    cfg.enable_stream(RS2_STREAM_DEPTH, width, height);
    cfg.enable_stream(RS2_STREAM_COLOR, width, height);
    pipe.start(cfg);
```

in order to test it with the resolution we are interested in, i.e., 640 x 480.

We got this result:

[image]

where the alignment is - according to us - **not correct**.

Indeed, by modifying the `rs-align` executable such that it **does not perform the alignment**, we got the following:

[image]

Thank you for your help!

MartyG-RealSense, 2023-05-09:
You are very welcome!  Can you confirm whether modifying rs-align to not perform alignment resolves your problem, please?

xEnVrE, 2023-05-09:
>  Can you confirm whether modifying rs-align to not perform alignment resolves your problem

It seems so @MartyG-RealSense. 

However, I was just wondering why using the `rs2::align`, as done inside the `rs-align` executable, breaks the alignment with this resolution. Can you confirm whether this is the expected behavior? 

Thank you

MartyG-RealSense, 2023-05-09:
I believe that you are the first person to report an issue with the C++ based rs-align example when using D405, as D405 users have always used the Python-based equivalent example **align_depth2color.py** example up until now.  rs-align and align_depth2color.py both perform alignment using the **align_to** instruction.

MartyG-RealSense, 2023-05-09:
Which cfg stream configuration instructions are you using to define 640x480 resolution in your rs-align custom version?  Is it something like this:

```
cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30)
cfg.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_RGB8, 30)
```

MartyG-RealSense, 2023-05-10:
Is it still misaligned if you change the **window size** back to 1280x270 instead of using 640x480 but still keep the streams at 640x480

` window app(1280, 720, "RealSense Align Example"); // Simple window handling `

MartyG-RealSense, 2023-05-15:
Hi @xEnVrE  Do you require further assistance with this case, please?  Thanks!

xEnVrE, 2023-05-15:
Hi @MartyG-RealSense, we kind of solved by disabling the alignment.

However the problem still persists for those who want to use the alignment for whatever reason.

I will try to perform the last test you asked today, sorry for the delay.

MartyG-RealSense, 2023-05-15:
That's no problem at all.  I look forward to your next report.  Thanks very much for the update!

xEnVrE, 2023-05-15:
> Is it still misaligned if you change the window size back to 1280x270 instead of using 640x480 but still keep the streams at 640x480

We made this test, it seems that the problem persists:

### With software-based alignment

[image]

### Without software-based alignment

[image]

MartyG-RealSense, 2023-05-15:
The above images do not look as though they are misaligned.  The black silhouette around the object is a common effect that occurs when depth and color are mapped together. It is called **occlusion**.  The occlusion in this image is actually good quality, as the silhouette surrounds the object evenly on all sides instead of the shadow being off to one side of the object in particular.

xEnVrE, 2023-05-15:
Dear @MartyG-RealSense thank you for your comment.

Unfortunately, I cannot agree with you on the fact that both images are not misaligned. 

To me, if you consider the second image, it is clear that the alignment is very good. For example, **the boundaries of the cup match when comparing RGB and depth**:

[image]

Instead, if you take the first image - where the `librealsense` SDK is processing the depth to align it with the RGB image - it is clear to me that the **alignment is quite off. The red arrow point at where the cup ends in the RGB image, while the green arrow points where the cup ends in the depth image.**

[image]

Please, let me know what do you think! Thank you

MartyG-RealSense, 2023-05-15:
Thanks very much for the clarification.  It looks though as though your workaround for using rs-align with D405 without alignment is going to be the best approach for using a custom resolution.

xEnVrE, 2023-05-21:
I would say no, we made the tests you asked for and we were waiting for a fix from your side. 

As you stated, one solution is to disable alignment.

However I think it should be stated somewhere that the software-based alignment is broken for some resolutions with the D405 so that other people are aware of it.

Thanks

MartyG-RealSense, 2023-05-21:
The alignment on rs-align became incorrect for you when a custom resolution of 640x480 was set and worked normally when using its default 1280x720 resolution.  So the example program is working as intended when the program is unmodified. 

I understand your concern though in regard to your wish to be able to customize the resolution used with alignment.  It is certainly an issue that can be returned to at a future date if others experience a similar issue with their D405.

xEnVrE, 2023-05-21:
> So the example program is working as intended when the program is unmodified.

For sure. However the problem is not with the specific executable per se, rather on the alignment procedure that your SDK offers. The fact that `rs-align` has a default resolution of 1280x720 is, at least in my opinion, just a coincidence. Afterall, `rs-align` is just an example, as you stated.

This issue is more about the alignment of the D405 in general than the `rs-align` executable though. Sorry for being picky here.

MartyG-RealSense, 2023-05-21:
It's no trouble at all.  I have highlighted your issue to my Intel RealSense colleagues to seek their advice regarding whether a custom resolution can be set with D405.

MartyG-RealSense, 2023-05-27:
My Intel RealSense colleagues responded that this issue with D405 mis-alignment is known to them and an internal report to look at the problem has already been created.  

In the meantime they suggested using 640x360 or 848x480 resolution instead of 640x480, though this may not be an option for you as the rest of your application relies on 640x480.

xEnVrE, 2023-05-27:
Hi @MartyG-RealSense,

Thank you for the update!

We could actually use those other resolutions if the alignment works with them. Moving from 640x480 to 640x360 should not be a big deal for us.

Thanks again for the update. Looking forward towards a final resolution for the issue.

MartyG-RealSense, 2023-06-02:
Hi @xEnVrE  Were you able to make use of 640x360 resolution to provide correct alignment, please?  Thanks!

xEnVrE, 2023-06-05:
Hi @MartyG-RealSense, unfortunately I do not have access to this setup these days. If possible, I will update the issue with the tests once ready. 

Thank you

MartyG-RealSense, 2023-06-05:
Thanks very much @xEnVrE for the update.

MartyG-RealSense, 2023-06-15:
Hi @Daa98  The RealSense Viewer tool does not support alignment in its 2D mode (shown above).  It only maps two different streams together in its **3D pointcloud** mode.  

On the D405 camera model, RGB color is provided by the left infrared sensor, so the left infrared and RGB images have the same viewpoint.  The two images are not having alignment performed on them though and are aligned by default simply because they share a sensor.

MartyG-RealSense, 2023-06-20:
Alignment between RGB and infrared is not usually an issue with RealSense cameras, as alignment between those stream types is not supported.  One of the two stream types has to be Depth (depth to color, or depth to infrared).  

The only available solutions for color to infrared alignment are to follow the complicated advice at https://github.com/IntelRealSense/librealsense/issues/1556#issuecomment-389061477 or to align all three streams including depth using the method at https://github.com/IntelRealSense/librealsense/issues/5093

xEnVrE, 2023-06-27:
At the moment we are happy with the solution consisting of disabling the software alignment. Any future software update will be welcome of course. Thanks for the support thus far!

MartyG-RealSense, 2023-06-27:
You are very welcome!  As you do not require further assistance right now, I will close this issue.  Please feel free to re-open it at a future date or create a new issue.  Thanks again!

closed 2023-06-27 by MartyG-RealSense