realsenseai/librealsense #11785: Client time out -3 when using RS server for wireless video streaming transmission

H-chun, 2023-05-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D457 } | 
| Firmware Version                   | 2.53.1 | 
| Operating System & Version |  jetson  | 
| Kernel Version (Linux Only)    |  5.10.104-tegra                                         | 
| Platform                                 | NVIDIA Jetson  |
| SDK Version                            | 2.53.1                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  Robot                         | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Hello! When I installed the [https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md](installation) description for SDK compilation, an RS server was generated, and I executed the RS server on an ORIN. Open realsense viewer on another orin. Because orin itself has a fixed IP, I directly entered the IP, but encountered an error of error: client time out -3. Can you help me analyze it? What angles do I need to analyze this error from in this situation? Thank you very much
[image]

MartyG-RealSense, 2023-05-09:
Hi @H-chun  When you used the **installation.md** instructions to build the RealSense SDK from source code with CMake, did you include the CMake build flag **-DBUILD_NETWORK_DEVICE=ON** in the CMake build instruction to build the SDK with rs-server networking support included?

If you do have rs-server installed and have run it on the other Orin, did you follow the procedure at the link below to configure the **static IP** and provide it with your Orin's fixed IP?

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#26-setting-static-ip

Once a static IP has been configured then you should be able to access the camera in the RealSense Viewer as a **network camera**.

H-chun, 2023-05-09:
Thank you for replying to me so late. I have found the problem, it is a static IP issue. But a new issue has arisen. When I try to open the Stereo module or rgb Camera in the viewer, the viewer will flash back directly. Here are the corresponding logs
[image]

 09/05 05:41:14,938 INFO [281473343638480] (synthetic-stream-gl.cpp:80) Initializing rendering, GLSL=1
 09/05 05:41:14,938 INFO [281473343638480] (synthetic-stream-gl.cpp:89)  0 GPU objects initialized
 09/05 05:41:14,938 INFO [281473343638480] (synthetic-stream-gl.cpp:132) Initializing processing, GLSL=1
 09/05 05:41:14,968 INFO [281473343638480] (synthetic-stream-gl.cpp:146)  0 GPU objects initialized
 09/05 05:41:15,104 INFO [281473343638480] (rs.cpp:2701) Framebuffer size changed to 1344 x 756
 09/05 05:41:15,105 INFO [281473343638480] (rs.cpp:2701) Scale Factor is now 1
RealSense error calling rs2_get_device_info(dev:0xaaaaf9390b10, info:Product Id):
    info Product Id not supported by the device!

MartyG-RealSense, 2023-05-09:
The D455 camera model was released after the rs-server tool was developed and can have problems with rs-server.  For example, the D455's 5 FPS minimum speed is not supported because earlier RealSense 400 Series camera models such as D415 and D435 / D435i have a minimum FPS speed of 6.

H-chun, 2023-05-09:
Can I manually modify the frame rate and resolution I want through code, and then use it when compiling SDK? If possible, modify it at that location in the code?

MartyG-RealSense, 2023-05-09:
Unfortunately on a network camera connection you are limited to the resolution / FPS combinations described in the **Software Limitations** table linked to below.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#36-software-limitations

I am not aware of a method for editing the code of the networking tool to change the supported configurations.

H-chun, 2023-05-10:
Is there any way for me to set the frame rate and resolution to the combination I want in the network camera connection?

H-chun, 2023-05-10:
Hello, I only want to obtain 720p, 30fps video stream data without rendering on Vimer. You can only obtain these video stream data. Is there any way?

MartyG-RealSense, 2023-05-10:
The rs-server code can be found in the **Tools** folder of the SDK's source code.

https://github.com/IntelRealSense/librealsense/tree/master/tools/rs-server

It looks as though the supported FPS and resolution are defined in the lines of **RsServer.cpp** linked to below.

https://github.com/IntelRealSense/librealsense/blob/master/tools/rs-server/RsServer.cpp#L106-L118

To implement these changes, you would have to download the librealsense SDK's source code, edit this RsServer.cpp file and then build the SDK from the amended source code folder.

**********

In regard to your second question, are you asking if you can access network camera streams from a script without using realsense-viewer, please?  If you are then yes, you can make a script network-enabled.  A C++ example of doing so is at https://github.com/IntelRealSense/librealsense/issues/11576

H-chun, 2023-05-12:
Hello, we have tested 720p 30fps in SDK2.53.1 and found a low depth map frame rate. How can we optimize this situation? Replace the device's network card with a better one? Because currently I am testing on orin. The method of setting the device as AP and SAT has also been attempted. Can you provide some suggestions?

I have changed the following parameters:

echo 8388608 > /proc/sys/net/core/wmem_ default

echo 8388608 > /proc/sys/net/core/rmem_ default

MartyG-RealSense, 2023-05-12:
If you are using ROS, https://github.com/IntelRealSense/librealsense/issues/10730 discusses possible solutions for slow frame rate when using wi-fi.

MartyG-RealSense, 2023-05-19:
Hi @H-chun  Do you require further assistance with this case, please?  Thanks!

H-chun, 2023-05-19:
No need at the moment. Thank you

MartyG-RealSense, 2023-05-19:
Thanks very much for the update.  As you do not require further assistance at this time, I will close this case.  Thanks again!