realsenseai/librealsense #11786: Compile with mingw64 not found __builtin_ia32_paddsw128

frbitten, 2023-05-09:
I'm trying to compile the project with mingw64 and I get the following error:

'__builtin_ia32_phaddsw128' was not declared in this scope; did you mean '__builtin_ia32_paddsw128'

As far as I understand it is the use of SSE that generates this faulty use.

```[build] [1/6  16% :: 0.212] Performing update step for 'realsense'
[build] [2/6  33% :: 0.295] No patch step for 'realsense'
[build] [3/6  50% :: 4.120] Performing configure step for 'realsense'
[build] -- Checking internet connection...
[build] -- Internet connection identified
[build] -- Info: REALSENSE_VERSION_STRING=2.53.1
[build] -- Setting Windows configurations
[build] -- Building libcurl enabled
[build] -- using RS2_USE_WMF_BACKEND
[build] -- Building with TM2
[build] -- Fetching recommended firmwares:
[build] -- D4XX_FW_VERSION: 5.14.0.0
[build] -- SR3XX_FW_VERSION: 3.26.1.0
[build] -- T26X_FW_VERSION: 0.2.0.951
[build] -- L51X_FW_VERSION: 1.5.8.1
[build] -- L53X_FW_VERSION: 3.5.5.1
[build] -- disable link time optimization for fw project
[build] -- https://librealsense.intel.com/Releases/RS4xx/FW/D4XX_FW_Image-5.14.0.0.bin
[build] -- Download firmware 0;"skipping download as file already exists with expected SHA1 hash" for D4XX_FW_Image-5.14.0.0.bin
[build] -- https://librealsense.intel.com/Releases/SR300/FW/SR3XX_FW_Image-3.26.1.0.bin
[build] -- Download firmware 0;"skipping download as file already exists with expected SHA1 hash" for SR3XX_FW_Image-3.26.1.0.bin
[build] -- https://librealsense.intel.com/Releases/TM2/FW/target/0.2.0.951/target-0.2.0.951.mvcmd
[build] -- Download firmware 0;"skipping download as file already exists with expected SHA1 hash" for target-0.2.0.951.mvcmd
[build] -- https://librealsense.intel.com/Releases/L5xx/FW/L51X_FW_Image-1.5.8.1.bin
[build] -- Download firmware 0;"skipping download as file already exists with expected SHA1 hash" for L51X_FW_Image-1.5.8.1.bin
[build] -- https://librealsense.intel.com/Releases/L5xx/FW/L53X_FW_Image-3.5.5.1.bin
[build] -- Download firmware 0;"skipping download as file already exists with expected SHA1 hash" for L53X_FW_Image-3.5.5.1.bin
[build] -- Configuring done (3.3s)
[build] -- Generating done (0.2s)
[build] -- Build files have been written to: E:/engine/PA_Engine/EngineEcosystem/Engine/build/deps/Build/realsense
[build] [4/6  66% :: 33.733] Performing build step for 'realsense'
[build] FAILED: deps/Stamp/realsense/realsense-build E:/engine/PA_Engine/EngineEcosystem/Engine/build/deps/Stamp/realsense/realsense-build 
[build] cmd.exe /C "cd /D E:\engine\PA_Engine\EngineEcosystem\Engine\build\deps\Build\realsense && "C:\Program Files\CMake\bin\cmake.exe" --build . && "C:\Program Files\CMake\bin\cmake.exe" -E touch E:/engine/PA_Engine/EngineEcosystem/Engine/build/deps/Stamp/realsense/realsense-build"
[build] [10/168   0% :: 12.043] Building CXX object CMakeFiles/realsense2.dir/include/librealsense2/utilities/concurrency/dispatcher.cpp.obj
[build] FAILED: CMakeFiles/realsense2.dir/include/librealsense2/utilities/concurrency/dispatcher.cpp.obj 
[build] D:\msys64\ucrt64\bin\c++.exe -DBUILD_EASYLOGGINGPP -DCHECK_FOR_UPDATES -DCOM_MULTITHREADED -DCURL_STATICLIB -DELPP_NO_DEFAULT_LOG_FILE -DELPP_THREAD_SAFE -DHWM_OVER_XU -DNOMINMAX -DRS2_USE_WMF_BACKEND -DUNICODE -DWITH_TRACKING=1 -D_CRT_SECURE_NO_WARNINGS -D__SSSE3__ -IE:/engine/PA_Engine/EngineEcosystem/Engine/build/deps/Source/realsense -IE:/engine/PA_Engine/EngineEcosystem/Engine/build/deps/Build/realsense -IE:/engine/PA_Engine/EngineEcosystem/Engine/build/deps/Source/realsense/src -IE:/engine/PA_Engine/EngineEcosystem/Engine/build/deps/Source/realsense/third-party/realsense-file/rosbag/console_bridge/include -IE:/engine/PA_Engine/EngineEcosystem/Engine/build/deps/Source/realsense/third-party/realsense-file/rosbag/cpp_common/include -IE:/engine/PA_Engine/EngineEcosystem/Engine/build/deps/Source/realsense/third-party/realsense-file/rosbag/rosbag_storage/include -IE:/engine/PA_Engine/EngineEcosystem/Engine/build/deps/Source/realsense/third-party/realsense-file/rosbag/roscpp_serialization/include -IE:/engine/PA_Engine/EngineEcosystem/Engine/build/deps/Source/realsense/third-party/realsense-file/rosbag/rostime/include -IE:/engine/PA_Engine/EngineEcosystem/Engine/build/deps/Source/realsense/third-party/realsense-file/rosbag/roscpp_traits/include -IE:/engine/PA_Engine/EngineEcosystem/Engine/build/deps/Source/realsense/third-party/realsense-file/rosbag/roslz4/include -IE:/engine/PA_Engine/EngineEcosystem/Engine/build/deps/Source/realsense/third-party/realsense-file/rosbag/msgs -IE:/engine/PA_Engine/EngineEcosystem/Engine/build/deps/Source/realsense/third-party/realsense-file/boost -IE:/engine/PA_Engine/EngineEcosystem/Engine/build/deps/Source/realsense/third-party/realsense-file/lz4/lib -IE:/engine/PA_Engine/EngineEcosystem/Engine/build/deps/Source/realsense/include -IE:/engine/PA_Engine/EngineEcosystem/Engine/build/deps/Build/realsense/common/fw -ID:/msys64/ucrt64/include/libusb-1.0 -O3 -DNDEBUG -std=gnu++11 -MD -MT CMakeFiles/realsense2.dir/include/librealsense2/utilities/concurrency/dispatcher.cpp.obj -MF CMakeFiles\realsense2.dir\include\librealsense2\utilities\concurrency\dispatcher.cpp.obj.d -o CMakeFiles/realsense2.dir/include/librealsense2/utilities/concurrency/dispatcher.cpp.obj -c E:/engine/PA_Engine/EngineEcosystem/Engine/build/deps/Source/realsense/include/librealsense2/utilities/concurrency/dispatcher.cpp
[build] In file included from D:/msys64/ucrt64/lib/gcc/x86_64-w64-mingw32/12.2.0/include/immintrin.h:37,
[build]                  from D:/msys64/ucrt64/lib/gcc/x86_64-w64-mingw32/12.2.0/include/x86intrin.h:32,
[build]                  from D:/msys64/ucrt64/include/winnt.h:1658,
[build]                  from D:/msys64/ucrt64/include/minwindef.h:163,
[build]                  from D:/msys64/ucrt64/include/windef.h:9,
[build]                  from D:/msys64/ucrt64/include/windows.h:69,
[build]                  from E:/engine/PA_Engine/EngineEcosystem/Engine/build/deps/Source/realsense/third-party/easyloggingpp/src/easylogging++.h:380,
[build]                  from E:/engine/PA_Engine/EngineEcosystem/Engine/build/deps/Source/realsense/include/librealsense2/utilities/easylogging/easyloggingpp.h:22,
[build]                  from E:/engine/PA_Engine/EngineEcosystem/Engine/build/deps/Source/realsense/include/librealsense2/utilities/concurrency/dispatcher.cpp:5:
[build] D:/msys64/ucrt64/lib/gcc/x86_64-w64-mingw32/12.2.0/include/tmmintrin.h: In function '__m128i _mm_hadd_epi16(__m128i, __m128i)':
[build] D:/msys64/ucrt64/lib/gcc/x86_64-w64-mingw32/12.2.0/include/tmmintrin.h:42:20: error: '__builtin_ia32_phaddw128' was not declared in this scope; did you mean '__builtin_ia32_paddw128'?
[build]    42 |   return (__m128i) __builtin_ia32_phaddw128 ((__v8hi)__X, (__v8hi)__Y);```

frbitten, 2023-05-09:
Hi @MartyG-RealSense 
The project I work on will need to be compiled on Windows (X86 and Arm) and Linux (x86 and Arm). To facilitate the cross-compile, I opted to compile the code and better manage the target platform. With MingW64 I keep the same GNU standard in all compilers and avoid having to deal with specific cases of each compiler in the code.

Another important point for using MingW64 is not having the Visual Studio license costs. As far as I understand the SDK on the windows platform can only be compiled in Visual Studio, correct?

In the windows_config.cmake file, SSE support is enabled, are these instructions essential for the SDK? Or can the SDK be used without SSE support? As far as I've seen the code, if I remove the directive -D__SSSE3__ I won't have the error anymore. Would this change be possible?

MartyG-RealSense, 2023-05-10:
SSSE3 is used if **CUDA** support in the SDK is not enabled.  It is enabled by building the SDK with the flag **-DBUILD_WITH_CUDA=true** included in the CMake build instruction.  CUDA is dependent on the computer having an Nvidia graphics GPU though, so would not be practical if your application has to be able to run on any computer and not just ones with an Nvidia GPU.

When SSSE3 is enabled, it is typically used for depth to color alignment.  My understanding of the SDK's alignment process is that it uses SSE optimizations if supported, otherwise it uses unoptimized align. 

I do not know what the result will be if you remove the directive -D__SSSE3__.

frbitten, 2023-05-10:
SSSE3 is being enabled in all scenarios when the platform is windows. The way cmake is, it doesn't matter if you use CUDA or not.

You can see in this file:
https://github.com/IntelRealSense/librealsense/blob/master/CMake/windows_config.cmake

MartyG-RealSense, 2023-05-10:
I have highlighted your issue to my Intel RealSense colleagues to seek their advice about whether SSSE3 can be disabled.

MartyG-RealSense, 2023-05-12:
Thanks very much for your patience.  My colleagues have responded with the advice below.

*********

We recommend to use Visual Studio to compile the SDK on Windows.  SSSE3 can be disabled, but expect potential performance impact in certain performance sensitive usages.

In the SDK, some image processing and transformation, for example, YUY2 image format conversion, depth alignment, and pointcloud, are very computing intensive.  A few different optimizations are implemented in the current SDK to accelerate this processing, including GLSL with GPU, CUDA on Nvidia, and SSSE3 on CPU.

In normal usages, when a graphics context is available (see examples like realsense-viewer and the pointcloud example), GLSL acceleration is used.  CUDA optimization (limited to alignment and pointcloud) will be used if Nvidia GPU is available and the CUDA cmake flag is enabled.

When GPU accelerations are not available, SSSE3 optimized code will be used.

If none of these is available then processing will not be accelerated and performance may be impacted.

So if SSSE3 is disabled, performance will be impacted only when GPU acceleration is also not available.

frbitten, 2023-05-12:
Thanks for the feedback. Usage on windows will probably be just for testing and development and we don't have high performance requirements. The release version will be used on a Jetson with Cuda enabled.

I'll try unavoiding SSSE and give feedback on the behavior.

MartyG-RealSense, 2023-05-12:
Thanks very much.  I look forward to your next report.  Good luck!

frbitten, 2023-05-19:
I had to work on other areas due to a priority change and I didn't have time to take the test. Once I get the test done I will post the feedback.

MartyG-RealSense, 2023-06-13:
Case closed due to no further comments received.

frbitten, 2023-06-22:
Hi @MartyG-RealSense, sorry for the delayed response. But I only now had time to look into this problem.

It is not possible to compile the SDK with MingW. Even removing SSSE3 and several error blocks if the visual studio version is old, the code uses Microsoft ATL that only exists in Visual Studio. And there is no way around it.

It would be interesting in the future to allow the SDK to be compiled with MingW for windows. Since it is compiled with GCC for Linux.