realsenseai/librealsense #11787: the buffer size was insufficient.

975085538, 2023-05-10:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Firmware Version                   | | 
| Operating System & Version |   Win11  | 
| Kernel Version (Linux Only)    |                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.0 }                          | 
| Language                            |  {C# }                          | 
| Segment			|   | 

### Issue Description
I wanted to use the image control in wpf to display the depth camera in real time, but an exception occurred during execution: **the buffer size was insufficient**. How to solve this problem? **Here is my code:**
```
using (var frames = pipeline.WaitForFrames())
{
        using (var depthFrame = frames.DepthFrame)
        {
                  if (depthFrame != null)
                  {
                      var bytesPerPixel = depthFrame.BitsPerPixel;
                      var pixelData = new byte[depthFrame.DataSize];
                      Marshal.Copy(depthFrame.Data, pixelData, 0, (int)depthFrame.DataSize);
                      var imageWidth = depthFrame.Width;
                      var imageHeight = depthFrame.Height;
                      **var bitmap = BitmapSource.Create(imageWidth, imageHeight, 96, 96, PixelFormats.Rgb24, null, pixelData, imageWidth * bytesPerPixel);**
                      this.depthImage.Source = bitmap;
                  }
         }
}
```
The problem is in this line:
**var bitmap = BitmapSource.Create(xxx, xxx, 96, 96, xxx, null, xxx, xxx);**

MartyG-RealSense, 2023-05-10:
I researched the **buffer size was insufficient** error in relation to BitmapSource.Create.  Whilst I am not familiar with this code, most of it seems to conform with other examples that my research found.  The problem is most likely in my opinion to be with the **stride** (calculated by imageWidth * bytesPerPixel) or with pixelData.

Because `pixelData = new byte[depthFrame.DataSize];`, if the error relates to an insufficient buffer then I would recommend investigating pixelData first because it involves DataSize.

MartyG-RealSense, 2023-05-10:
Does it make any difference if you enclose imagewidth * bytesPerPixel inside brackets as it is multiplying two values together?

`var bitmap = BitmapSource.Create(imageWidth, imageHeight, 96, 96, PixelFormats.Rgb24, null, pixelData, (imageWidth * bytesPerPixel))`

975085538, 2023-05-10:
> 如果将图像宽度 * 字节每像素括在括号内，因为它将两个值相乘，这有什么区别吗？
> 
> `var bitmap = BitmapSource.Create(imageWidth, imageHeight, 96, 96, PixelFormats.Rgb24, null, pixelData, (imageWidth * bytesPerPixel))`

I tried it and it made no difference.

MartyG-RealSense, 2023-05-10:
A RealSense depth image is 16 bit.  Your line uses the RGB format Rgb24 though.  So might the bytesPerPixel be 24 instead of 16?

My C# programming knowledge is limited, so thanks very much for your patience!

975085538, 2023-05-10:
> 实感深度图像为 16 位。不过，您的线路使用 RGB 格式 Rgb24。那么每像素字节数可能是 24 而不是 16 吗？
> 
> 我的C#编程知识有限，非常感谢您的耐心等待！

Thanks for your reply, then PixelFormats are not using RGB24 but which should be used instead?

975085538, 2023-05-10:
> 如果您使用的是深度帧，则实感深度使用 16 位 **Z16** 格式。
> 
> 我发现的示例代码（如下）表明，如果格式为 Z16，则使用 **Gray16** 作为像素格式。
> 
> ```
>      BitmapSource CreateBitmapSource( Device device, StreamType stream )
>         {
>             var width = device.GetStreamWidth( stream );
>             var height = device.GetStreamHeight( stream );
>             var format = device.GetStreamFormat( stream );
> 
>             var pixelFormat = PixelFormats.Rgb24;
>             if ( format == FormatType.y8 ) {
>                 pixelFormat = PixelFormats.Gray8;
>             }
>             else if ( format == FormatType.z16 ) {
>                 pixelFormat = PixelFormats.Gray16;
>             }
> 
>             var bpp = pixelFormat.BitsPerPixel / 8;
> 
>             return BitmapSource.Create( width, height, 96, 96, pixelFormat, null,
>                 device.GetFrameData( stream ), width * height * bpp, width * bpp );
>         }
> ```
Thanks for your reply. I found that there is no StreamType class in SDK and GetStreamFormat() methods in Device. The version I used is 2.53.1.0. Here is the code for the Device class:
```
public class Device : RefCountedPooledObject
    {
        protected static Hashtable refCountTable = new Hashtable();

        protected static readonly object tableLock = new object();

        public InfoCollection Info
        {
            get;
            private set;
        }

        public ReadOnlyCollection<Sensor> Sensors => QuerySensors();

        internal override void Initialize()
        {
            lock (tableLock)
            {
                if (refCountTable.Contains(base.Handle))
                {
                    refCount = (refCountTable[base.Handle] as RefCount);
                }
                else
                {
                    refCount = new RefCount();
                    refCountTable[base.Handle] = refCount;
                }

                Retain();
            }

            Info = new InfoCollection(NativeMethods.rs2_supports_device_info, NativeMethods.rs2_get_device_info, base.Handle);
        }

        protected override void Dispose(bool disposing)
        {
            if (m_instance.IsInvalid)
            {
                return;
            }

            lock (tableLock)
            {
                IntPtr handle = base.Handle;
                base.Dispose(disposing);
                if (refCount.count == 0)
                {
                    refCountTable.Remove(handle);
                }
            }
        }

        internal Device(IntPtr ptr)
            : base(ptr, null)
        {
            Initialize();
        }

        internal Device(IntPtr ptr, Deleter deleter)
            : base(ptr, deleter)
        {
            Initialize();
        }

        internal static T Create<T>(IntPtr ptr) where T : Device
        {
            return ObjectPool.Get<T>(ptr);
        }

        internal static T Create<T>(IntPtr ptr, Deleter deleter) where T : Device
        {
            T val = ObjectPool.Get<T>(ptr);
            val.Reset(ptr, deleter);
            return val;
        }

        public ReadOnlyCollection<Sensor> QuerySensors()
        {
            object error;
            IntPtr ptr = NativeMethods.rs2_query_sensors(base.Handle, out error);
            using (SensorList sensorList = new SensorList(ptr))
            {
                Sensor[] array = new Sensor[sensorList.Count];
                sensorList.CopyTo(array, 0);
                return Array.AsReadOnly(array);
            }
        }

        public void HardwareReset()
        {
            NativeMethods.rs2_hardware_reset(base.Handle, out object _);
        }

        public bool Is(Extension extension)
        {
            object error;
            return NativeMethods.rs2_is_device_extendable_to(base.Handle, extension, out error) != 0;
        }

        public T As<T>() where T : Device
        {
            return Create<T>(base.Handle);
        }
    }
```

MartyG-RealSense, 2023-05-10:
I was suggesting just to change PixelFormats.Rgb24 to **PixelFormats.Gray16** to see what happens  :)

975085538, 2023-05-10:
Thank you for your reply, but there is no difference after I tried to modify it to Gray16:
`var bitmap = BitmapSource.Create(imageWidth, imageHeight, 96, 96, PixelFormats.Gray16, null, pixelData, (imageWidth * bytesPerPixel));`

975085538, 2023-05-10:
> 我找不到任何其他建议。我道歉。

Please let me know if you have any other suggestions. Thank you for your reply!

MartyG-RealSense, 2023-05-10:
Another approach my research found was to convert depth pixels into color pixels and use BGR2 as the PixelFormat.

https://stackoverflow.com/questions/8107762/kinect-background-removal

```
 //Convert depth information for a pixel into color information
                byte[] ColoredBytes = GenerateColoredBytes(e.ImageFrame);

                depthImage = e.ImageFrame.Image;
                image2.Source = BitmapSource.Create(depthImage.Width, depthImage.Height, 96, 96, PixelFormats.Bgr32, null,
                    ColoredBytes, depthImage.Width * PixelFormats.Bgr32.BitsPerPixel / 8);
```

975085538, 2023-05-11:
> Another approach my research found was to convert depth pixels into color pixels and use BGR32 as the PixelFormat.
> 
> https://stackoverflow.com/questions/8107762/kinect-background-removal
> 
> ```
>  //Convert depth information for a pixel into color information
>                 byte[] ColoredBytes = GenerateColoredBytes(e.ImageFrame);
> 
>                 depthImage = e.ImageFrame.Image;
>                 image2.Source = BitmapSource.Create(depthImage.Width, depthImage.Height, 96, 96, PixelFormats.Bgr32, null,
>                     ColoredBytes, depthImage.Width * PixelFormats.Bgr32.BitsPerPixel / 8);
> ```

Thank you for your reply. I have carefully studied the case you provided. But I didn't find the GenerateColoredBytes () method in the case, and what does the 'e' in 'e.mageframe' stand for? Hope to get more help, thank you!

975085538, 2023-05-12:
> 引用的脚本使用**GenerateColoredBytes（e.ImageFrame）**;用于称为Microsoft Kinect的旧非RealSense深度相机，因此代码不适用于RealSense。我将其用作一般示例来描述解决将深度像素转换为彩色像素问题的可能方法。
> 
> 如果在这种特殊情况下使用 WPF 而不是 WinForms，则实感用户建议使用以下代码片段来表示流式处理深度，他们说这些代码片段适用于 WPF。
> 
> ```
> private void SetupWindow(PipelineProfile pipelineProfile, out Action<VideoFrame> depth, out Action<VideoFrame> color)
>         {
>             using (var p = pipelineProfile.GetStream(Stream.Depth) as VideoStreamProfile)
>                 imgDepth.Source = new WriteableBitmap(p.Width, p.Height, 96d, 96d, PixelFormats.Rgb24, null);
>             depth = UpdateImage(imgDepth);
> }
> ```

Can you provide me with the complete code,？it doesn't look very clear without some methods，thank you！

MartyG-RealSense, 2023-05-12:
That code section was all that was provided in the case that I found it in, unfortunately.

At that time there was another RealSense user with the same problem at https://github.com/IntelRealSense/librealsense/issues/3413 but the script that they provided was WinForms instead of WPF.

975085538, 2023-05-13:
Thank you for your reply. I have referred to your example and solved the exception. But now there is a new problem, I cannot display depth camera and color camera in the window, the program did not report error, may I ask what is the reason? Here is my code:
```
private void RunInThread()
        {
            this.pipeline = new Pipeline();
            this.colorizer = new Colorizer();

            Config cfg = new Config();

            cfg.EnableStream(Stream.Depth, 640, 480, Format.Z16, 30);
            cfg.EnableStream(Stream.Color, 640, 480, Format.Rgb8, 30);

            this.pipeline.Start(cfg);

            using (var frames = this.pipeline.WaitForFrames())
            {
                var colorFrame = frames.ColorFrame.DisposeWith(frames);
                var depthFrame = frames.DepthFrame.DisposeWith(frames);

                try
                {
                    //imageSource = colorizer.Process<VideoFrame>(depthFrame).DisposeWith(colorFrame);

                    var colorized_depth = colorizer.Process(depthFrame).DisposeWith(frames);
                    var DepthImg = new Bitmap(640, 480, 1920, PixelFormat.Format24bppRgb, colorized_depth.Data);//新建深度图像的内容
                    var ColorImg = new Bitmap(460, 300, colorFrame.Stride, PixelFormat.Format24bppRgb, colorFrame.Data);//新建rgb图像的内容

                    BitmapSource bitmapSourceDep = Imaging.CreateBitmapSourceFromHBitmap(DepthImg.GetHbitmap(),IntPtr.Zero,Int32Rect.Empty,BitmapSizeOptions.FromEmptyOptions());
                    BitmapSource bitmapSourceCol = Imaging.CreateBitmapSourceFromHBitmap(DepthImg.GetHbitmap(),IntPtr.Zero,Int32Rect.Empty,BitmapSizeOptions.FromEmptyOptions());

                    var dispatcher = depthImage.Dispatcher;
                    dispatcher.Invoke(() =>
                    {
                        depthImage.Source = bitmapSourceDep;
                        colorImage.Source = bitmapSourceCol;
                    });
                }
                catch (Exception e)
                {
                    MessageBox.Show(e.Message);
                }
            }
        }
```

MartyG-RealSense, 2023-05-13:
Is there any difference if you move colorFrame and depthFrame inside the Try block?

```
try
{
var colorFrame = frames.ColorFrame.DisposeWith(frames);
var depthFrame = frames.DepthFrame.DisposeWith(frames);

  //imageSource = colorizer.Process<VideoFrame>(depthFrame).DisposeWith(colorFrame);
```

975085538, 2023-05-13:
Putting two lines of code in a try block makes no difference. In addition, I debug the program and each object has content and is not null.I don't know what the problem is.

MartyG-RealSense, 2023-05-13:
As your code is nested within a void - RunInThread() - presumably the code will not run until RunInThread() is called in your script somewhere.  Have you called it with a line like this:

`RunInThread(); `

975085538, 2023-05-13:
Hello, I'm using thread calls：
```
            Thread thread = new Thread(RunInThread);
            thread.Start();
```

MartyG-RealSense, 2023-05-13:
If you insert an instruction that causes a message to be printed then you can use it for debugging to see whether that section of code is being accessed by the script.  For example:

Console.WriteLine($"Yes");

975085538, 2023-05-13:
I use the following code to test, can pop up the form, but still can't show the camera：
`MessageBox.Show($"Yes");`

MartyG-RealSense, 2023-05-13:
The next thing you could check is your opening and closing brackets { and } and whether you have one too many closing brackets somewhere.  For example:

[image]

975085538, 2023-05-13:
Upon my inspection, this was not caused by parentheses, and the program compiled smoothly, but the camera was not displayed.

MartyG-RealSense, 2023-05-13:
Regarding this line:

`BitmapSource bitmapSourceCol = Imaging.CreateBitmapSourceFromHBitmap(DepthImg.GetHbitmap(),IntPtr.Zero,Int32Rect.Empty,BitmapSizeOptions.FromEmptyOptions());`

Would bitmapSourceCol not be **ColorImg.GetHbitmap** instead of DepthImg.GetHbitmap ?

975085538, 2023-05-15:
> Regarding this line:
> 
> `BitmapSource bitmapSourceCol = Imaging.CreateBitmapSourceFromHBitmap(DepthImg.GetHbitmap(),IntPtr.Zero,Int32Rect.Empty,BitmapSizeOptions.FromEmptyOptions());`
> 
> bitmapSourceCol looks as though it should be **ColorImg.GetHbitmap** instead of DepthImg.GetHbitmap ?

I've tried DepthImg.GetHbitmap and ColorImg.GetHbitmap for this line, but neither shows the camera directly.

MartyG-RealSense, 2023-05-15:
https://github.com/IntelRealSense/librealsense/issues/4848 takes the approach of putting the wait_for_frames() section within a **while** loop.

closed 2023-05-18 by 975085538