realsenseai/librealsense #11789: Visual Studio Code can't see imports

mericgeren, 2023-05-10:
Required Info:

  Camera Model:  D435 
  Firmware Version:  v2.50.0 
  Operating System & Version:  Linux (Ubuntu 18.04) 
  Kernel Version:  5.3.0-59-generic 
  Platform:  PC
  SDK Version:  2.3.2 
  Language:  python 
  Segment:  Robot 

***Issue Description***
<After successfully setting up and being able to display live feed using `roslaunch realsense2_camera rs_camera.launch align_depth:=false
` and RViz, i installed pyrealsense2 with command `pip install pyrealsense2`. Then, created and built a catkin package with dependencies rospy and sensor_msgs and with scripts directory inside. To write a ros node in Python i opened Visual Studio, selected **File**, then **Open File..** and there i moved to root of the package i created and opened from there. I have imported pyrealsense2, rospy, sensor_msgs and numpy. Although, i didn't run across any error or warning like "import could not be resolved" i can't get intellisense for members of sensor_msgs and pyrealsense2 and Visual Studio Code also seems to not be able to find members of these too. Could you help me on this issue please? >

MartyG-RealSense, 2023-05-10:
Hi @mericgeren  Intellisense does not provide auto-complete suggestions for all pyrealsense2 instructions.  This subject is discussed at https://github.com/IntelRealSense/librealsense/issues/10231 with three RealSense users of pyrealsense2 in that discussion reporting the same problem.

It is also discussed at https://github.com/IntelRealSense/librealsense/issues/10859 

A RealSense user at https://github.com/IntelRealSense/librealsense/pull/11624 submitted a solution at the end of March 2023.

mericgeren, 2023-05-10:
Hi @MartyG-RealSense i mean for example when i write "pyrealsense2" and then, press enter i can't see functions like config or pipeline in window appearing and after writing the function nevertheless, when i ctrl + right-click i am not directed into the code, to the place where originally these functions have been implemented as i would have with other classes like numpy. So, Visual Studio Code behaves just like i am writing something new.

MartyG-RealSense, 2023-05-10:
If Visual Studio Code is not providing Intellisense auto-completion at all then Microsoft recommend trying a restart of VS Code. 
 This can be done by performing the keyboard shortcut **Ctrl + Shift + P** and then input the command **Restart Window**.

mericgeren, 2023-05-10:
Actually, i got this issue only for pyrealsense2 and when i try to input the command ***Restart Window*** it shows me a message states "No Matching Commands"

mericgeren, 2023-05-10:
Thank you for your help. Just, couldn't get how to create that stub file with .pyi extension?

35selim, 2023-05-10:
Hi,
As stated in the issue, whose link is provided by @MartyG-RealSense above, you need to place the stub file along with the init file. Since class attributes and methods are reachable to you at the run time, there won't be any errors using the stub file. 
You should be aware of that this file is my very first implementation and not official so it can contain some type unmatchings.

mericgeren, 2023-05-11:
Just copied the contents of __init__.pyi file here: "https://github.com/IntelRealSense/librealsense/pull/11624/commits/8b5d4dfd9302873279ff02e4c7e0c3deb5dea7e9" into a blank text editor file and save it as "__init__.pyi" inside directory "librealsense/ wrappers/ python/ pyrealsense2 ". After doing that opened Visual Studio Code. But, unfortunately while i didn't get any "import could not be resolved" before saving "__init.pyi__". Now, i get such error for sensor_msgs and rospy and still can't get any suggestions for pyrealsense2.

35selim, 2023-05-11:
You can have a look at [here](https://bobbyhadz.com/blog/python-import-could-not-be-resolved-from-source-pylance-reportmissingmodulesource) for "import could not be resolved" errors. Be sure that your python interpreter, that is used by VScode currently, is using the `__init__.py` file inside `librealsense/wrappers/python/pyrealsense2`. If it is not the case; then, place the stub file inside the directory where the init file located. (Probably, it is located at something like **/lib/python3.*/site-packages/pyrealsense2)

mericgeren, 2023-05-11:
Thanks. Now i can list members of pyrealsense2 and when i do ctrl + left-click a tab displaying ' \_\_init__.pyi ' is opened. Also, tried 'pip install requests' and i got the message stating: "Requirement already satisfied: requests in ./.local/lib/python3.6/site-packages (2.27.1)
" and checked i am still using the correct interpreter.

MartyG-RealSense, 2023-05-11:
Thanks so much @35selim for your kind assistance to @mericgeren  :)

mericgeren, 2023-05-12:
Now code runs without any issue. Thanks @35selim and @MartyG-RealSense for all the help you offer. If you don't mind i am closing this issue as solved.

MartyG-RealSense, 2023-05-12:
That's excellent news, @mericgeren - thanks very much for the update  :)