realsenseai/librealsense #11790: D435i mechanical parameters for installation between various camera modules?

JACKLiuDay, 2023-05-11:
Hi, thank you for you guys great work about librealsense. I have some problem with my realsense D435i. I want to use my lidar to provide depth param for my D435i color image. And I have the installation params between my lidar and D435i. So I need the translation and RPY params between RGB module and my ouster lidar. I found some calibration project for calculating the extrinsic parameter for lidar and camera. While these projects seem to be cumbersome. Could you give me some advice?

MartyG-RealSense, 2023-05-11:
Hi @JACKLiuDay  You can obtain the extrinsic parameters between two sensors on the same camera (such as color to depth or depth to infrared1) by launching the RealSense SDK's 'rs-enumerate-devices' tool in calibration information mode with the command **rs-enumerate-devices -c**

JACKLiuDay, 2023-05-12:
> Hi @JACKLiuDay You can obtain the extrinsic parameters between two sensors on the same camera (such as color to depth or depth to infrared1) by launching the RealSense SDK's 'rs-enumerate-devices' tool in calibration information mode with the command **rs-enumerate-devices -c**

Hi,thank you for your quick reply. I will try it with my D435i camera.

closed 2023-05-15 by JACKLiuDay