realsenseai/librealsense #11792: Calibration problem

zrd1234, 2023-05-11:
Hello, may I ask the difference between the internal parameters fx and fy of D435i camera and the calibration results by kalibr is more than 100, may I ask which is more accurate

zrd1234, 2023-05-11:
The reprojection error is 0.5

MartyG-RealSense, 2023-05-11:
Hi @zrd1234  I would always recommend relying on the calibration values of official RealSense tools, which can access the calibration data stored inside the camera hardware.  An example is launching the 'rs-enumerate-devices' tool in calibration information mode with the command **rs-enumerate-devices -c**

Whilst some RealSense users do use Kalibr, it is not designed specifically for use with RealSense cameras like the official tools are.

zrd1234, 2023-05-11:
> Hi @zrd1234 I would always recommend relying on the calibration values of official RealSense tools, which can access the calibration data stored inside the camera hardware. An example is launching [...]

Hi, I just want to check the camera's fx,fy.cx.cy and distortion parameters. camera_info does not see the distortion parameter.

MartyG-RealSense, 2023-05-11:
rs-enumerate devices -c can provide all of this information.  On RealSense cameras, cx and cy (the principal point) are named ppx and ppy.

[image]

On a D435i camera the five distortion coefficients are always all '0' for all stream types.  A RealSense team member provides the reason for this at https://github.com/IntelRealSense/librealsense/issues/1430#issuecomment-375945916

MartyG-RealSense, 2026-10-23:
Case closed due to no further comments received after a long time period.

closed 2026-10-23 by MartyG-RealSense