realsenseai/librealsense #11795: Settings of RGB sensor

r91andersson, 2023-05-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | 2021-05 | 
| Operating System & Version |  Linux (Ubuntu 18.04)  | 
| Kernel Version (Linux Only)    | -                                         | 
| Platform                                 |NVIDIA NX   |
| SDK Version                            | - }                          | 
| Language                            |  {C }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
I have two images with different settings (I suppose), the quality of the image is completely different if you look at the two images below, could you tell me which setting on the realsense camera affects the image like this?

[image]

[image]

MartyG-RealSense, 2023-05-12:
Hi @r91andersson  The warm intensity of the colors in the lower image looks as though the value of the RGB **White Balance** option has been increased.  As White Balance is increased above the default value of '4600', the RGB image's colors become progressively stronger, like pale green grass becoming a strong, vibrant green.

MartyG-RealSense, 2023-05-23:
Hi @r91andersson  Do you have an update about this case that you can provide, please?  Thanks!

r91andersson, 2023-05-30:
> Hi @r91andersson <https://github.com/r91andersson> Do you require further
> assistance with this case, please? Thanks!

No thanks! Thanks for the help

closed 2023-06-06 by MartyG-RealSense