realsenseai/librealsense #11796: A problem when converting cv::mat to rs2::frame

YANSU7, 2023-05-12:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435  | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   {Win 10  | 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 | PC  |
| SDK Version                            |  v2.49.0 }                          | 
| Language                            |  C++ }                          | 
| Segment			|                           | 

### Issue Description
Hello, IntelRealSense Customer Service
I tried to convert cv::Mat to rs2::frame using the code inside this URL(https://github.com/IntelRealSense/librealsense/issues/2634).
I successfully converted rs2::frame to cv::Mat and displayed it. (Never mind about the colour)(fig.1)
[image]

However, I found that if I used the code
memcpy((void*)res.get_data(), image.data, w * h * 2).
the displayed image would disappear partially.(fig.2)
[image]

Of course there could be other reasons.
I'm not a professional programmer, so I'd really appreciate your help, thanks!

Here is my code:

// License: Apache 2.0. See LICENSE file in root directory.
// Copyright(c) 2017 Intel Corporation. All Rights Reserved.

#include <librealsense2/rs.hpp> // Include RealSense Cross Platform API
#include <opencv2/opencv.hpp>   // Include OpenCV API
#include "example.hpp" 

int main(int argc, char* argv[]) try
{
    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_RGB8, 30);
    cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);
    rs2::pipeline pipe;
    auto profile = pipe.start(cfg);
    auto sensor = profile.get_device().first<rs2::depth_sensor>();

    // Set the device to High Accuracy preset of the D400 stereoscopic cameras   
    if (sensor && sensor.is<rs2::depth_stereo_sensor>())
    {
        sensor.set_option(RS2_OPTION_VISUAL_PRESET, RS2_RS400_VISUAL_PRESET_HIGH_ACCURACY);
    }
    auto stream = profile.get_stream(RS2_STREAM_DEPTH).as<rs2::video_stream_profile>();
    using namespace cv;
    const auto window_name = "Display Image";
    namedWindow(window_name, WINDOW_AUTOSIZE);

    // Create custom processing block and its output queue:
    rs2::frame_queue q;
    rs2::processing_block pb(
        [](rs2::frame f, rs2::frame_source& src)
        {
            // For each input frame f, do:

            const int w = f.as<rs2::video_frame>().get_width();
            const int h = f.as<rs2::video_frame>().get_height();

            // frame --> cv
            Mat image(Size(w, h), CV_16UC1, (void*)f.get_data(), Mat::AUTO_STEP);
            // do some (silly) processing
            //cv::GaussianBlur(image, image, cv::Size(0, 0), 3);

            // Allocate new frame. Copy all missing data from f.
            // This assumes the output is same resolution and format
            // if not true, need to specify parameters explicitly
            auto res = src.allocate_video_frame(f.get_profile(), f);

            // copy from cv --> frame
            memcpy((void*)res.get_data(), image.data, w * h * 2);

            // Send the resulting frame to the output queue
            src.frame_ready(res);

        });
    pb.start(q); // Bind output of the processing block to be enqueued into the queue

    while (cv::waitKey(1) < 0 && cv::getWindowProperty(window_name, cv::WND_PROP_VISIBLE) >= 1)
    {
        rs2::frameset data = pipe.wait_for_frames(); // Wait for next set of frames from the camera
        rs2::video_frame color = data.get_color_frame();

        // Send color frame for processing
        pb.invoke(color);
        // Wait for results
        color = q.wait_for_frame();

        // Query frame size (width and height)
        const int w = color.as<rs2::video_frame>().get_width();
        const int h = color.as<rs2::video_frame>().get_height();

        // Create OpenCV matrix of size (w,h) from the colorized depth data     frame -> CV
        Mat image(Size(w, h), CV_8UC3, (void*)color.get_data(), Mat::AUTO_STEP);

        // Update the window with new data
        imshow(window_name, image);
    }

    return EXIT_SUCCESS;
}
catch (const rs2::error& e)
{
    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    return EXIT_FAILURE;
}
catch (const std::exception& e)
{
    std::cerr << e.what() << std::endl;
    return EXIT_FAILURE;
}

YANSU7, 2023-05-12:
Thank you very much for your reply.
It is great that the colour of the image returns to normal after the BGR change.

Unfortunately, the image is still not displayed, even after I have modified the W and H of the image.
I have changed the values of W and H to fixed values, but the images are not displayed.
I have also tried to completely remove the following codes:
```
            // For each input frame f, do:
            const int w = 640;
            const int h = 480;
```
But this causes the program to report an error.
So I deleted the program set to high accuracy. Still not working.
```
auto sensor = profile.get_device（）.firstrs2：:d epth_sensor（）;
// Set the device to High Accuracy preset of the D400 stereoscopic cameras   
if (sensor && sensor.is<rs2::depth_stereo_sensor>())
{
    sensor.set_option(RS2_OPTION_VISUAL_PRESET, RS2_RS400_VISUAL_PRESET_HIGH_ACCURACY);
}
auto stream = profile.get_stream(RS2_STREAM_DEPTH).as<rs2::video_stream_profile>();
```
Here's a graph of my display results.
[image]

And here is my code:
```
// License: Apache 2.0. See LICENSE file in root directory.
// Copyright(c) 2017 Intel Corporation. All Rights Reserved.

#include <librealsense2/rs.hpp> // Include RealSense Cross Platform API
#include <opencv2/opencv.hpp>   // Include OpenCV API
#include "example.hpp" 

using namespace cv;

int main(int argc, char* argv[]) try
{
    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_BGR8, 30);
    cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);
    rs2::pipeline pipe;
    auto profile = pipe.start(cfg);

    const auto window_name = "Display Image";
    namedWindow(window_name, WINDOW_AUTOSIZE);

    // Create custom processing block and its output queue:
    rs2::frame_queue q;
    rs2::processing_block pb(
        [](rs2::frame f, rs2::frame_source& src)
        {
            // For each input frame f, do:

            const int w = 640;
            const int h = 480;

            // frame --> cv
            Mat image(Size(w, h), CV_16UC1, (void*)f.get_data(), Mat::AUTO_STEP);

            // Allocate new frame. Copy all missing data from f.
            // This assumes the output is same resolution and format
            // if not true, need to specify parameters explicitly
            auto res = src.allocate_video_frame(f.get_profile(), f);

            // copy from cv --> frame
            memcpy((void*)res.get_data(), image.data, w * h * 2);

            // Send the resulting frame to the output queue
            src.frame_ready(res);

        });
    pb.start(q); // Bind output of the processing block to be enqueued into the queue

    while (cv::waitKey(1) < 0 && cv::getWindowProperty(window_name, cv::WND_PROP_VISIBLE) >= 1)
    {
        rs2::frameset data = pipe.wait_for_frames(); // Wait for next set of frames from the camera
        rs2::video_frame color = data.get_color_frame();

        // Send color frame for processing
        pb.invoke(color);
        // Wait for results
        color = q.wait_for_frame();

        // Query frame size (width and height)
        const int w = 640;
        const int h = 480;

        // Create OpenCV matrix of size (w,h) from the colorized depth data     frame -> CV
        Mat image(Size(w, h), CV_8UC3, (void*)color.get_data(), Mat::AUTO_STEP);

        // Update the window with new data
        imshow(window_name, image);
    }
    return EXIT_SUCCESS;
}
catch (const rs2::error& e)
{
    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    return EXIT_FAILURE;
}
catch (const std::exception& e)
{
    std::cerr << e.what() << std::endl;
    return EXIT_FAILURE;
}
```

MartyG-RealSense, 2023-05-12:
Is the black area still there if you remove the '2' multiplication from the memcpy instruction?

`memcpy((void*)res.get_data(), image.data, w * h);`

YANSU7, 2023-05-13:
Thank you very much, I have solved the problem.
The result of the debugging is this:
If I change this line of code to this:
```
            // copy from cv --> frame
            memcpy((void*)res.get_data(), image.data, w * h * 1);
```
I will get about 1/3 of the original image.
If I change this line of code to this:
```
            // copy from cv --> frame
            memcpy((void*)res.get_data(), image.data, w * h * 2);
```
I will get about 2/3 of the original image.
So I changed the final constant to 3 and got a really great image. (I'm sorry, I forgot to take a screenshot)
```
            // copy from cv --> frame
            memcpy((void*)res.get_data(), image.data, w * h * 3);
```
Finally, I also tried to change the final constant to 4, but the program would report an error. I guess it was because I didn't have enough information in the image to cause the error, but I've found the problem, thanks a lot!

MartyG-RealSense, 2023-05-13:
You are very welcome.  It's excellent news that you have achieved a solution.  Thanks very much for the update and for sharing what worked for you!

closed 2023-05-21 by MartyG-RealSense