realsenseai/librealsense #11798: Ubuntu 22 + Docker + ROS Humble

stephenadhi, 2023-05-12:
Hi, 
I use the lines for the installation process from this [Dockerfile](https://github.com/IntelRealSense/librealsense/blob/master/scripts/Docker/Dockerfile) 
`# Build and install
RUN cd /usr/src/librealsense \
    && mkdir build && cd build \
    && cmake \
    -DCMAKE_C_FLAGS_RELEASE="${CMAKE_C_FLAGS_RELEASE} -s" \
    -DCMAKE_CXX_FLAGS_RELEASE="${CMAKE_CXX_FLAGS_RELEASE} -s" \
    -DCMAKE_INSTALL_PREFIX=/opt/librealsense \    
    -DBUILD_GRAPHICAL_EXAMPLES=OFF \
    -DBUILD_PYTHON_BINDINGS:bool=true \
    -DCMAKE_BUILD_TYPE=Release ../ \
    && make -j$(($(nproc)-1)) all \
    && make install `

#0 738.1   Add the installation prefix of "realsense2" to CMAKE_PREFIX_PATH or set
[2023-05-12T12:59:50.154Z] #0 738.1   "realsense2_DIR" to a directory containing one of the above files.  If
#0 738.1   "realsense2" provides a separate development package or SDK, be sure it has
#0 738.1   been installed.

Where should I provide the CMAKE_PREFIX_PATH to?

I also tried to install the Realsense SDK in a container following this instruction for jetson https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md but got stuck in instruction 3, the error is unable to locate librealsense-utils and librealsense-dev. Can anyone provide further assistance?

MartyG-RealSense, 2023-05-12:
Hi @stephenadhi  Are you using an Nvidia Jetson as your reference to the installation_jetson.md instructions suggests?  If you are then the Dockerfile that you linked to is part of a Docker tutorial that is designed for use with computers with an **x86** architecture (like desktop and laptop PCs).  Jetson boards have an **Arm** architecture.

stephenadhi, 2023-05-12:
Hi, yes I know, so my custom dockerfile is already for jetson using ARM architecture. https://github.com/stephenadhi/docker-ros-containers/blob/humble-jetpack-car-dev/docker/Dockerfile

MartyG-RealSense, 2023-05-12:
My knowledge of Docker is limited, unfortunately.  My research into your question did though find a guide for installing RealSense ROS in a Docker container that recommended installing from source code on Arm64 using the instructions at the link below.

https://github.com/2b-t/realsense-ros2-docker#12-installation-from-source

The instructions use a Dockerfile inspired by the Intel x86 Dockerfile that you linked to earlier.

MartyG-RealSense, 2023-05-21:
Hi @stephenadhi  Do you require further assistance with this case, please?  Thanks!

stephenadhi, 2023-05-21:
Hi, thank you for this link, I managed to build the Dockerfile from the source!

> My knowledge of Docker is limited, unfortunately. My research into your question did though find a guide for installing RealSense ROS2 in a Docker container that recommended installing from source code on Arm64 using the instructions at the link below.
> 
> https://github.com/2b-t/realsense-ros2-docker#12-installation-from-source
> 
> The instructions use a Dockerfile inspired by the Intel x86 Dockerfile that you linked to earlier.

closed 2023-05-21 by stephenadhi

MartyG-RealSense, 2023-05-21:
You are very welcome.  It's great to hear that you achieved a solution.  Thanks very much for the update!