realsenseai/librealsense #11800: Processing blocks don't always retain the new values set in the Unity editor

nataliesettles, 2023-05-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.14.00.00 | 
| Operating System & Version |  Win 10  | 
| Platform                                 | PC  |
| SDK Version                            |  2.53.1.4623  | 
| Language                            |  Unity  2021.2.14 | 

### Issue Description

When I set new values in the RS processing profile and blocks that I've created for my project, the values sometimes revert to the defaults or sometimes retain the new values - often a mix of these two things. Do I have to create a new copy of the script and hardcode my choices in order to retain them, or is there another way of assuring that the choices I make in the GUI are retained when the project runs? 

For example, right now I have a very basic profile running in a project with just a Colorizer block. When I set the Colorizer to "Fixed" and "Hue" in the drop-down menus and press play, the settings retain "Fixed", but change "Hue" to "Jet", and when I stop play "Jet" is now selected in the drop-down menu and my choice of "Hue" is gone.  This sort of value swapping happens with other blocks too - this is just one example.

Thanks!

MartyG-RealSense, 2023-05-12:
Hi @nataliesettles  I was not able to test this due to the RealSense Unity wrapper's example scenes not having a configurable Colorizer.  Could you provide a link to an .exe build folder of your project that I can download and test to see whether I can replicate the settings changes that you describe, please.  Thanks!

nataliesettles, 2023-05-15:
Hi @MartyG-RealSense - here is a scene and processing profile to test the colorizer settings. It's for Unity 2021.2.14 with the RS SDK 2.0. Once you open the scene, click the processing profile in the ProcessingPipe and adjust the values as you wish in the Inspector. Press play in the editor and some if not all of these values will change on play. Why does the profile have no persistent memory of the values set in the editor/inspector? Do I need to set them by script instead? Thanks!

[ColorizerTest.zip](https://github.com/IntelRealSense/librealsense/files/11480024/ColorizerTest.zip)

MartyG-RealSense, 2023-05-16:
I imported your scene but could not find the Colorizer block that you mentioned and so could not change the colorizer settings with drop-down menus.  Can you explain how to find these options, please?

nataliesettles, 2023-05-16:
Hi @MartyG-RealSense - In the scene click on the ProcessingPipe in the hierarchy and then in the Inspector double-click the scriptable object called `TouchDetectionImageProcessing` which is assigned to the field called "Profile." When you've double-clicked that it should show up in the Inspector and will show the colorized module with its values. If you leave that open in the inspector while you change its values and then press play, you should see the issue I mentioned.

nataliesettles, 2023-05-16:
Oh! Right. Sorry - yes, that's the right name. I gave the profile a new name `TestProcessing` in the minimal test scene to show this problem.

MartyG-RealSense, 2023-05-16:
Thanks very much.  I was able to successfully test the colorizer using the method that you described. 

The color scheme that Hue changes to apparently depends on which Visual Preset is selected.  If **Dynamic** was selected instead of Fixed then the Hue setting remained Hue when play is pressed.

If you open the **RsColorizer** script in Visual Studio, you can edit line 24 to change Fixed from a value of '1' to a value of '0'.  As Dynamic is set to 0, this should cause the Fixed preset to remain on Hue when play is pressed.

[image]

nataliesettles, 2023-05-16:
Unfortunately, that doesn't fix it on my end - I lose both values when I do that (they revert to Dynamic and Jet). I was also hoping to find a solution that makes the code reusable instead of hard-coded if possible.

I'm still trying a few things.

MartyG-RealSense, 2023-05-17:
You can force Hue to be the only supported color scheme by removing all other settings from the ColorScheme list whilst still having the full range of Visual Presets available.

[image]

nataliesettles, 2023-05-17:
I finally fixed it. The mess was in the logic for the `UpdateOptions` function. With this change, there is no need to hardcode any of the fields in order to assure that the options chosen in the inspector persist when played in the editor.

**Original version of `UpdateOptions` in SDK rsColorizer that did not retain changes made in the Inspector:**
`private void UpdateOptions()
    {
        if (presetOption.Value != (float)visualPreset)
        {
            presetOption.Value = (float)visualPreset;

            colorScheme = (ColorScheme)schemeOption.Value;
            histogramEqualization = histEqOption.Value != 0f;
            minDist = minDistOption.Value;
            maxDist = maxDistOption.Value;
        }
        else
        {
            if (schemeOption.Value != (float)colorScheme)
                schemeOption.Value = (float)colorScheme;

            if (histEqOption.Value != (float)(histogramEqualization ? 1 : 0))
                histEqOption.Value = (float)(histogramEqualization ? 1 : 0);

            if (minDistOption.Value != minDist)
                minDistOption.Value = minDist;

            if (maxDistOption.Value != maxDist)
                maxDistOption.Value = maxDist;
        }
    }`

**Changed (Fixed!) version of `UpdateOptions` that retained field assignments made in the Inspector:**
`private void UpdateOptions()
    {
        if (presetOption.Value != (float)visualPreset)
            presetOption.Value = (float)visualPreset;
        
        if (schemeOption.Value != (float)colorScheme)
            schemeOption.Value = (float)colorScheme;

        if (histEqOption.Value != (float)(histogramEqualization ? 1 : 0))
            histEqOption.Value = (float)(histogramEqualization ? 1 : 0);

        if (minDistOption.Value != minDist)
            minDistOption.Value = minDist;

        if (maxDistOption.Value != maxDist)
            maxDistOption.Value = maxDist;
    }`

closed 2023-05-17 by nataliesettles