realsenseai/librealsense #11802: Runtime Error: Device not Connected.

mamanyusufkhan, 2023-05-14:
My Colab cannot detect my Realsense Depth Camera D435 although when I run it in my local drive, it is able to detect it. Can you help me with it?

MartyG-RealSense, 2023-05-21:
Hi @mamanyusufkhan  Do you require further assistance with this case, please?  Thanks!

mamanyusufkhan, 2023-05-21:
No I am good for now. However, do you have any suggestions on how I can
directly take a video input instead of uploading a .bag file?

MartyG-RealSense, 2023-05-21:
I could only find information about accessing ordinary webcam video through Colab but not accessing RealSense SDK functions.

https://dev.to/0xkoji/use-webcam-on-google-colaboratory-in-1-min-1mo7

In the RealSense Colab example project that I linked to above, you can test with a live camera instead of a bag by commenting out the **cfg.enable_device_from_file** line.

# Setup:
pipe = rs.pipeline()
cfg = rs.config()
**cfg.enable_device_from_file("object_detection.bag")**
profile = pipe.start(cfg)

MartyG-RealSense, 2023-05-27:
Hi @mamanyusufkhan  Do you have an update about this case that you can provide, please?  Was the camera still undetected if you commented out the enable_device_from_file instruction?

MartyG-RealSense, 2023-05-29:
If you have a bag file that was recorded with the RealSense SDK then the SDK's rs-convert tool can extract the bag's frames to .bin depth matrix files.

https://github.com/IntelRealSense/librealsense/tree/master/tools/convert

If the bag file contains an RGB stream then 3D XYZ values for a specific coordinate instead of the entire image can be obtained using the SDK instruction rs2_project_color_pixel_to_depth_pixel which - as the name suggests - converts a 2D color pixel to a 3D depth pixel.  Below are some examples of script references for this instruction.

C++
https://github.com/IntelRealSense/librealsense/issues/2982

Python
https://github.com/IntelRealSense/librealsense/issues/5603#issuecomment-574019008

mamanyusufkhan, 2023-06-03:
For the time being, No. But If I need anything else, I'll let you know.
Thanks for helping!