realsenseai/librealsense #11803: How to set up Camera L515 in c++

0DWADE0, 2023-05-15:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.53.1 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {AR }                          | 

### Issue Description
<Hi, I am developing L515 using c++ and I am having a problem setting the camera parameters. I found the Intel Realsense Cross-platform API, but I couldn't figure out how to set it up.
[image]
>

MartyG-RealSense, 2023-05-15:
Hi @0DWADE0  The RealSense SDK supports C++ by default so you do not need to install an additional compatibility 'wrapper'.  The link below has details about how to install the SDK on Windows and set up a RealSense project in Visual Studio using **.props** property sheet files provided by the SDK.

https://support.intelrealsense.com/hc/en-us/community/posts/4412175411859/comments/4412182682387

In regard to the specific example of accessing L515 Visual Presets with C++, https://github.com/IntelRealSense/librealsense/issues/9071#issuecomment-878552273 provides advice about how to do so.

MartyG-RealSense, 2026-10-23:
This open GitHub issue relating to the RealSense L515 camera is being closed because the L515 model is End of Life (EOL) and no longer supported.  We welcome the creation of new issues relating to supported cameras.

closed 2026-10-23 by MartyG-RealSense