realsenseai/librealsense #11805: Isn't working SDK app on Win10 with DBUILD_NETWORK_DEVICE=ON

MrHellBond, 2023-05-15:
Hello!
I tried to build new SDK with flag DBUILD_NETWORK_DEVICE=ON.
1.I prepared CMAKE (cmake-gui) 3.26.3 with flag DBUILD_NETWORK_DEVICE=ON for Configure and Generate Project
These steps passed good.
2.Opened project in Visual Studio 2022 and built new SDK.
3.Tried to open new Realsense's exe-file and still have closed network features.

Could you help me to solve this issue?

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 
| Firmware Version                   | 05.14.00.00
| Operating System & Version |   {Windows 10Pro 22H2, 19045.2965 
| Platform                                 | PC-Windows10/Raspberry Pi 
| SDK Version                           |  2.53.14623                                             
| Segment			        |  other
[image]
s                      
![2](https://github.com/IntelRealSense/librealsense/assets/13
[log-cmake.txt](https://github.com/IntelRealSense/librealsense/files/11475396/log-cmake.txt)
3221298/6aa9c14c-80c0-4561-a380-c47e1d73e42f)
[log-VS.txt](https://github.com/IntelRealSense/librealsense/files/11475398/log-VS.txt)

MartyG-RealSense, 2023-05-15:
Hi @MrHellBond  The **rs-server** component of the open-source ethernet networking tool that is necessary for accessing a RealSense camera remotely over a network can only be compiled on Linux, unfortunately, as advised at the link below.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#32-building-from-source

This networking tool is being removed from the RealSense SDK in the next version, likely numbered 2.54.0.  This is because it was an experimental proof of concept rather than an official Intel feature of the RealSense SDK.

MrHellBond, 2023-05-15:
> 

Hello again!
Thanks for your information. But please clerify some questions:
1. I need to build SDK only for Linux. Ubuntu for an example?
2. Network tools will be closed in next version of SDK. Is it mention for WIndows version or for Linux or for both systems?
3. Could you give more detail information about not-supported settings for D455 camera?

MartyG-RealSense, 2023-05-15:
1.  The networking should work if built on Linux, including Ubuntu.  

2.  It would be for both Windows and Linux.

3.  Any mode in the table at the link below that uses 6 FPS would be incompatible.

closed 2023-05-27 by MartyG-RealSense