realsenseai/librealsense #11807: Compiling error on arm64, RK3588. unresolvable R_AARCH64_ADR_PREL_PG_HI21 relocation against symbol `malloc@@GLIBC_2.17'

dongdongbh, 2023-05-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   | - | 
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    | 5.10.110-rockchip-rk3588                                        | 
| Platform                                 | rk3588s [Orangepi 5B](http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/details/Orange-Pi-5B.html)  |
| SDK Version                            |  v2.50.0                          | 
| Language                            |  C                       | 
| Segment			|  Robot                          | 

### Issue Description
Compiling error with `v2.50.0`
```
cmake ../ -DFORCE_RSUSB_BACKEND=ON -DCMAKE_BUILD_TYPE=Release
sudo make uninstall && make clean && make -j8
..................
[ 97%] Building CXX object tools/depth-quality/CMakeFiles/rs-depth-quality.dir/depth-quality-model.cpp.o
[ 98%] Linking CXX executable realsense-viewer
[ 99%] Building CXX object tools/depth-quality/CMakeFiles/rs-depth-quality.dir/__/__/third-party/imgui/imgui.cpp.o
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(easy.c.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `malloc@@GLIBC_2.17' which may bind externally can not be used when making
a shared object; recompile with -fPIC
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(easy.c.o)(.text+0x30): unresolvable R_AARCH64_ADR_PREL_PG_HI21 relocation against symbol `malloc@@GLIBC_2.17'
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
make[2]: *** [tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/build.make:498: tools/realsense-viewer/realsense-viewer] Error 1
make[1]: *** [CMakeFiles/Makefile2:2076: tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 99%] Building CXX object tools/depth-quality/CMakeFiles/rs-depth-quality.dir/__/__/third-party/imgui/imgui_draw.cpp.o
[ 99%] Building CXX object tools/depth-quality/CMakeFiles/rs-depth-quality.dir/__/__/third-party/imgui/imgui_impl_glfw.cpp.o
[100%] Building C object tools/depth-quality/CMakeFiles/rs-depth-quality.dir/__/__/third-party/glad/glad.c.o
[100%] Building C object tools/depth-quality/CMakeFiles/rs-depth-quality.dir/__/__/third-party/tinyfiledialogs/tinyfiledialogs.c.o
[100%] Building CXX object tools/depth-quality/CMakeFiles/rs-depth-quality.dir/__/__/third-party/easyloggingpp/src/easylogging++.cc.o
[100%] Linking CXX executable rs-depth-quality
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(easy.c.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `malloc@@GLIBC_2.17' which may bind externally can not be used when making
a shared object; recompile with -fPIC
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(easy.c.o)(.text+0x30): unresolvable R_AARCH64_ADR_PREL_PG_HI21 relocation against symbol `malloc@@GLIBC_2.17'
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
make[2]: *** [tools/depth-quality/CMakeFiles/rs-depth-quality.dir/build.make:513: tools/depth-quality/rs-depth-quality] Error 1
make[1]: *** [CMakeFiles/Makefile2:2106: tools/depth-quality/CMakeFiles/rs-depth-quality.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

MartyG-RealSense, 2023-05-16:
Hi @dongdongbh  Using **-DFORCE_RSUSB_BACKEND=ON** is the correct action when building librealsense for Rockchip (RK) devices.  Using a -J number for the **make** instruction can cause random build crashes on devices with a low-end hardware specification though.  Please try building with a lower -J number such as **-j4** or by leaving out the J-number and just doing **sudo make uninstall && make clean && make** (though the build time will be longer).

dongdongbh, 2023-05-16:
I build with  -j4 which shows same error. I think it is an error, not a build crash. The error related to `_AARCH64_ADR_PREL_PG_HI21 relocation against symbol 'malloc@@GLIBC_2.17' `

MartyG-RealSense, 2023-05-16:
The errors seem to relate to graphical tools (realsense-viewer and rs-depth-quality).  So you could try skipping the installation of these graphics-based tools by adding **-DBUILD_GRAPHICAL_EXAMPLES=FALSE** to the CMake build flags to see whether it makes a difference.

dongdongbh, 2023-05-16:
``` 
cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_GRAPHICAL_EXAMPLES=FALSE
sudo make uninstall && make clean && make -j8
```
works!

closed 2023-05-16 by dongdongbh