realsenseai/librealsense #11811: Is the depth solution of D435 based on binocular parallax principle or structured light scheme?

pgswag, 2023-05-17:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Is the depth solution of D435 based on binocular parallax principle or structured light scheme?>

MartyG-RealSense, 2023-05-17:
Hi @pgswag  The RealSense 400 Series camera range, including D435, uses **stereo depth** constructed from left and right sensors.  The depth pixel value is a measurement from the parallel plane of the imagers and not the absolute range.

[image]

The left and right imagers capture the scene and send imager data to the depth imaging (vision) processor, which calculates depth values for each pixel in the image by correlating points on the left image to the right image and via the shift between a
point on the Left image and the Right image.  The depth pixel values are processed to generate a depth frame. Subsequent depth frames create a depth video stream. 

The RealSense SR300 and SR305 models use **coded light** technology, wich is similar to structured light.

pgswag, 2023-05-18:
Thank you very much for your answer.

pgswag, 2023-05-18:
“The depth pixel value is a measurement from the parallel plane of the imagers and not the absolute range.”
So that means the depth is “a” as shown in the figure below instead of “b”?
[image]

pgswag, 2023-05-18:
When I hold D435 at an Angle of zero to the wall, is the depth of all points in D435 field of vision theoretically the same? For example, the point in the center of the field of view has the same depth as the vertex of the four corners of the field of view?
[image]
In other words, the depth is the projection onto the z axis not the distance from the origin, right?

MartyG-RealSense, 2023-05-18:
A diagram at https://github.com/IntelRealSense/librealsense/issues/7279#issuecomment-690188950 provided by a RealSense team member further illustrates the relationship between range and Z-depth.

They add "The content of the depth frame is "Z" values calculated for every pixel in the camera's frustum (or cropped FOV).  In the sketch it is clear that while range (or radial distance) may coincide with depth (or "Z"), in 99.99% they will be different".

pgswag, 2023-05-31:
Thank you for your answer!

closed 2023-05-31 by pgswag

5204338, 2024-04-01:
> The left and right imagers capture the scene and send imager data to the depth imaging (vision) processor, which calculates depth values for each pixel in the image by correlating points on the left [...]

Based on your response, the principle of depth measurement of the Realsense D435 camera is based on binocular vision, while the Realsense D300 series camera is based on structured light measurement. Is that correct?

> Hi @pgswag The RealSense 400 Series camera range, including D435, uses **stereo depth** constructed from left and right sensors. The depth pixel value is a measurement from the parallel plane of the imagers and not the absolute range.
> 
> ![image](https://private-user-images.githubusercontent.com/41145062/238959809-bd34d0ec-4641-4344-be73-4d289ff99af6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTE5NDA4NTEsIm5iZiI6MTcxMTk0MDU1MSwicGF0aCI6Ii80MTE0NTA2Mi8yMzg5NTk4MDktYmQzNGQwZWMtNDY0MS00MzQ0LWJlNzMtNGQyODlmZjk5YWY2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDAxVDAzMDIzMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWE1OGE0NThkNGQ4NGM1YWE4ZGU5MjI1ZjAyZGVlYzgyODVkZTU0MjZiNTUwOTQ5MzBlZTUxNzJjMDUyYzEyNDAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.onxABlBbbLtvKbeYEm8n8tVr7K9H4FnAGjet-NKQo6w)
> 
> The left and right imagers capture the scene and send imager data to the depth imaging (vision) processor, which calculates depth values for each pixel in the image by correlating points on the left image to the right image and via the shift between a point on the Left image and the Right image. The depth pixel values are processed to generate a depth frame. Subsequent depth frames create a depth video stream.
> 
> The RealSense SR300 and SR305 models use **coded light** technology, which is similar to structured light.

Based on your response, the principle of depth measurement of the Realsense D435 camera is based on binocular vision, while the Realsense D300 series camera is based on structured light measurement. Is that correct?

5204338, 2024-04-01:
OK,I know the result，Thank you very much.