realsenseai/librealsense #11812: How to save a file in point cloud format?

mengxinglvhappy, 2023-05-17:
| Camera Model |D415 |
| Firmware Version | 05.13.00.50 |
| Operating System & Version | WIN11 |
| Platform | PC|
| SDK Version | SDK2.0 |
| Language | C++ |
| Segment |others |
### Issue Description
<Describe your issue / question / feature request / etc..>
I recorded a video using RealSense Viewer, and the video is saved as a .bag file.After i use this code：

```#include <librealsense2/rs.hpp>
#include <iostream>
int main()
{
    rs2::pointcloud pc;
    rs2::points points;
    rs2::pipeline pipe;
    rs2::config cfg;
    cfg.enable_device_from_file("kailelvbo.bag");
    pipe.start(cfg);
    auto frames = pipe.wait_for_frames();
    auto depth = frames.get_depth_frame();
    auto RGB = frames.get_color_frame();
    pc.map_to(RGB);
    points = pc.calculate(depth);
    auto vertices = points.get_vertices();
    auto tex_coords = points.get_texture_coordinates();
    points.export_to_ply("ak.ply",RGB);} 
```

I get .ply file,but when I opened it, it was in mesh format instead of point cloud format.
[image]

After I use the example：pointcloud .this one:
[image]
Here is my code

```#include <librealsense2/rs.hpp>
#include "example.hpp"   
#include <algorithm>           
void register_glfw_callbacks(window& app, glfw_state& app_state);
int main(int argc, char* argv[]) try
{
    rs2::config cfg;
    cfg.enable_device_from_file("kailelvbo.bag");
    window app(1280, 720, "RealSense Pointcloud Example");
    glfw_state app_state;
    register_glfw_callbacks(app, app_state);
    rs2::pointcloud pc
    rs2::points points;
    rs2::pipeline pipe;
    pipe.start(cfg);
    while (app) 
    {
        // Wait for the next set of frames from the camera
        auto frames = pipe.wait_for_frames();
        auto color = frames.get_color_frame();
        if (!color)
            color = frames.get_infrared_frame();
        pc.map_to(color);
        auto depth = frames.get_depth_frame();
        points = pc.calculate(depth);
        app_state.tex.upload(color);
        draw_pointcloud(app.width(), app.height(), app_state, points);
    }
    return EXIT_SUCCESS;
}
catch (const rs2::error& e)
{
    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    return EXIT_FAILURE;
}
catch (const std::exception& e)
{
    std::cerr << e.what() << std::endl;
    return EXIT_FAILURE;
}
```

**but I only get a window with a pointcloud ,i don't know how to save this pointcloud.How can I save a PLY file in point cloud format instead of mesh format?Like the feature in Realsense Viewer：**
[image]

MartyG-RealSense, 2023-05-17:
Hi @mengxinglvhappy  The **save_to_ply()** instruction provides the configuration options that are offered by the RealSense Viewer's exporter.  An example of a C++ script that demonstrates its use can be found at https://github.com/IntelRealSense/librealsense/issues/6677#issuecomment-1101712660

The official SDK documentation for its configuration options can be found at the link below.

https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1save__to__ply.html

mengxinglvhappy, 2023-05-18:
I have read #6677，and i have solved my problem
```
#include <librealsense2/rs.hpp>
#include <iostream>
#include <rs_export.hpp>
int main()
{
    rs2::pointcloud pc;
    rs2::points points;
    rs2::pipeline pipe;
    rs2::config cfg;
    cfg.enable_device_from_file("kailelvbo.bag");
    pipe.start(cfg);
    auto frames = pipe.wait_for_frames();
    auto depth = frames.get_depth_frame();
    auto RGB = frames.get_color_frame();
    pc.map_to(RGB);
    points = pc.calculate(depth);
    auto vertices = points.get_vertices();
    auto tex_coords = points.get_texture_coordinates();
   // points.export_to_ply("ak.ply",RGB);
    rs2::save_to_ply exporter("231.ply",pc);
    exporter.set_option(rs2::save_to_ply::OPTION_PLY_BINARY, 0.f);
    exporter.set_option(rs2::save_to_ply::OPTION_IGNORE_COLOR, 0.f);
    exporter.set_option(rs2::save_to_ply::OPTION_PLY_MESH, 0.f);
    exporter.export_to_ply(points, RGB);}
```
should change this “private” in rs_export.hpp to “public”
[image]

[image]

MartyG-RealSense, 2023-05-18:
Hi @mengxinglvhappy  I see that you re-opened the issue after achieving a solution.  Do you require further assistance, please?  Thanks!

closed 2023-05-18 by mengxinglvhappy