realsenseai/librealsense #11814: when open the togle the jetson nano freeze

LeoVerheyden, 2023-05-17:
| | |
|---|---|
|**librealsense**|2.53.1 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D435I|
|**Serial Number**|239122070694|
|**Firmware Version**|05.14.00.00|
|**Advanced Mode**|NO|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|243323067229|
|**Firmware Update Id**|243323067229|

Please provide a description of the problem

MartyG-RealSense, 2023-05-18:
Hi @LeoVerheyden  Could you provide more information about your problem, please?  What it is that you are toggling?  Thanks!

MartyG-RealSense, 2023-05-27:
Hi @LeoVerheyden  Do you require further assistance with this case, please?  Thanks!

dmipx, 2023-05-29:
@LeoVerheyden please submit `dmesg` log when it happens.

MartyG-RealSense, 2023-06-03:
Hi @LeoVerheyden  Are you able to generate a dmesg log as @dmipx suggested in the comment above, please?  Thanks!

closed 2023-06-21 by MartyG-RealSense