realsenseai/librealsense #11815: Librealsense needs sudo on macOS Monterey and Ventura

jackjansen, 2023-05-17:
I am opening this issue because #9916 has become unwieldy and has an awful lot of information that is beside the point.

Maintainers: feel free to close-as-duplicate if you think this is a bad idea.

My hope is that there are some people out there who have an idea which MacOS change is the cause of this issue, then we should be able to come up with a fix or workaround.

I presume the root cause is the continuous effort by Apple to harden MacOS against malware, that started with MacOS 10.8 with signed applications, System Integrity Protection, and all that. But I have no idea which specific feature has caused this behaviour.

### Issue Description

Since MacOS Monterey any program that uses librealsense needs `sudo` rights. The error differs depending on the program, but it will often say something like "failed to set power state" or "failed to claim usb interface" or both:

```
 17/05 21:52:52,400 INFO [0x1f8359b40] (context.cpp:382) Found 1 RealSense devices (mask 0xff)
 17/05 21:52:52,416 ERROR [0x16e0df000] (handle-libusb.h:125) failed to claim usb interface: 0, error: RS2_USB_STATUS_ACCESS
 17/05 21:52:52,416 ERROR [0x16df3b000] (sensor.cpp:664) acquire_power failed: failed to set power state
```

```
There are 1 connected RealSense devices.
rs_error was raised when calling rs2_create_device(info_list:0x6000024b7360, index:0):
    failed to set power state
```

Requiring superuser rights is always inconvenient, but sometimes impossible (for example when trying to use librealsense from within `Unity`).

MartyG-RealSense, 2023-05-18:
Hi @jackjansen  Whilst I do not have advice that I can offer on this subject, let's keep the case open for a couple of weeks to give other Mac users the opportunity to comment.  You could also post a message at https://github.com/IntelRealSense/librealsense/issues/9916 inviting the Mac users in that discussion to comment with advice here.

jackjansen, 2023-05-22:
I have been searching through the libusb issues, and a few seem promising as they discuss a problem similar to ours (unable to open USB devices on Monterey or later). Also: these issues are often with "camera-like" devices (which the Realsense also is: it shows up in Apple applications like Photo Booth).

The central issue in the libusb repo seems to be https://github.com/libusb/libusb/issues/1014

A potential workaround seems to be in https://github.com/libusb/libusb/issues/972#issuecomment-1411919134

That references Apple documentation https://developer.apple.com/documentation/coremediaio/overriding_the_default_usb_video_class_extension?language=objc
which states:

> Starting in macOS 12.3, the operating system provides a class-compliant Core Media I/O (CMIO) extension that enables the use of USB cameras and other video capture devices. When you connect a USB camera, the system finds its default driver and connects it with the camera.

This could be the smoking gun: if Apple has handed the Realsense camera (or one of its sensors) off to its own driver then `libusb` (and therefore `librealsense`) can no longer access it.

jackjansen, 2023-05-22:
Searching a bit further has finally led me to a libusb FAQ entry that seems to indicate the issue is unsolvable:

https://github.com/libusb/libusb/wiki/FAQ#how-can-i-run-libusb-applications-under-mac-os-x-if-there-is-already-a-kernel-extension-installed-for-the-device-and-claim-exclusive-access

I have tried unloading the KEXT that handles USB video cameras, by doing

```
sudo kmutil unload -b com.apple.UVCService
```

but it fails with 

```
Error Domain=KMErrorDomain Code=71 "Kernel request failed: (libkern/kext) kext is in use or retained (cannot unload) (-603946984)" UserInfo={NSLocalizedDescription=Kernel request failed: (libkern/kext) kext is in use or retained (cannot unload) (-603946984)}
```

timprepscius, 2023-05-22:
As I understand it, Apple started requiring elevated authority to draw more power than "normal."

Out of curiosity, I asked ChatGPT if it knew of any work arounds.  I wonder if the powered USB hub would have any result.

> USB devices that require more power than what is typically provided by a USB port may require higher privileges, such as root access, to establish the necessary processes and enable their functionality. This restriction is in place for security reasons and to prevent unauthorized devices from drawing excessive power, potentially causing damage to the system or compromising its stability.
> 
> By default, macOS enforces power limits on USB ports to ensure the safe operation of connected devices. When a USB device exceeds these limits, the operating system will prevent it from fully functioning unless appropriate authorization is granted. This authorization can be obtained by running the process with root privileges.
> 
> However, there are potential workarounds you can try:
> 
> Use a powered USB hub: If your USB device requires additional power, connecting it through a powered USB hub that has its own power supply can provide the necessary power without triggering the system's power restrictions.
> 
> Adjust power settings: In some cases, adjusting the power settings of your USB ports may help. You can try accessing the Energy Saver preferences in System Preferences and disabling any power-saving options related to USB devices. Keep in mind that altering these settings may affect the overall power management of your system.
> 
> Utilize third-party software: There are third-party applications available that may allow you to adjust power settings or bypass certain restrictions on USB devices. However, be cautious when using such software, as it may have compatibility issues or security implications.
> 
> It's important to note that modifying power settings or bypassing security measures can have unintended consequences and may potentially compromise the stability or security of your system. Therefore, exercise caution and ensure that any modifications you make align with your specific needs and are in accordance with the manufacturer's recommendations for the USB device in question.
>

cansik, 2023-05-22:
@jackjansen I just started looking into this issue again and I would agree with you. RealSense cameras are shown as UVC cameras, which means they are already accessed by the kext (`com.apple.UVCService`). So the idea would be to create a custom USB video class extension for the realsense cameras, which runs in user space and can be detached by libusb, right?

Otherwise, if it would be possible to disable UVC in the FW, macOS would maybe not attach the device.

jen-co, 2023-05-23:
Thanks for opening this fresh new issue @jackjansen - the spam from the previous one was getting annoying. 

While I don't have any insights on how to fix the root issue I noticed you mentioned that you weren't able to access the camera from Unity. In case you haven't tried it already if you open the Unity editor with root privileges from the command line then camera access is possible. 

Thanks also for all the digging so far to try get to the bottom of this issue!

MartyG-RealSense, 2023-05-23:
Thanks very much @timprepscius @jen-co  and @cansik for your advice on this issue!

jackjansen, 2023-06-06:
Thank you @timprepscius for that note on USB power draw. My feeling is that this is a completely different issue from the UVC issue. Also: it seems this power-draw issue *cannot* be worked around by running with `sudo`. At least: not always.

I've been doing a couple of experiments with a D415 on two machines running Ventura 13.3 or Ventura 13.4: Mac mini M1 and an Intel MacBook Pro 2019.

The Mac mini M1 does worst: nothing works. I always get 

```
 06/06 13:11:39,221 ERROR [0x16fbbf000] (sensor.cpp:664) acquire_power failed: failed to set power state
```

Tried with and without `sudo`. Tried a USB-A to USB-C cable. Tried a USB-C to USB-C cable connected to the USB4 port. Tried USB-A cable connected to powered USB3 hub (RSHTECH RSH-516) and another USB-A to C cable to the D415.

I have not been able to find the Energy Saver settings that ChatGPT came up with, so it must have hallucinated those (at least for Ventura).

The Intel MacBook does a bit better, but in a very weird way:
- USB-C to USB-C cable: `sudo real sense-viewer` fails with `failed to set power state`.
- USB-C from front port on the MacBook(the one far from the screen) to Satechi Type C Multiport Adapter, then USB-A to C cable to D415: does not work. It makes no difference if I connect the power supply to the Satechi or not.
- USB-C from back port on the MacBook(the one near to the screen) to Satechi Type C Multiport Adapter, then USB-A to C cable to D415: Works??!?!? Only with `sudo`, but it works. It does not matter whether I connect the power supply to the Satechi or not.

My current theory is that there is something with the USB port on the computer (maybe having PD enabled?) that makes a difference. If other people could experiment and report that would be helpful.

MartyG-RealSense, 2023-06-16:
Hi @jackjansen  Do you require further assistance with this case, please?  Thanks!

MMYoloCoding, 2023-06-20:
Thank you everyone for the support on this issue. I am a little confused, however. The depth camera can still be opened and used, but it requires an annoying sudo everytime, is this the problem or the camera simply doesn't work on Mac Ventura? I just bought a D435f and a D435if online but I have only a M2 macbook, i'm hoping to help solve this issue before my cameras arrive.

MMYoloCoding, 2023-06-20:
Has any Mac user tried to use parallel windows to open the camera? I hope someone can test try this method since my cameras haven't arrived.

yugasun, 2023-06-27:
> 

> Has any Mac user tried to use parallel windows to open the camera? I hope someone can test try this method since my cameras haven't arrived.

@MMYoloCoding  I have run librealsense successfully, below is my system environment:

Required Info |  
-- | --
Camera Model | D435
Firmware Version | N/A
Operating System & Version | **MacOS M2, Ventura (13.2)**
SDK Version | 2.51.1
Language | python3.10

Refer to my repo: https://github.com/yugasun/pyrealsense2-mac

cansik, 2023-06-27:
@yugasun It would be more interesting to here how you solved it, instead of posting a link to your repository? And what does _solve_ mean? Does it run without sudo, without any problems?

Looking at your repository, it seems that you translated the powershell script to bash. Are there changes fixing the libusb kext problem?

yugasun, 2023-06-28:
@cansik By using [pyrealsense2-mac](https://github.com/yugasun/pyrealsense2-mac), you can run without `sudo`.

yugasun, 2023-06-28:
> @yugasun It would be more interesting to hear more about how you solved it, instead of just posting a link to your repository? And what does _solve_ mean? Does it run without sudo, without any problems?
> 
> Looking at your repository, it seems that you translated the powershell script to bash. Are there changes fixing the libusb kext problem?

@cansik Just run sucessfully with `sudo`.

jackjansen, 2023-06-28:
@cansik Do you have any idea _why_ your pyrealsense2-mac fixes the issue? (Unfortunately I cannot try it myself at the moment)

cansik, 2023-06-28:
@jackjansen My build script uses the most recent libusb version, which allowed it to support M1 chips. But other than that the main reason for the repository was to create macOS releases, because Intel is not publishing wheels for macOS on pypi.

jackjansen, 2023-06-29:
@cansik I have no success with your https://github.com/yugasun/pyrealsense2-mac repository. I finally got it to build `realsense-viewer` and the resulting executable behaved exactly like the `brew`-installed one. Works with `sudo` but not without.

But I had to make various changes to your build script before it built `realsense-viewer` at all:
- Changed cmake options to build examples and graphical examples
- Build only single-architecture
- `brew install glfw`

cansik, 2023-06-29:
@jackjansen Ok, I guess there is a misunderstanding and this is exactly what I was worried about: The confusion between @yugasun's fork and the one I was working on for over two years (https://github.com/cansik/pyrealsense2-macosx/). But regarding your question: Yes, the `pyrealsense-macosx` version needs sudo as well. There is no workaround at the moment.

@yugasun Could you please explain the purpose behind launching your own version instead of contributing to the existing project through a pull request? Now we have `pyrealsense2-macosx` and `pyrealsense2-mac` on github and pypi, which seems a bit confusing for the users and even myself.

yugasun, 2023-06-30:
> 

> @jackjansen Ok, I guess there is a misunderstanding and this is exactly what I was worried about: The confusion between @yugasun's fork and the one I was working on for over two years [...]

@cansik Sorry for the mistake, I just publish `pyrealsense2-mac` for personal use, I will unpublish `pyrealsense2-mac`.

AkshajGupta, 2023-07-04:
@yugasun 
I am following the instructions and am running into errors 
I ran 
`pip install pyrealsense2-mac`
and got 
```
ERROR: Could not find a version that satisfies the requirement pyrealsense2-mac (from versions: none)
ERROR: No matching distribution found for pyrealsense2-mac
```

AkshajGupta, 2023-07-04:
Update: I figured out the issue in the last message I sent

I am running into new error when using `sudo realsense-viewer`
` Error during polling error handler: failed to set power state`

MartyG-RealSense, 2023-07-18:
Hi @AkshajGupta  Were you able to resolve your problem, please?  Thanks!

sidekicktec, 2023-09-09:
My Solution in TouchDesigner:
https://forum.derivative.ca/t/intel-realsense-with-m1-chips/352284/4?u=sidekicktec

MartyG-RealSense, 2023-09-09:
Thanks so much @sidekicktec for sharing the M1 installation solution that worked for you!

closed 2023-09-15 by MartyG-RealSense

MartyG-RealSense, 2024-09-03:
Hi everyone, 

The current latest 2.56.1 version of the librealsense SDK contains a fix for SDK compilation on MacOS Apple Silicon 'M' processors and also an update for MacOS 14 (Sonoma). 

These changes are confirmed in 2.56.1's release notes.

[image]

2.56.1 is currently an 'interim beta' release of the SDK which can only be installed by source code compilation. 

https://github.com/IntelRealSense/librealsense/releases/tag/v2.56.1

jackjansen, 2024-12-18:
@MoAbbasid here is what I see with a D435 on my M1 Mac mini 2020 (running Sequoia 15.1.1, to be precise).

System Information -> Hardware -> USB:

```
Intel(R) RealSense(TM) Depth Camera 435 :

  Product ID:	0x0b07
  Vendor ID:	0x8086  (Intel Corporation)
  Version:	51.00
  Serial Number:	843213020743
  Speed:	Up to 5 Gb/s
  Manufacturer:	Intel(R) RealSense(TM) Depth Camera 435 
  Location ID:	0x00230000 / 7
  Current Available (mA):	900
  Current Required (mA):	720
  Extra Operating Current (mA):	0
```

And it is indeed detected as a UVC device: if I open Photo Booth it immediately shows the image that comes from the D435 RGB sensor...

yrichi, 2026-10-29:
For L515 owners finding this thread: the 2.57.6+ macOS fixes mentioned above never reached the L515, since L500 support was removed in v2.55.1. I've published a fork that restores it on v2.59.1 and makes it work **natively on Apple Silicon** (M-series, macOS 26, fw 1.5.8.1): https://github.com/yrichi/librealsense (branch `exp1-merge-2.59.1`, builds on the L500/T265 restoration by @eoffermann).

Measured: depth 1024×768 @ 30 fps for 2 h with 0 dropped frames, 100/100 open/close cycles, IMU at full rate (gyro ~200 Hz), `.bag` recording.

On the original question of this issue — *why sudo*: since macOS 12.3, `UVCAssistant` claims the camera's UVC interfaces exclusively. libusb on Darwin can only detach it by capturing the **whole device**, which requires root or the `com.apple.vm.device-access` entitlement, which Apple reserves for VM software (libusb/libusb#1014, closed as "known limitation"). So sudo is structural, not a librealsense bug.

Two hard-won warnings in the fork's README: the hidapi HID path is dead under whole-device capture (the IMU needs the generic libusb path), and a process that dies while capturing leaves the L515's depth function wedged **on every host** until the camera is unplugged for ~2 minutes — fw 1.5.8.1 is final, so that will never be fixed.

MartyG-RealSense, 2026-10-30:
Thanks so much @yrichi for the detailed explanation and for creating a custom fork of the SDK with modernized L515 support for MacOS users who want to use their camera with a recent SDK!

yrichi, 2026-10-30:
Thanks @MartyG-RealSense ! 
Since fixing the SDK, I’ve been using it quite extensively, so we’ll see if any other minor issues pop up.
If they do, I’ll try to fix them and push the changes to the fork.

I’m really glad this camera doesn’t have to end up gathering dust in a drawer.