realsenseai/librealsense #11816: I want to use Dual cam D435f cameras

KITAE-KIM-MOBYUS, 2023-05-18:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435f | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win 10)  | 
| Kernel Version (Linux Only)    |                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C#}                          | 
| Segment			|  {Robot                        | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I want to use two D435f cameras and see color images from each selection. Can I watch the video at the same time? Or can I see them one by one? I would like an article worth reading.

MartyG-RealSense, 2023-05-18:
Hi @KITAE-KIM-MOBYUS  I believe that this question is related to your earlier issue at https://github.com/IntelRealSense/librealsense/issues/11756

 You should be able to access a color stream from both cameras if you have the two cameras on different pipelines like in https://github.com/IntelRealSense/librealsense/issues/3432#issuecomment-472570380

I do not have a script reference for **displaying** streams from the different cameras though.  The structure of the code is likely to depend on whether your C# script is using WPF or WinForms.

MartyG-RealSense, 2023-05-27:
Hi @KITAE-KIM-MOBYUS  Do you require further assistance with this case, please?  Thanks!

dmipx, 2023-05-29:
@KITAE-KIM-MOBYUS 
> I want to use two D435f cameras and see color images from each selection. Can I watch the video at the same time? Or can I see them one by one? I would like an article worth reading.
you can use "+ Add Source" button on realsense-viewer and select second camera.
[image]

MartyG-RealSense, 2023-06-03:
Hi @KITAE-KIM-MOBYUS  Was the suggestion above from @dmipx to add the second camera to the RealSense Viewer using the 'Add Source' option so that both cameras could be viewed helpful to you, please?  Thanks!

closed 2023-06-14 by MartyG-RealSense