realsenseai/librealsense #11817: Cannot stream images from D457 on Jetson Orin

dmellosanjay, 2023-05-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D457 | 
| Firmware Version                   | 05.14.00.00 | 
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    | 5.10.104                                         | 
| Platform                                 | NVIDIA Jetson Orin  |
| Language                            |  C++                         | 

### Issue Description

I can stream images from the D457's devices (IMU, RGB, Depth etc) using `v4l2-ctl`. But when I try to stream images using v2.53 of `librealsense` I see these error messages in the kernel logs.

```
[ 1490.024072] d4xx 72-001b: DEPTH START, stream_status 0x1004:2, config_status 0x4800:1
[ 1490.025233] bwmgr API not supported
[ 1490.200903] tegra-camrtc-capture-vi tegra-capture-vi: corr_err: discarding frame 0, flags: 0, err_data 131072, vc: 2
[ 1490.201838] (NULL device *): vi_capture_control_message: NULL VI channel received
[ 1490.202061] t194-nvcsi 13e40000.host1x:nvcsi@15a00000: csi5_stream_close: Error in closing stream_id=0, csi_port=0
[ 1490.202361] (NULL device *): vi_capture_control_message: NULL VI channel received
[ 1490.202575] t194-nvcsi 13e40000.host1x:nvcsi@15a00000: csi5_stream_open: VI channel not found for stream- 0 vc- 2
[ 1490.285087] d4xx 74-001b: Y8 START, stream_status 0x1014:2, config_status 0x4808:1
[ 1491.936240] d4xx 72-001b: ds5_send_hwmc(): HWMC failed, ret: 0, status: 1, error code: -21
[ 1492.299329] d4xx 72-001b: ds5_get_hwmc(): HWMC failed, ret: 0, error code: 256
[ 1498.654867] d4xx 72-001b: DEPTH STOP, stream_status 0x1004:1, config_status 0x4800:0

```

I am using Connect Tech's BSP for Jetpack 5.0.2 that has the drivers required for the D457 to work with Connect Tech's Expansion Board (JCB002)

MartyG-RealSense, 2023-05-19:
At https://github.com/IntelRealSense/librealsense/issues/10984#issuecomment-1486939651 a D457 user states that Connect Tech's instructions were to use JetPack **4.6.1**.  They comment on this again at https://github.com/IntelRealSense/librealsense/issues/10984#issuecomment-1518671668

The librealsense SDK does not have official support for JetPack 5.0.2 at the time of writing this.  It has been added to the next SDK version, which will likely be numbered 2.54.0.  This support is currently on the development branch of the SDK, which becomes the latest 'master' SDK when pushed to release.

https://github.com/IntelRealSense/librealsense/tree/development

It is possible to use Orin with JetPack 5 and SDK 2.53.1 if a kernel patch is not applied or if the SDK is built from source code using the **RSUSB backend** installation method to bypass the kernel - see https://github.com/IntelRealSense/librealsense/issues/10722#issuecomment-1230005808

Ultimately though, I recommend enquiring to Connect Tech's support if you continue to experience problems with this issue.

dmellosanjay, 2023-05-22:
@MartyG-RealSense  
> At [#10984 (comment)](https://github.com/IntelRealSense/librealsense/issues/10984#issuecomment-1486939651) a D457 user states that Connect Tech's instructions were to use JetPack **4.6.1**. They comment on this again at [#10984 (comment)](https://github.com/IntelRealSense/librealsense/issues/10984#issuecomment-1518671668)

I did see these posts, but since I'm using an Orin I can only use a Jetpack 5-compatible BSP.

> It is possible to use Orin with JetPack 5 and SDK 2.53.1 if a kernel patch is not applied or if the SDK is built from source code using the **RSUSB backend** installation method to bypass the kernel - see [#10722 (comment)](https://github.com/IntelRealSense/librealsense/issues/10722#issuecomment-1230005808)

The D457 is detected only when I disable the RSUSB backend.

> Ultimately though, I recommend enquiring to Connect Tech's support if you continue to experience problems with this issue.

 I will contact Connect Tech to check what version of librealsense their BSP([CTI-L4T-AGX-35.1.0-V001-REALSENSE](https://connecttech.com/ftp/Drivers/CTI-L4T-ORIN-AGX-35.1.0-V001-INTELRS.tgz)) was tested against.

Thanks for your quick reply!

MartyG-RealSense, 2023-05-23:
You are very welcome.  I look forward to your next report.  Good luck!

dmipx, 2023-05-29:
Hi @dmellosanjay.
We have suport for JetPack 5.0.2. Do you have any updates on this issue?

dremsol, 2023-06-02:
@dmipx I'm receiving an Jetson Orin Nano devkit next week. How could I test the support?

MartyG-RealSense, 2023-06-02:
My understanding at the time of writing this is that official support for 5.0.2 has been added to the **development** branch of librealsense already but is not yet in the master branch (currently 2.53.1) until the development branch gets pushed to release and becomes the next 'latest' master version of the SDK.  I believe that in SDK 2.53.1, JetPack 4.6 is still currently the newest officially supported JetPack.

https://github.com/IntelRealSense/librealsense/tree/development

However, JetPack 5.0.2 can be used with Orin now, by either not applying kernel patching or by building librealsense from source code with CMake with the **RSUSB backend** installation method - see https://github.com/IntelRealSense/librealsense/issues/10722#issuecomment-1230250762

MartyG-RealSense, 2023-06-07:
Hi everyone,

The new 2.54.1 version of librealsense with JetPack 5.0.2 support is now available.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.54.1

Release notes:
https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2541

dmellosanjay, 2023-06-14:
I was able to stream frames from the depth and RGB sensors by running the SDK examples `save-to-disk` and `hello-realsense`

Thanks for your support!

MartyG-RealSense, 2023-06-14:
You are very welcome.  It's great to hear that the examples were able to stream successfully for you.  Thanks very much for the update!