realsenseai/librealsense #11818: Cannot flash D457 over I2C/GMSL

dmellosanjay, 2023-05-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D457 | 
| Firmware Version                   | 05.14.00.00 | 
| Operating System & Version |   Ubuntu 20.04  | 
| SDK                                          | 2.53.1   |
| Kernel Version (Linux Only)    | 5.10.104                                      | 
| Platform                                 | NVIDIA Jetson Orin  |

### Issue Description

With the device in MIPI/GMSL mode, I ran the command to update the firmware, but it failed.
`rs-fw-update -f Signed_Image_UVC_5_14_0_0.bin` 

After setting it in USB mode I was able to successfully update the firmware.

MartyG-RealSense, 2023-05-19:
Hi @dmellosanjay  I will enquire to my Intel RealSense colleagues regarding the possibility of updating firmware over a GMSL connection instead of USB.

MartyG-RealSense, 2023-05-19:
My colleagues responded that you are an Intel non-disclosure agreement (NDA) customer and could you therefore submit your question on the Zendesk support ticket system, please.  Thanks!

MartyG-RealSense, 2023-05-27:
Hi @dmellosanjay  Have you submitted a support ticket on Zendesk, please?  Thanks!

dmellosanjay, 2023-05-27:
@MartyG-RealSense When I try to login to realsensesupport.intel.com (the new portal?) it says social account login is not allowed. I was told the team at Intel is looking into it a week ago.

MartyG-RealSense, 2023-05-27:
Hi @dmellosanjay  I am not familiar with that site but I was able to log into it and it just appears to be an old list of information resource links.  The address of the other main RealSense support forum, which is not Zendesk, is https://support.intelrealsense.com/hc/en-us/community/topics

dmipx, 2023-05-29:
Hi @dmellosanjay 
I'm not sure this feature fully verified with realsense tools.
Please try the direct way for d457 case:
`cat Signed_Image_UVC_5_14_0_0.bin > /dev/d4xx-dfu504`

dmellosanjay, 2023-05-31:
@dmipx I tried running the command, but there were no progress indicators in the kernel logs even after 20 minutes.

hsuys, 2023-05-31:
Please check if the `/dev/d4xx-dfu504` is enumerated.
The `cat Signed_Image_UVC_5_14_0_0.bin > /dev/d4xx-dfu504` takes approximately 90 seconds to complete. 
A screenshot for example.
[image]

MartyG-RealSense, 2023-06-06:
Hi @dmellosanjay  Was the advice of @dmipx and @hsuys in the comment above helpful to you, please?  Thanks!

MartyG-RealSense, 2023-06-18:
Case closed due to no further comments received.

closed 2023-06-18 by MartyG-RealSense

tvonhausenCTI, 2023-11-03:
@hsuys @MartyG-RealSense 

I've created a ticket with Intel Realsense as well but will open back this thread. 
I've tried the method to update firmware using the command `cat Signed_Image_UVC_5_14_0_0.bin > /dev/d4xx-dfu504` but do not see any activity in the dmesg when running this and the command hangs.
I've checked the /dev/ list and /dev/d4xx-dfu504 is enumerating properly.

Are you aware of any issues with this process?
We are running this on our AGX201 carrier for Jetson AGX Orin. JetPack version 5.1.1.

MartyG-RealSense, 2023-11-04:
Hi @tvonhausenCTI  I provided a response on the Intel RealSense Help Center to your question there.

https://support.intelrealsense.com/hc/en-us/community/posts/22439941418003/comments/22462284261139

If you have created a private Zendesk ticket then that will be the best place to receive advice on this issue.

Therkelsen, 2024-05-16:
Any updates on this?

MartyG-RealSense, 2024-05-16:
Hi @Therkelsen  The release notes for the current latest 2.55.1 version of librealsense list "rs-fw-update tool - handle D457 MIPI device"

dmipx, 2024-05-19:
> Any updates on this?

Hi. We are working on refactoring D457 firmware update procedures in SDK, PR will be submitted soon.