realsenseai/librealsense #11819: Attempt to uncompress the video stream in RS server for transmission

H-chun, 2023-05-19:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D457 } | 
| Firmware Version                   | 2.53.1 | 
| Operating System & Version |   jetpack 5.1.1  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | NVIDIA Jetson  |
| SDK Version                            |  2.53.1                         | 
| Language                            |  c++                         | 
| Segment			|  Robot                         | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Hello! At present, I have found that during the process of transmitting video streams on RS server, both depth maps and rgb maps have been compressed. I would like to modify the code myself to not compress the transmission.
There are currently two issues:
1. On the server side, where is the specific SDK for obtaining video streams, encoding, and transmitting the code
2. On the client side, where is the SDK for obtaining video streams through IP, decoding, and rendering the code to the UI interface

MartyG-RealSense, 2023-05-19:
Hi @H-chun  The issue of controlling compression in rs-server may be simpler than the two questions that you asked.  The section of the **RsServer.cpp** code at the link below highlights the section of the file that determines if compression is enabled.

https://github.com/IntelRealSense/librealsense/blob/master/tools/rs-server/RsServer.cpp#L42-L56

H-chun, 2023-05-25:
Hello! I have a few more questions I would like to ask you
1. I saw that setting - c after rs server indicates compression transmission, and not setting - c means no compression transmission. Is that correct?
2. How can I check the size of a frame of image in RS server transmission, and where can I see it in the code?
3. The last question is that the 720p, 30fps rs server has been supported in the latest sdk, but the depth map will be too laggy. Will you fix this. Or how can we improve this phenomenon?

MartyG-RealSense, 2023-05-25:
1.  It does appear that using -c enables compression according to line 42 of RsServer.cpp.

https://github.com/IntelRealSense/librealsense/blob/master/tools/rs-server/RsServer.cpp#L42

2.  Data size can be retrieved from a frame's 'frame handle' - see https://github.com/IntelRealSense/librealsense/issues/5087#issuecomment-550459833 -  with the instruction **get_data_size**.

https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1frame.html#ab3665db740b156cc9ef447796d349d31

https://github.com/IntelRealSense/librealsense/issues/5718 may be a helpful C++ reference for using get_data_size.

4.  The networking tool is being removed from the RealSense SDK in the next version, likely numbered 2.54.0.  This is because the tool was an experimental proof of concept rather than an official, validated Intel RealSense feature.  So it will not receive further updates, unfortunately.

You could try implementing a post-processing **Decimation** filter to reduce the complexity of the depth image, like in the C++ **rs-post-processing** example program.

https://github.com/IntelRealSense/librealsense/blob/master/examples/post-processing/rs-post-processing.cpp#L73

H-chun, 2023-05-26:
Hello, I have looked at the code and found that it does not implement uncompressed transmission? Because I tried using the following two commands to transmit video streams and found that both were compressed?
rs-server -c
rs-server

MartyG-RealSense, 2023-05-26:
The section of the tool's documentation linked to below explains which compression methods are applied.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#33-software-architecture

H-chun, 2023-05-26:
Hello, what I mean is that as you mentioned earlier, the transmission can be uncompressed. You only need to set the corresponding parameters, such as rs-server-c and - c, to change the compression mode of the transmission. But currently, the results of my testing have not changed. Adding - c and not, both are for compressed transmission

MartyG-RealSense, 2023-05-26:
I said that it appeared that -c affected compression, judging by the code of RsServer.cpp.  I was not certain of this though, which is why I said 'appeared', and unfortunately I do not have any further advice that I can offer about this particular issue.  I do apologize.

closed 2023-06-06 by MartyG-RealSense