realsenseai/librealsense #11820: Can't play recorded video due to bus errorr

aphinex1110, 2023-05-19:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux (Ubuntu 18.04.6 LTS)   | 
| Kernel Version (Linux Only)    |  4.9.253                                         | 
| Platform                                 |  NVIDIA Jetson Nano |
| SDK Version                            |  2.50.0                          | 
| Language                            |  C++                           | 
| Segment			|  Object detection                     | 

### Issue Description

Hello together,

I always receive a "Bus error" message, as soon as I try to play a recorded video (bag format). After I get this error, my application crashes. 

However I tried the rs-record-playback example from **examples/record-playback/rs-record-playback.cpp** (Version 2.53.1), which works perfectly fine. 

The "Bus error" message directly appears when I try to start the pipeline with pipe->start(cfg). Recording however, seems to work as intended. 

Below you can see my code so far, which is basically the same approach as in **examples/record-playback/rs-record-playback.cpp**: 

```cpp
    rs2::frameset frames;
    rs2::frame depth;              
                    
    auto pipe = std::make_shared<rs2::pipeline>();
    pipe->start();
    std::cerr << "Pipe 1 started" << std::endl;
    rs2::device device = pipe->get_active_profile().get_device();

    if (!device.as<rs2::playback>())
    {       
        frames = pipe->wait_for_frames(); // wait for next set of frames from the camera
        std::cerr << "Got some frames" << std::endl;
        depth = color_map.process(frames.get_depth_frame()); // Find and colorize the depth data
    }
        
    // START RECORDING
    if (!device.as<rs2::recorder>())
    {       
        pipe->stop(); // Stop streaming with default configuration
        pipe = std::make_shared<rs2::pipeline>();
        rs2::config cfg;
        cfg.enable_record_to_file("a2.bag");
        std::cerr << "Enable device to file" << std::endl;
        pipe->start(cfg); //File will be opened in read mode at this point
        std::cerr << "Pipe 2 started" << std::endl;
        device = pipe->get_active_profile().get_device();
    }

    // recording some frames
    for(int i = 0; i < 30; i++){
        std::cerr << "Recording frame.." << std::endl;
         frames = pipe->wait_for_frames(); // wait for next set of frames from the camera      
    }      

    // STOP RECORDING
    std::cerr << "Stop recording" << std::endl;
    pipe->stop(); // Stop streaming with default configuration
    pipe = std::make_shared<rs2::pipeline>();
    pipe->start();
    device = pipe->get_active_profile().get_device();

    // PLAY recorded device
    if (!device.as<rs2::playback>())
    {       
        pipe->stop(); // Stop streaming with default configuration
        pipe = std::make_shared<rs2::pipeline>();
        rs2::config cfg;
        cfg.enable_device_from_file("a2.bag");
        std::cerr << "Enable device from file" << std::endl;
        pipe->start(cfg); //File will be opened in read mode at this point
        std::cerr << "Pipe 3 started" << std::endl;
        device = pipe->get_active_profile().get_device();
    }  
                  
    rs2::playback playback = device.as<rs2::playback>();
    if (pipe->poll_for_frames(&frames)) // Check if new frames are ready
    {
        depth = color_map.process(frames.get_depth_frame()); // Find and colorize the depth data for rendering
    }     
```

The "Bus error" message I receive exactly after I called cfg.enable_device_from_file("a2.bag") and want to start the pipeline with pipe->start(cfg) right after: (please see below)  
```cpp
    // PLAY recorded device
    if (!device.as<rs2::playback>())
    {       
        pipe->stop(); // Stop streaming with default configuration
        pipe = std::make_shared<rs2::pipeline>();
        rs2::config cfg;
        cfg.enable_device_from_file("a2.bag");
        std::cerr << "Enable device from file" << std::endl;
        pipe->start(cfg); //File will be opened in read mode at this point
        std::cerr << "Pipe 3 started" << std::endl;
        device = pipe->get_active_profile().get_device();
    }                   
```

Right now I am using librealsense2 version 2.50.0 on my Jetson Nano. Could that be the root cause why I'm getting a "Bus error"?

I also tried to play a recorded video, which was generated by this script **examples/record-playback/rs-record-playback.cpp**, but I still receive the same error. 

I don't know how to proceed anymore, so I decided to create this issue. I would really appreciate it, if anyone has some ideas.

Thank you and best regards,
aphinex1110

MartyG-RealSense, 2023-05-20:
Hi @aphinex1110  If you were previously using SDK version 2.53.1 and the **rs-record-playback.cpp** example program worked, do you have firmware version **5.14.0.0** installed in your camera, please?  If you do then the firmware version should be downgraded to **5.13.0.50** when using SDK 2.50.0, as firmware 5.14.0.0 is not compatible with SDK versions older than 2.53.1.

aphinex1110, 2023-05-20:
@MartyG-RealSense Thank you for your prompt answer.

I just checked which firmware version I have with **_rs-fw-update -l_** command.
It says firmware version: 5.12.14.5
So firmware version should be fine for SDK 2.50.0 or do I need to upgrade to version 5.13.0.50?

MartyG-RealSense, 2023-05-20:
5.12.14.5 is only one firmware version older than the recommended one for 2.50.0. (5.13.0.50) so it should be okay for use with 2.50.0.

A RealSense team member provides advice about the error at https://github.com/IntelRealSense/librealsense/issues/4519#issuecomment-515759414

aphinex1110, 2023-05-20:
@MartyG-RealSense kk thank you very much for the information.

Unfortunatly, I am still receiving the same bus error message. 

Now I am trying to install the newest SDK version 2.51.3 on my Jetson Nano. 

For this purpose I first uninstalled and removed all realsense2 data from previous verison with the following command:
**_dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge_**

Then I installed librealsense2 newest version according to the instructions from [librealsense](https://github.com/IntelRealSense/librealsense)/[doc](https://github.com/IntelRealSense/librealsense/tree/master/doc)/installation_jetson.md

Everything got installed correctly without any issues. But when I execute my cmake file, I always get the following error message after calling find_package(realsense2 REQUIRED):

```console
CMake Error at /usr/local/lib/cmake/realsense2/realsense2Targets.cmake:92 (message):
  The imported target "realsense2::realsense2" references the file

     "/usr/local/lib/librealsense2.so.2.50.0"

  but this file does not exist.  Possible reasons include:

  * The file was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and contained

     "/usr/local/lib/cmake/realsense2/realsense2Targets.cmake"

  but not all the files it references.

Call Stack (most recent call first):
  /usr/local/lib/cmake/realsense2/realsense2Config.cmake:35 (include)
  CMakeLists.txt:26 (find_package)
```

It is still looking for /usr/local/lib/librealsense2.so.2.50.0 even though it does not exist anymore.

aphinex1110, 2023-05-20:
Short Update:

New SDK version 2.53.1 is now working as intended. 

However I still receive the same bus error, as soon as I try to play a recorded bag file...

MartyG-RealSense, 2023-05-21:
Checking your script against rs-record-playback, I note that you have the following section added in the recording part of your script:

```
// recording some frames
for(int i = 0; i < 30; i++){
std::cerr << "Recording frame.." << std::endl;
frames = pipe->wait_for_frames(); // wait for next set of frames from the camera      
}    
```

rs-record-playback only has one wait_for_frames() instruction, whilst adding this section of new code means that your script has two wait_for_frames().  So you could try commenting out this section in case it is corrupting your bag recording.

If you are using a for(int) to skip the first several frames in order to enable auto-exposure to settle down first, you could move the for(int)  to directly above the remaining one wait_for)frames.

```
   if (!device.as<rs2::playback>())
    {       

    for(int i = 0; i < 30; i++)
    {

        frames = pipe->wait_for_frames(); // wait for next set of frames from the camera

}

        std::cerr << "Got some frames" << std::endl;
        depth = color_map.process(frames.get_depth_frame()); // Find and colorize the depth data
    }

```

aphinex1110, 2023-05-21:
I just tried your suggested steps but I still receive the same bus error message. 

I have also tried to use the recorded bag file from rs-record-playback.cpp example as input for my script. This way I modified the code, so that only playing the bag file will be executed:
```cpp
std::cout << "RS2 Version = " << RS2_API_VERSION_STR << std::endl;

rs2::frameset frames;
rs2::frame depth;             

auto pipe = std::make_shared<rs2::pipeline>();
rs2::config cfg;
cfg.enable_device_from_file("a.bag");
pipe->start(cfg); 
             
if (pipe->poll_for_frames(&frames)) // Check if new frames are ready
{
    depth = color_map.process(frames.get_depth_frame()); // Find and colorize the depth data for rendering
}     
```
However, I still receive the same bus error when calling pipe->start(cfg), even with the generated and working bag file from rs-record-playback.cpp example.

I also checked if I'm really using realsense2 version 2.53.1 with:
```cpp
std::cout << "RS2 Version = " << RS2_API_VERSION_STR << std::endl;
```
And it says 2.53.1, so everything should be installed correctly. 

But I still don't get it, why the rs-record-playback.cpp example is working. In my opinion the root cause must be either some wrong function call in my script or some part of realsense2 is still not installed and missing?

MartyG-RealSense, 2023-05-22:
If you are copying code from the rs-record-playback.cpp example project into a new project that you have created and are compiling with CMake in Ubuntu then there may be missing linkages to the librealsense SDK in the CMakeLists.txt file.  

The link below has a simple C++ starter-project tutorial for using CMake to build a simple application that integrates the SDK.

https://github.com/zivsha/librealsense/tree/getting_started_example/examples/getting-started

aphinex1110, 2023-05-23:
Goods news.

I just used the same steps in my CMakeLists.txt as in https://github.com/zivsha/librealsense/tree/getting_started_example/examples/getting-started and finally it works. 

Apparently I had to use 
`find_library(REALSENSE2_FOUND realsense2 HINTS ${LIBRARY_DIR} REQUIRED)`
instead of 
`find_package(realsense2 REQUIRED)` in my CMake. 

@MartyG-RealSense Thank you so much for your help, I wouldn't have found it on my own. 

Best regards and all the best
aphinex1110

MartyG-RealSense, 2023-05-23:
You are very welcome.  It's great to hear that you were successful.  Thanks very much for the update!

closed 2023-05-30 by MartyG-RealSense