realsenseai/librealsense #11821: Maximum Resolution Change Yields Failed

shihaocao, 2023-05-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { d435f } | 
| Firmware Version                   | 05.14.00.00 | 
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.14.0-1059-oem                                         | 
| Platform                                 | NVidia Jetson Orin  |
| SDK Version                            |  { librealsense2.53 }                          | 
| Language                            |  {C++}                          | 
| Segment			|  { Robotics }                          | 

### Issue Description

## What changed:
I recently upgraded our code to capture at 1920 x 1080 for color stream and  depth streams from our intel realsense d435f. In the past we ran 848x400 for both cameras. And then we started noticing the issue.

_note all of the below are using our own internal frame numbers (not RS frame number)_

## The Issue:

It has caused a very peculiar issue where after exactly 3 frames, we drop frames 4 - 17 inclusive. After which the next wait_for_frames(), repeats frame 3 forever. When this issue occurs the depth frames are always black and 0 valued everywhere. This has happened twice so far, and its always 3 good frames, then 4 - 17 are null, and then frame 18 returns the same data as frame 3.

This issue will persist for 10000+ frames.

The settings we use are applied here:
    FPS = 10
    depth_width: 1280,
    depth_height: 720,
    color_width: 1920,
    color_height: 1080
[image]

My capture code and frame checking occurs here:
[image]
[image]

[image]

Here's how I can tell that the frames continue to be stale and how my frames 4 - 17 are dropped. Notice how the file sizes stop changing on the color frames (_C_)
[image]

I call my wait_for_frames in a loop that fires at about 9.5Hz.

The issue fixes itself if I stop my binary and restart it. This issue happens about 5% of my binary runs so far, over at least two runs.

## Request

Is there any way to fix this issue without restarting my binary? (and closing and reopening the pipeline) I am running this code on agricultural equipment and once we have started capturing frames its imperative that the depth code actually works, otherwise, the equipment will fail to do its job.

Please let me know what other information I can provide!

MartyG-RealSense, 2023-05-20:
Hi @shihaocao  Does the problem still occur if you use 1280x720 color instead of 1920x1080, please?

When auto-exposure is enabled, the first few frames after starting a program are spent settling down the auto-exposure.  So it may be worth testing disabling auto-exposure to see whether it is the cause of your issue when using 1920x1080 color.  It can be disabled using code like this:

`sensor.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 0);`

shihaocao, 2023-05-20:
Hi Marty!

Unfortunately, configuration changes like that are a bit hard to test at the moment since all of our realsenses are out on production units... and we need the full resolution of the color sensor and auto exposure. I can definitely try at my next chance though.

### New information:

Most interestingly though, this appears to happen around about 1/25 to 1/50 captures, where capture is a pipeline start, followed by calls to `wait_for_frames()`. This also makes it very hard to test if configuration/if the resolution change actually caused the issue or if I just didn't notice it before...

The "issue" seems slightly more likely to strike if the previous binary was interrupted/killed instead of naturally shutting down and perhaps letting destructors be called properly?

I have seen it occur on a fresh jetson boot + realsense powerup + binary run. Could anything persist on the realsense despite a full system power cycle?

Also, ctually checked my code, and I don't ever call `.stop()` on my pipeline. Do I need to make sure that is called even if I only call pipeline `.start()` once per program?

Thanks!

MartyG-RealSense, 2023-05-20:
If your program needs to run continuously then don't call **stop()**.  A program should be able to run continuously for a very long time period anyhow, as the pipeline can only hold up to 16 frames at a time in its 'queue' for each stream type by default and drops old frames from the queue as new ones arrive, releasing memory resources.

If a program is improperly closed then problems can occur in a subsequent session of that program when it is run again.  

When rebooting a computer to try to resolve a problem, a cold shutdown (full power off and start-up with the power button) is more likely to resolve the issue than a reset, where the issue may persist after reboot.

closed 2023-06-02 by MartyG-RealSense