realsenseai/librealsense #11823: "not found applicable to the completion of the Image processing components of the operation."

975085538, 2023-05-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D455 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (11)  | 
| Kernel Version (Linux Only)    |                  | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.53.1 }                          | 
| Language                            |  {C# }                          | 
| Segment			|  {others }                          | 

### Issue Description
I want by WPF open folder import raw format file and displayed in the Image, but there have been some exceptions: **System. A NotSupportedException:** "not found applicable to the completion of the Image processing components of the operation."

Here's my code: 
```
// 读取 file path
                string filePath = openFileDialog.FileName;

                // 创建新的 BitmapImage 对象
                BitmapImage bitmapImage = new BitmapImage();

                if (filePath.EndsWith(".raw")) //如果是 .raw 格式文件
                {
                    using (FileStream stream = File.Open(filePath, FileMode.Open))
                    {
                        int width = 424;
                        int height = 240;
                        int bytesPerPixel = 1;

                        // 计算文件流的长度
                        long fileLength = stream.Length;

                        // 计算图像的总像素数
                        int pixelCount = width * height;

                        // 计算每个像素的字节数
                        int bytesPerRow = width * bytesPerPixel;
                        byte[] rawData = new byte[fileLength];

                        // 从文件中读取所有数据
                        stream.Read(rawData, 0, (int)fileLength);

                        // 调整的原始数组以匹配给定的宽度和高度
                        byte[] rgbaData = new byte[pixelCount * 4];
                        for (int i = 0; i < pixelCount; i++)
                        {
                            int index = i * bytesPerPixel;
                            rgbaData[i * 4 + 0] = rawData[index];
                            rgbaData[i * 4 + 1] = rawData[index];
                            rgbaData[i * 4 + 2] = rawData[index];
                            rgbaData[i * 4 + 3] = 255;
                        }

                        bitmapImage.BeginInit();
                        bitmapImage.CreateOptions = BitmapCreateOptions.None;
                        bitmapImage.CacheOption = BitmapCacheOption.OnLoad;
                        bitmapImage.StreamSource = new MemoryStream(rgbaData);
                        bitmapImage.DecodePixelWidth = width;
                        bitmapImage.DecodePixelHeight = height;
                        bitmapImage.EndInit();
                    }
                }
```

The exception appears on this line:**bitmapImage.EndInit();**

MartyG-RealSense, 2023-05-22:
There are no past RealSense C# cases involving this error to refer to, unfortunately.  

I also looked at non-RealSense cases.  One that resembles your problem is at the link below.

https://stackoverflow.com/questions/26404496/wpf-bitmapimage-endinit-throwing-notsupportedexception

MartyG-RealSense, 2023-05-30:
Do you require further assistance with this case, please?  Thanks!

975085538, 2023-05-31:
> 请问您需要有关此案的进一步帮助吗？谢谢！

I will try to find a feasible method again, thank you for your help!

closed 2023-05-31 by 975085538