realsenseai/librealsense #11833: Realsense interferes with libcamera

kjfullert, 2023-05-23:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                        |                         D435                              | 
| Firmware Version                    |                   05.12.07.150                         | 
| Operating System & Version  |             Linux Raspberry Pi OS 64 bit      | 
| Kernel Version (Linux Only)     |                      5.15.61-v8+                       | 
| Platform                                  |                         Raspberry Pi                   |
| SDK Version                            |                              2.51.1                        | 
| Language                                |                                  C                           | 
| Segment			         |                                 Robot                     | 

### Issue Description
In the following scenario librealsense appears to cause libcamera to crash:

1. Plug in a realsense camera (D435) and a csi camera (ov9281)
2. Run libcamera-hello -t 50000000 in a terminal 
3. Wait for it to start streaming video
4. Run rs-align in another terminal
5. The libcamera-hello application will crash.  The librealsense file will continue to run.

This makes it difficult to use the librealsense camera with other cameras at the same time.  Can this be solved? 

Here is the output from libcamera-hello.  The error occurs at 0:01:14.
[image]

MartyG-RealSense, 2023-05-24:
Hi @kjfullert  The camera firmware driver 5.12.7.150 is very old, dating back to August 2020, and was designed for librealsense version **2.38.1**.   The recommended firmware version for librealsense 2.51.1 is **5.13.0.50**, which is seven versions newer than 5.12.7.150.  Please first try updating your firmware to see whether this makes a difference to your problem.  Thanks!

kjfullert, 2023-05-24:
Hi @MartyG-RealSense! I update the firmware as instructed, but it doesn't solve the problem.

MartyG-RealSense, 2023-05-24:
Does libcamera crash if **rs-hello-realsense** (a depth-only example program) is run instead of rs-align?  If libcamera is not crashed by rs-hello-realsense then it might suggest that rs-align is trying to access RGB from the csi camera.

As you are using a Raspberry Pi, another possibility is that the Pi does not have enough power to support two cameras simultaneously.  Usually with RealSense cameras, only one camera can be used on a Pi board.

kjfullert, 2023-05-24:
Hi @MartyG-RealSense thanks for the quick responses! Much appreciated. Even running rs-hello-realsense instead of rs-align will cause the other camera to crash. 

We are using a RaspberryPi, however, when I run the realsense camera first, and then the OV9281 csi camera second and leave enough time for the realsense to start running, then the csi camera and the realsense run together fine.  This suggests it isn't a problem with the raspberry pi not able to support the power from them both.

kjfullert, 2023-05-24:
@MartyG-RealSense I've also tested the realsense + an usb webcam on an earlier version of 32 bit raspberry pi OS (buster), which is not using libcamera and I can get 5 video streams (usb webcam, realsense rgb, realsense depth, realsense ir - right and left) running simultaneously at approximately 30fps without browning out as seen in the attached picture. This suggests that the raspberry pi has the ability to support the realsense and another camera. Unfortunately we cannot use buster for our final product and need the 64bit raspberry pi OS (bullseye). 

[image]

MartyG-RealSense, 2023-05-25:
Thanks very much for the information about using Bullseye.  There have been past Raspberry Pi reports of RealSense cameras working fine on Buster but not on Bullseye.  A solution that a RealSense user shared at https://github.com/IntelRealSense/librealsense/issues/9943#issuecomment-966048785 is to compile Buster and then copy two binary files from it over to Bullseye.

MartyG-RealSense, 2023-05-31:
Raspberry Pi provide a 64-bit version of Buster at the link below.

https://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2021-05-28/

By default, librealsense on Linux uses V4L2 as its backend instead of RSUSB when installing from packages or when building from source code with the CMake flag **-DFORCE_RSUSB_BACKEND=FALSE**

When the V4L2 backend is used, standard Linux tools are able to access the raw RealSense data streams, though they will not be colored as nicely as when the streams are accessed through librealsense.

MartyG-RealSense, 2023-06-01:
I do not have information about the two binary files, but the commenter in that thread says that they copied files from the **Documents** folder, so maybe you could look there to see if there are only two binaries.  If there are more, then copy over all the binaries in that folder to Bullseye and see what happens.

MartyG-RealSense, 2023-06-13:
Case closed due to no further comments received.

closed 2023-06-13 by MartyG-RealSense

kjfullert, 2023-07-09:
Never solved this problem but we switched back to Buster to move forward with our project.

MartyG-RealSense, 2023-07-09:
Okay, thanks very much for the update!