realsenseai/librealsense #11835: Realsens-viewer Invalid argument

Hayouung-Park, 2023-05-24:
| | |
|---|---|
|**librealsense**|2.53.1 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D455|
|**Serial Number**|035322250731|
|**Firmware Version**|05.14.00.00|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|025523050851|
|**Firmware Update Id**|025523050851|

Please provide a description of the problem

Hello, I tried to connect my d445 depth camera to realsense-viewer.
I installed SDK2.0
But, when I strated realsense-viewer, the error was occured.
[image]
[image]

MartyG-RealSense, 2023-05-24:
Hi @Hayouung-Park  A **xioctl(VIDEOC_C_CTRL)** error typically indicates a conflict with the Linux kernel.  Which kernel version are you using, please?  Thanks!

Hayouung-Park, 2023-05-24:
my kernel version is Ubuntu 20.04
Thnaks!
[image]

MartyG-RealSense, 2023-05-24:
Thank you.  20.04 is the Ubuntu version.  The kernel version will typically begin with '5.', such as '5.15'.  You can find your kernel version with the Ubuntu command **uname -r**

Hayouung-Park, 2023-05-24:
sorry.
my kernel version is 5.15
[image]

MartyG-RealSense, 2023-05-24:
Thank you.  Which method did you use to install librealsense?  Was it from packages or built from source code?

Hayouung-Park, 2023-05-24:
I followed all steps of github method.
[here](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md)

MartyG-RealSense, 2023-05-24:
As you installed librealsense from packages, it will automatically include a kernel patch for 5.15 and so it is it necessary to patch the kernel. This does not prevent the possibility of a conflict with the kernel for some reason though, perhaps due to something on the particular computer that you are using.

Which computer / computing device are you using, please (Raspberry Pi, Nvidia Jetson, desktop/ laptop PC, etc).

Hayouung-Park, 2023-05-24:
I'm using my laptop PC

MartyG-RealSense, 2023-05-24:
A patch only needs to be applied if building librealsense from source code instead of packages.  When building from source code and using kernel 5.15 and Ubuntu 20.04, patch script **./scripts/patch-realsense-ubuntu-lts-hwe.sh** would be run

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

When was the last time that the camera worked correctly?  Has it been stored unused in a drawer or cupboard for weeks / months?

Hayouung-Park, 2023-05-24:
I bough it three weeks ago...
And I have used it from two weeks ago. The error also has been occured when I used it first time.
I already patched. And other webcams are working well.

MartyG-RealSense, 2023-05-24:
As you applied a kernel patch before performing a package installation of librealsense (which has the kernel patch included in the packages already), I would recommend next trying to remove the patched kernel using the instructions in the link below.  Then install an unpatched 5.15 kernel.

https://askubuntu.com/questions/1338398/how-do-i-remove-newest-kernel

Hayouung-Park, 2023-05-25:
1. Sure
2. Core dupmed when I did rs-capture which is in the usr/local/bin folder.

[image]

3. It has didn't work from the first time I used.

MartyG-RealSense, 2023-05-25:
Please next try resetting the camera to its default factory-new calibration settings using the instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487

If this does not work then it is likely that your D455 has an uncorrectable internal hardware fault.  In that case it should be returned to the retailer that it was purchased from.

Hayouung-Park, 2023-05-25:
Thank you for answer.
[image]
I did what you suggested. But it doesn't work. 
I also think it should be returned to the retailer. I'll answer about this issue after receive new camera.

Hayouung-Park, 2023-05-30:
I received new camera. And it is working correctly.
So I think it was hardware problem.
Thank you for answer

MartyG-RealSense, 2023-05-30:
It's great to hear that a replacement camera resolved your problem.  Thanks very much for the update!

closed 2023-05-30 by MartyG-RealSense