realsenseai/librealsense #11836: A problem was encountered when measuring the distance between two points

YANSU7, 2023-05-24:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435  | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 | PC  |
| SDK Version                            | v2.49.0                      | 
| Language                            |  C++                        | 
| Segment			|                        | 

### Issue Description
Hi，MartyG.
I managed to get the code in rs-measure.cpp to run in my own project using VS2019. However, I encountered some problems during the subsequent distance measurement.
I obtained the outline of the target by means of target detection and found two points, which I am sure have the correct coordinates in the RGB image. I then found that due to the alignment function of the align_to function, I had to do some conversion of the coordinates obtained from the RGB image to make it fit the input of get_distance.
Unfortunately, I did not calculate the correct distance between the two points by this conversion, and the results I calculated were often infinite or unusually large. I have filled all the voids, so voids should not be the problem.
My RGB image and depth image both have a resolution of 640*480, but the image gets smaller after alignment, as shown below：
[image]
[image]
[image]

Here are my codes:
```
float dist_3d(const rs2::depth_frame& frame, cv::Point2d points1, cv::Point2d points2)
{
    float upixel[2]; // From pixel    
    float upoint[3]; // From point    (in 3D)
    float vpixel[2]; // To pixel      
    float vpoint[3]; // To point      (in 3D)

    // points1.x and points1.y are both decimals.
    int point_start_x = static_cast<int>(95 + points1.x * 417);
    int point_start_y = static_cast<int>(83 + points1.y * 303);
    int point_end_x = static_cast<int>(95 + points2.x * 417);
    int point_end_y = static_cast<int>(83 + points2.y * 303);

    if (point_start_x < 94 || point_start_x > 512 || point_start_y < 94 || point_start_y > 512)
    {
        point_start_x = 95;
        point_start_y = 83;
        point_end_x = 95;
        point_end_y = 83;
    }
  
    cout << point_start_x << "**" << point_start_y << endl;
    cout << point_end_x << "**" << point_end_y << endl;

    float udist = frame.get_distance(point_start_x, point_start_y);  
    float vdist = frame.get_distance(point_end_x, point_end_y);   
    // Deproject from pixel to point in 3D
    rs2_intrinsics intr = frame.get_profile().as<rs2::video_stream_profile>().get_intrinsics(); // Calibration data
    rs2_deproject_pixel_to_point(upoint, &intr, upixel, udist);
    rs2_deproject_pixel_to_point(vpoint, &intr, vpixel, vdist);
    // Calculate euclidean distance between the two points
    return sqrt(pow(upoint[0] - vpoint[0], 2.f) +
        pow(upoint[1] - vpoint[1], 2.f) +
        pow(upoint[2] - vpoint[2], 2.f));
}
```

MartyG-RealSense, 2023-05-24:
Hi @YANSU7  The rs-measure example program is an unusual alignment program as it uses the very rare **color to depth** alignment method instead of the commonly used depth to color.  What this means is that the field of view size of the color sensor is increased to match the field of view size of the depth sensor, which causes the aligned image to stretch out to fill the screen.  

My understanding is that use of color to depth alignment is not vital to the program and it was used simply to provide a demonstration of color to depth alignment.  You could try changing the align_to instruction to be **depth to color** alignment and see if that improves your results.

`rs2::align align_to(RS2_STREAM_COLOR);`

YANSU7, 2023-05-25:
I am very pleased to receive your reply.
I tried to make some changes to the code as you requested and my test results did not show much improvement.
Also, I am sure that the method I used for target detection to get the points of type Point is fine.
All my doubts are actually concentrated in this latter piece of code:
```
    float udist = frame.get_distance(point_start_x, point_start_y);  
    float vdist = frame.get_distance(point_end_x, point_end_y);   
    // Deproject from pixel to point in 3D
    rs2_intrinsics intr = frame.get_profile().as<rs2::video_stream_profile>().get_intrinsics(); // Calibration data
    rs2_deproject_pixel_to_point(upoint, &intr, upixel, udist);
    rs2_deproject_pixel_to_point(vpoint, &intr, vpixel, vdist);
    // Calculate euclidean distance between the two points
    return sqrt(pow(upoint[0] - vpoint[0], 2.f) +
        pow(upoint[1] - vpoint[1], 2.f) +
        pow(upoint[2] - vpoint[2], 2.f));
```
Does the get_distance in this code require any input parameters? It would be good to specify the exact type of the parameter, I'm a bit PTSD on the type error issue!
If you have time I would also like you to explain in detail how the camera calculates the distance in 3D space from the pixel coordinates (x, y) step by step.
I look forward to your detailed reply!

YANSU7, 2023-05-26:
Hello MartyG!
Thank you so much for your prompt reply.
I have changed the alignment.`rs2::align align_to(RS2_STREAM_COLOR);`

What function should I use to change the depth intrinsics or the colour intrinsics that you mentioned? I'm sorry I'm not very good with your code.

Then I would also like to ask you a logical question:
First I use target detection to get the pixel coordinates of the two coordinates I need to detect, and then I input this coordinate value into get_distance. 
Is the above process consistent with the process of measuring the distance between two points?
Do I need to do any other transformations before inputting the pixel coordinates into get_distance?
Am I missing something that is causing my measurement to have inaccuracy problems?

MartyG-RealSense, 2023-05-26:
Your intrinsics look as though they are already set up in the line below.

`rs2_intrinsics intr = frame.get_profile().as<rs2::video_stream_profile>().get_intrinsics(); // Calibration data`

Your get_distance code also looks fine.

YANSU7, 2023-05-27:
Hi MartyG!
I rebuilt my program with the code from rs-measure, unlike last time I didn't use dual threads, but this gave better results.
Thank you very much for your answers to my questions!
I hope you are working well!

MartyG-RealSense, 2023-05-27:
You are very welcome.  I'm pleased to hear that you were successful  :)

closed 2023-06-02 by MartyG-RealSense