realsenseai/librealsense #11838: Some Problems on the Internal Parameter Matrix of L515 Camera

HGS-1998, 2023-05-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515} | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) } | 
| Kernel Version (Linux Only)    |  ？？？      | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
How to adjust the internal reference matrix of two L515 cameras to be the same?

This is the internal reference matrix of one of my cameras：
Deep frame internal parameter matrix：
[ 960x540  p[495.936 270.658]  f[674.108 674.523]  Brown Conrady [0.179601 -0.559083 -0.00375945 0.000501796 0.527874] ]
Color frame internal reference matrix：
[ 960x540  p[495.936 270.658]  f[674.108 674.523]  Brown Conrady [0.179601 -0.559083 -0.00375945 0.000501796 0.527874] ]

This is the internal parameter matrix of another camera：
Deep frame internal parameter matrix：
 [ 960x540  p[491.242 274.202]  f[675.649 675.635]  Brown Conrady [0.162009 -0.504752 -0.00187483 0.000331119 0.475713] ]
Color frame internal reference matrix：
[ 960x540  p[491.242 274.202]  f[675.649 675.635]  Brown Conrady [0.162009 -0.504752 -0.00187483 0.000331119 0.475713] ]

How to adjust the internal parameter matrices of these two cameras to be consistent?
Please help me！

The problem I want to solve is:
Fixed camera position, fixed position of measured object
At the same camera position, switch to a different L515 camera and measure the three-dimensional coordinates of the object, in order to keep the three-dimensional coordinates of the object unchanged. Is it determined by unifying the internal parameter matrices of two L515 cameras?

MartyG-RealSense, 2023-05-25:
Hi @HGS-1998  https://github.com/IntelRealSense/librealsense/issues/4061 provides an example of Python scripting for manually defining the intrinsic values to use.

HGS-1998, 2023-05-25:
> Hi #4061 提供了一个 Python 脚本示例，用于手动定义要使用的内部值。

Thanks for your help,I have another question that I would like to ask you to help me solve.

Will the internal reference matrices of the camera's depth frames and color frames change after registration?

This is an internal parameter matrix that has not been registered:

[image]

This is the internal parameter matrix for registration:
[image]

Does this mean that the internal parameter matrix will change when the deep stream aligns with the color stream for registration?

MartyG-RealSense, 2023-05-31:
Hi @HGS-1998  Do you require further assistance with this case, please?  Thanks!

HGS-1998, 2023-05-31:
> Hi @HGS-1998 Do you require further assistance with this case, please? Thanks!

There are currently no issues. Thank you for your help

MartyG-RealSense, 2023-05-31:
You are very welcome.  As you do not require further assistance, I will close this issue.  Thanks very much for the update!

closed 2023-05-31 by MartyG-RealSense

HGS-1998, 2023-06-04:
> 非常欢迎你。由于您不需要进一步的帮助，我将关闭此问题。非常感谢您的更新！

Hello, I'm very sorry to bother you again. My question is: I have two cameras with model L515, how can I ensure that they have the same world coordinate system in the same space? (Simply put, I want two cameras to shoot the same object in the same position and obtain the same three-dimensional coordinates.)

MartyG-RealSense, 2023-06-04:
A way to calibrate multiple L515 cameras together relative to each's world position is to use place boards with **Apriltag** images on them behind the camera.  Intel demonstrated this with nine L515 at the CES electronics show in 2020, as shown in the YouTube video link below..

https://www.youtube.com/watch?v=UzIfn667abE&t=7s

***********

Another approach is to use an **affine transform** to set the images from all cameras to the same position and rotation in 3D space.  The SDK instruction **rs2_transform_point_to_point** is a standard affine transformation using a 3x3 rotation matrix and a 3-component translation vector.  

The function accepts the following parameters :

- "rs2_extrinsics" struct that contains parameters describing relationship between the separate 3D coordinate systems.
-  3D point coordinates.

https://github.com/IntelRealSense/librealsense/issues/1231#issuecomment-368421888
 has an example of a Python script that makes use of rs2_transform_point_to_point.

[image]

HGS-1998, 2023-06-05:
> 根据每个相机的世界位置一起校准多个 L515 相机的一种方法是在相机后面使用带有 **Apriltag** 图像的位置板。英特尔在 515 年的 CES 电子展上用九台 L2020 展示了这一点，如下面的 YouTube 视频链接所示。
> 
> https://www.youtube.com/watch?v=UzIfn667abE&t=7s
> 
> 另一种方法是使用**仿射变换**将所有相机的图像设置为3D空间中的相同位置和旋转。SDK 指令**rs2_transform_point_to_point**是使用 3x3 旋转矩阵和 3 分量平移向量的标准仿射变换。
> 
> 该函数接受以下参数：
> 
> * “rs2_extrinsics”结构，包含描述各个 3D 坐标系之间关系的参数。
> * 3D 点坐标。
> 
> [#1231（评论）](https://github.com/IntelRealSense/librealsense/issues/1231#issuecomment-368421888)有一个使用rs2_transform_point_to_point的Python脚本示例。
> 
> [image]

This is not the answer I want, my question is：I want two cameras to shoot the same object in the same position and obtain the same three-dimensional coordinates.
Using this affine transformation, we can get the transformation between the image coordinate system of an L515 camera, the camera coordinate system and the world coordinate system. But what I mean is, how to make two L515 cameras have the same world coordinate system, you can see the schematic diagram:
[image]

Why are the 3D coordinates not the same when the relative positions of the camera and the object remain unchanged？

MartyG-RealSense, 2023-06-05:
Each individual RealSense camera has unique intrinsics and extrinsics due to the manufacturing process in the factory when the camera is constructed.

MartyG-RealSense, 2023-06-05:
Other than setting custom calibration values for a camera in a script as suggested above at https://github.com/IntelRealSense/librealsense/issues/11838#issuecomment-1562748349 I cannot think of another approach, unfortunately.