realsenseai/librealsense #11839: Realsense L515 camera datastream pulsing issue

srirampr22, 2023-05-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |L515| 
| Firmware Version                   | 01.05.03.00,| 
| Operating System & Version |   Linux (Ubuntu 20.04) | 
| Kernel Version (Linux Only)    |  5.15.0-72-generic                                     | 
| Platform                                 | PC |
| SDK Version                            |  2.53.1                     | 
| Language                            |  ROS1/REalsense-viewer                     | 
| Segment			|  Robot                      | 

### Issue Description
Issue Description

I have been experiencing two issues with my Realsense L515 camera:

    Overheating issue: The camera gets "feverish" and heats up after running for a while. I suspect this is an issue with the hardware and would appreciate any suggestions on how to deal with this.

    Streaming error: When running the realsense-viewer, I get the following error messages only for the depth sensor and the IR sensor:
[image]

I have tried unplugging and flipping the type C end of the connector, have tried downgrading to other compatible firm wares but nothing works.

MartyG-RealSense, 2023-05-25:
Hi @srirampr22  If an L515 is pulsing then it means that the camera has an internal hardware failure, unfortunately.  The issue is discussed at https://github.com/IntelRealSense/librealsense/issues/9505

If the camera was purchased less than a year ago then it should be returned to the retailer that it was purchased from.

MartyG-RealSense, 2023-06-12:
Case closed due to no further comments received.

closed 2023-06-12 by MartyG-RealSense

srirampr22, 2023-06-17:
Hi @MartyG-RealSense sorry for not replying soon, Thank you for your assistance, the issue has been resolved after replacing the cameras with the replacement sent by intel.

MartyG-RealSense, 2023-06-17:
It's no problem at all.   I'm pleased that we could help.  Thanks very much for the update!