realsenseai/librealsense #11843: T265 camera is not detected when i open realsense-viewer.

PotroCoelho, 2023-05-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265 } | 
| Firmware Version                   | (0.2.0.951)| 
| Operating System & Version |   {Ubuntu 22.04 LTS} | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Raspberry Pi
| SDK Version                            |  { 2.53.1)                          | 
| Language                            |  {python}                          | 
| Segment			|  {others }                          | 

### Issue Description
When I open the realsense viewer i get this output 

25/05 17:06:47,633 ERROR [281473491857632] (librealsense-exception.h:52) /dev/video10 is no video capture device Last Error: Invalid argument
 25/05 17:06:47,647 ERROR [281473491857632] (sensor.cpp:664) acquire_power failed: /dev/video10 is no video capture device Last Error: Invalid argument
 25/05 17:06:47,679 WARNING [281473491857632] (rs.cpp:310) null pointer passed for argument "device"
 25/05 17:06:47,680 WARNING [281473491857632] (rs.cpp:2704) Couldn't refresh devices - /dev/video10 is no video capture device Last Error: Invalid argument

The viewer opens, it does not detec a camera and presents the 2 same erros and warnings in the command line. If i unplug and plug the camera again, it is detected and everything runs smoothly. What kind of issue is this? Is it only a detection failure?

Btw, i installed it via source code

DragonOnGit, 2023-05-31:
hi, I am facing the same issue, my friend once solved this problem, but I forgot the method...., I am trying to fix it.

PotroCoelho, 2023-06-01:
@DragonOnGit Good to know that I'm not the only one. If you manage to solve it I'd like to know the solution.

DragonOnGit, 2023-06-02:
By checking the issues before, I found that T265 on Raspi has been given up now. The issues related like #4681 and #4979 said they were moving on to a new product. So if we still want to use this old one, the issue needs to be fixed by ourselves.

Furthermore, I tried many other methods online, while I finish make codes and plug in the camera, use the `lsusb` command in the terminal, the output is 
```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 03e7:2150
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
And use command `dmesg` the output is 
```
[ 5469.101857] usb 1-1.2: new high-speed USB device number 4 using xhci_hcd
[ 5469.202637] usb 1-1.2: New USB device found, idVendor=03e7, idProduct=2150, bcdDevice= 0.01
[ 5469.202654] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 5469.202666] usb 1-1.2: Product: Movidius MA2X5X
[ 5469.202677] usb 1-1.2: Manufacturer: Movidius Ltd.
[ 5469.202686] usb 1-1.2: SerialNumber: 03e72150
```

So still trying to fix it. Do you or anybody have some other solutions?

PotroCoelho, 2023-06-09:
I don't know why but reinstalling librealsense again solved it for me. I dont know why but it is working just fine

finickyDrone, 2023-08-22:
I'm seeing the same issue on the T264 in Ubuntu 20.04 with the latest v2.54.1 version of the RealSense ADK. dmesg shows:
`
[  522.301524] videodev: Linux video capture interface: v2.00
[  522.307186] usb 3-5: Found UVC 1.00 device Integrated_Webcam_HD (0c45:6720)
[  522.314602] input: Integrated_Webcam_HD: Integrate as /devices/pci0000:00/0000:00:14.0/usb3/3-5/3-5:1.0/input/input34
[  522.314684] usbcore: registered new interface driver uvcvideo
[  577.814598] usb 3-1: new high-speed USB device number 7 using xhci_hcd
[  577.963883] usb 3-1: New USB device found, idVendor=03e7, idProduct=2150, bcdDevice= 0.01
[  577.963886] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  577.963888] usb 3-1: Product: Movidius MA2X5X
[  577.963888] usb 3-1: Manufacturer: Movidius Ltd.
[  577.963889] usb 3-1: SerialNumber: 03e72150
`

and lsusb shows:
`Bus 003 Device 007: ID 03e7:2150 Intel Myriad VPU [Movidius Neural Compute Stick]
`
It seems to be misidentifying the T265 for some reason. This is on the latest version of RealSense ADK and I already rebooted and reinstalled RealSense.

FPSychotic, 2023-09-13:
For my life I won't buy nevermore a Intel product.

AllenIsaacJose, 2023-09-15:
SOLVED

```
sudo sh -c "echo 141a0000.pcie > /sys/bus/platform/drivers/tegra194-pcie/unbind"
sudo sh -c "echo 141a0000.pcie > /sys/bus/platform/drivers/tegra194-pcie/bind"
```
Find the PCIe card to which the USB is connected. In my case `141a0000.pcie`. The above command will turn the power supply to the PCIe card off and then on thereby making my T265 tracker camera detectable.

LuanVuNguyen, 2023-11-30:
> SOLVED
>
> ```
> sudo sh -c "echo 141a0000.pcie > /sys/bus/platform/drivers/tegra194-pcie/unbind"
> sudo sh -c "echo 141a0000.pcie > /sys/bus/platform/drivers/tegra194-pcie/bind"
> ```
>
> Find the PCIe card to [...]

Hello,
I also had the same problem. How to search for "141a0000.pcie". Thank you
[image]

AllJordanSS, 2024-04-22:
> RESOLVIDO
> 
> ```
> sudo sh -c "echo 141a0000.pcie > /sys/bus/platform/drivers/tegra194-pcie/unbind"
> sudo sh -c "echo 141a0000.pcie > /sys/bus/platform/drivers/tegra194-pcie/bind"
> ```
> 
> Encontre a placa PCIe à qual o USB está conectado. No meu caso `141a0000.pcie`. O comando acima desligará e ligará a fonte de alimentação da placa PCIe, tornando minha câmera rastreadora T265 detectável.

Resolvi da mesma forma, reiniciando a fonte de energia do hub usb. Em controladoras que não conseguem reiniciar a fonte de energia do hub, é necessário seccionar a alimentação do cabo usb da câmera, adicionando um relé acionado por uma gpio. Assim é possível reiniciar a câmera e utilizá-la sem problemas.
Eu utilizava essa resolução quando usava a câmera com uma Raspberry PI 4, no entanto quando mudei para Jetson Orin Nano, que tem a função de reiniciar a fonte de alimentação do seu hub usb via terminal, não precisei mais.

AllenIsaacJose, 2024-04-22:
> Hello, I also had the same problem. How to search for "141a0000.pcie". Thank you [...]

Hi,

Sorry for the late reply.

This method works only if the USB is connected to a separate PCIe card whose power can be turned off and on programmatically. To find the PCIe card info, if I remember correctly, you can open the realsense-viewer and connect the camera. On the left side, there would be an information panel where you could click on the info button and see the complete path to the USB port including the PCIe card information in the physical port section.

DragonOnGit, 2024-04-28:
> Hi,
>
> Sorry for the late reply.
>
> This method works only if the USB is connected to a separate PCIe card whose power can be turned off and on programmatically. To find the PCIe card info, if I [...]

Do you mean that the power supplement is the reason that the driver recognizeing the camera?

AllenIsaacJose, 2024-04-28:
> Do you mean that the power supplement is the reason that the driver recognizeing the camera?

I am not very sure about the internal software stack. But when the power to the T265 camera is received before the host system is fully booted up, the camera goes into low power mode. So by diagnosing the USB ports, we could see the USB connection to the camera is in 2.0 mode instead of the expected 3.0 mode. As a result, we had to make sure the camera power was received only after the host system was booted up.

ntsarb, 2024-08-06:
I've reached this discussion after testing my RealSense T264 on Windows 11 and finding out it's not recognized in the RealSense Viewer (v2.55.1). Do I need to revert back to an older version in order to use the camera?

closed 2024-08-06 by MartyG-RealSense

ntsarb, 2024-08-06:
> Hi @ntsarb T265 support was removed in version 2.54.1 as it is a retired camera model. You should be able to use your T265 with the Viewer if you download version 2.51.1 of **Intel.RealSense.Viewer.exe** in the 'Assets' file list at the link below.
> https://github.com/IntelRealSense/librealsense/releases/tag/v2.51.1
> As T265 is no longer supported, I will close this issue. You are welcome to respond in the comments though.

Thanks for letting me know. However, these are expensive devices, intended for robotic applications, which are also expensive, and we need to be able to support them for much longer. Please give it some thought because I will be avoiding Intel Realsense devices going forward, if software support is dropped like this.

shubham-shahh, 2024-11-12:
To use T265 with Ubuntu 22.04, we can use v2.51.1 with a small fix

`git clone https://github.com/IntelRealSense/librealsense.git`
`cd librealsense `
`./scripts/setup_udev_rules.sh`
`mkdir build && cd build`
`cmake ../ -DBUILD_EXAMPLES=true -DBUILD_WITH_CUDA=true -DBUILD_WITH_TM2=ON`
`sudo make uninstall && make clean && make && sudo make -j$(($(nproc)-4)) install`

If you get a error related to  Unicode character `´` in file src/libusb/libusb.h
use this patch https://github.com/IntelRealSense/librealsense/pull/10818/files
and run the make command again and the camera should work fine

note: the latest version of SDK does not have examples like rs-pose

phxhuy, 2025-01-15:
For anyone still plays with this legacy device, solution from @shubham-shahh works, just make sure to checkout for branch v2.51.1. You may need to use the branch ros1-legacy as well if you use the realsense_ros wrapper. 

I got it works fine with ros 1 package build from source for ubuntu 22.04. 

For the default ros2 version, I managed to fix the "Unsupported device!" error, but got this error going further:

```
[WARN] [1736951241.119069051] [camera.camera]: Module Name "Tracking Module" does not define a callback.
[ERROR] [1736951241.119112249] [camera.camera]: Error starting device: No known base_stream found for transformations.
```
It certainly needs some further tweaks to work with ros2.

ylenianistico, 2025-01-22:
> Hello, I also had the same problem. How to search for "141a0000.pcie". Thank you [image]

Hi @LuanVuNguyen we're facing the exact same problem. Do you remember if you fixed it, and how?

musstt, 2025-05-05:
> For anyone still plays with this legacy device, solution from [@shubham-shahh](https://github.com/shubham-shahh) works, just make sure to checkout for branch v2.51.1. You may need to use the branch [...]

hey are you fix itu?

66Lau, 2025-05-13:
> To use T265 with Ubuntu 22.04, we can use v2.51.1 with a small fix
>
> `git clone https://github.com/IntelRealSense/librealsense.git` `cd librealsense ` `./scripts/setup_udev_rules.sh` `mkdir build && [...]

After execute these command, `rs-enumerate-devices -s` could recognize T265

But If you also want a ros wrapper,  remember that do NOT install ROS2 wrapper like this `sudo apt install ros-<ROS_DISTRO>-realsense2-*`, they also remove T265 support in the newest version, and this would make your previous compilation fail.

My solution is:

```
mkdir -p ~/ros2_ws/src 
cd ~/ros2_ws/src/
git clone https://github.com/IntelRealSense/realsense-ros.git
cd realsense-ros/
git checkout 4.51.1
cd ~/ros2_ws
sudo apt-get install python3-rosdep -y && rosdep update
rosdep install -i --from-path src --rosdistro humble --skip-keys=librealsense2 -y
colcon build
source /opt/ros/humble/setup.bash
cd ~/ros2_ws
. install/local_setup.bash
```

Then type `rs-enumerate-devices -s` to check connections and `ros2 launch realsense2_camera rs_launch.py serial_no:="'<your device serial number>'"`